Hieromon/AutoConnect #285: PlatformIO does not show v1.2.2 of AutoConnect library.

v-c, 2020-12-13:
Here's an output from my PIO terminal window:

PS D:\GitRepos\TestOTA> date 
Sun, Dec 13, 2020 12:16:29 AM
PS D:\GitRepos\TestOTA> pio lib update
Library Storage: D:\GitRepos\TestOTA\.pio\libdeps\esp32cam
Updating hieromon/AutoConnect                 1.2.1 @ ^1.2.1                     [Up-to-date]

v-c, 2020-12-13:
Also see information here: https://platformio.org/lib/show/2678/AutoConnect/installation

[image]

Hieromon, 2020-12-13:
Hello @v-c,

Thank you for letting me know. The version description of `library.json` as a manifest has been updated correctly, so it's possible that the PIO crawler hasn't caught the update yet.
Release v.1.2.2 commit 0d02b06a3b476371a9f23bae045176af8705fc4b

https://github.com/Hieromon/AutoConnect/blob/0d02b06a3b476371a9f23bae045176af8705fc4b/library.json#L28

I'm not sure about the scan cycle of the PIO library crawler and have no idea to update an AutoConnect registry forcibly in the PIO library. Let me see how it goes.
That being said, 14 hours have passed since the release, so it'd be nice if the PIO library manager would catch the v1.2.2 release.

v-c, 2020-12-13:
Thanks Hieromon. I think the issue is on the platformio side.

v-c, 2020-12-13:
Looks like the crawler finally ran. Marking the issue as closed.

closed 2020-12-13 by v-c

Hieromon, 2020-12-14:
@v-c 
Ok, 
The delay in updating the library registry with crawl seems to have been talked about in the past. It efforts are being made to improve the current "within 24 hours".
https://community.platformio.org/t/web-hook-or-request-based-library-crawler/9367

According to the topic, it would allow for manual refreshing of metadata as well. But It's possible that I'm missing to manually update the registry due to my lack of knowledge of the PIO library commands I have.
Maybe this?
https://docs.platformio.org/en/latest/core/userguide/lib/cmd_update.html