Hieromon/AutoConnect #290: Server does not respont after being inactive for a day

1JonaB1, 2020-12-28:
Hello, Im using the Autoconnect libary in my project on an esp32. I don't know if it is an Feature but after an hour the server takes a while on first use, which would be ok however after a day it doesn't work entirely.

Hieromon, 2021-01-04:
Can you provide logs with AC_DEBUG enabled?
Follow the documentation, see [here](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options) and [here](https://hieromon.github.io/AutoConnect/faq.html#still-not-stable-with-my-sketch).

1JonaB1, 2021-01-05:
Thank you for your answere. Here are my logs:
https://pastebin.com/RnDUNdVq
At the start i needed to connect manualy over the saved networks
At the end i connected to the accespoint again (don't know why)

Hieromon, 2022-01-29:
AutoAutoConnect appears to be working fine.

closed 2022-12-14 by Hieromon