Hieromon/AutoConnect #291: porting code to Platformio

sgrizzi, 2021-01-03:
Hi, I am trying to port my code for ESP32, which is using the Autoconnect, library from the Arduino IDE to Platformio.
The code is running perfectly in Arduino.
When I try and compile the code under Platformio I get the following error which I cannot correct:
....
Compiling .pio/build/esp32doit-devkit-v1/lib509/WiFi/ETH.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib509/WiFi/WiFi.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib509/WiFi/WiFiAP.cpp.o
In file included from .pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnect.h:39:0,
                 from include/PLH_web_handling.h:10,
                 from src/PLH_web_handling.cpp:8:
.pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnectOTA.h:32:13: error: 'SPIFFSFS' in namespace 'fs' does not name a type
 typedef fs::SPIFFSFS  SPIFFST;
             ^
In file included from .pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnect.h:39:0,
                 from include/PLH_web_handling.h:10,
                 from src/PLH_web_handling.cpp:8:
.pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnectOTA.h:104:3: error: 'SPIFFST' does not name a type
   SPIFFST*  _fs;                /**< Filesystem for the native file uploading */
   ^
Compiling .pio/build/esp32doit-devkit-v1/lib509/WiFi/WiFiClient.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib509/WiFi/WiFiGeneric.cpp.o
....
In file included from .pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnect.h:39:0,
                 from include/Zmotion_blev_conn_OTA_EEv1.h:33,
                 from src/Zmotion_blev_conn_OTA_EEv1.cpp:36:
.pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnectOTA.h:32:13: error: 'SPIFFSFS' in namespace 'fs' does not name a type
 typedef fs::SPIFFSFS  SPIFFST;
             ^
In file included from .pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnect.h:39:0,
                 from include/Zmotion_blev_conn_OTA_EEv1.h:33,
                 from src/Zmotion_blev_conn_OTA_EEv1.cpp:36:
.pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnectOTA.h:104:3: error: 'SPIFFST' does not name a type
   SPIFFST*  _fs;                /**< Filesystem for the native file uploading */
   ^
Compiling .pio/build/esp32doit-devkit-v1/lib509/WiFi/WiFiMulti.cpp.o

I get the error twice during compilation.
I installed bot Autoconnect and Pagebuilder libraries and this is what I see lib_deps
lib_deps = 
	...
	hieromon/AutoConnect@^1.2.2
	hieromon/PageBuilder@^1.4.2
	
It looks as if I am missing some basic building block... but I cannot find what is missing.
I am not using SPIFFS anywhere in my code.

Can anybody give me some help, please?

Rgds.

Hieromon, 2021-01-04:
Could you post the `platformio.ini`?
If you want to share AutoConnect installed with Arduino IDE's Library Manager with PlatformIO, it needs a little work for `platformio.ini`.

sgrizzi, 2021-01-04:
> Reply to: Hieromon/AutoConnect <reply@reply.github.com>

Hi, this is the Platformio.ini file

; PlatformIO Project Configuration File

;

;   Build options: build flags, source filter

;   Upload options: custom upload port, speed and extra flags

;   Library options: dependencies, extra library storages

;   Advanced options: extra scripting

;

; Please visit documentation for the other options and examples

; https://docs.platformio.org/page/projectconf.html

 

[env:esp32doit-devkit-v1]

platform = espressif32

board = esp32doit-devkit-v1

framework = arduino

lib_extra_dirs = ~/Documents/Arduino/libraries

lib_deps = 

    sinricpro/SinricPro@^2.8.1

    nkolban/ESP32 BLE Arduino@^1.0.1

    hieromon/AutoConnect@^1.2.2

    hieromon/PageBuilder@^1.4.2

 

 

Fyi, the source code has been ported from Arduino.ino format to *.cpp + *.h files.

 

I also tried to import the original Arduino project directly into Platformio, including by default also the Arduino libraries, but when I compile I do get exactly the same error.

 

Rgds,

Hieromon, 2021-01-04:
What does the `Dependency Graph` at the build in PlatformIO look like when you remove `lib_deps` and add `lib_ldf_mode=chain+` instead?

sgrizzi, 2021-01-04:
Hi, with the following “new” .ini file:

[env:esp32doit-devkit-v1]

platform = espressif32

board = esp32doit-devkit-v1

framework = arduino

lib_extra_dirs = ~/Documents/Arduino/libraries

lib_ldf_mode=chain+

 

this is the dependency graph+output I get:

> Executing task: pio run <

 

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)

-------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html

PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)

PACKAGES: 

 - framework-arduinoespressif32 3.10004.201016 (1.0.4) 

 - tool-esptoolpy 1.30000.201119 (3.0.0) 

 - toolchain-xtensa32 2.50200.80 (5.2.0)

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF Modes: Finder ~ chain+, Compatibility ~ soft

Found 65 compatible libraries

Scanning dependencies...

Dependency Graph

|-- <SinricPro> 2.7.6

|   |-- <ArduinoJson> 6.17.2

|   |-- <WiFi> 1.0

|   |-- <WebSockets> 2.3.0

|-- <ArduinoJson> 6.17.2

|-- <WiFi> 1.0

|-- <WebSockets> 2.3.0

|-- <EEPROM> 1.0.3

[... 310 lines cut ...]
* Web  > https://platformio.org/lib/search?query=header:WiFi.h

*

**************************************************************

 

 #include <WiFi.h>

                  ^

compilation terminated.

*** [.pio/build/esp32doit-devkit-v1/lib544/WebSockets/WebSocketsServer.cpp.o] Error 1

================================================================ [FAILED] Took 11.39 seconds ================================================================

The terminal process "pio 'run'" terminated with exit code: 1.

 

Terminal will be reused by tasks, press any key to close it.

Date: Monday, 4 January 2021 at 22:31

Hieromon, 2021-01-04:
The version of AutoConnect recognized by PlatformIO is as follows.

`|-- <AutoConnect> 1.1.7`

I think there are multiple old versions of AutoConnect left in your environment.

sgrizzi, 2021-01-04:
I made a copy of the /Arduino/libraries folder. I had two “Autoconnect” in there and removed both of them.

Now the version recognized is Autoconnect 1.2.2…. but I get even more errors….

Can you still give me some hints?

Rgds, Gabriele

Compiling .pio/build/esp32doit-devkit-v1/src/Zmotion_blev_conn_OTA_EEv1.cpp.o

Compiling .pio/build/esp32doit-devkit-v1/libf6c/WiFi/WiFiUdp.cpp.o

Compiling .pio/build/esp32doit-devkit-v1/libf6c/WiFi/utility/server_drv.cpp.o

Compiling .pio/build/esp32doit-devkit-v1/libf6c/WiFi/utility/spi_drv.cpp.o

In file included from /Users/gabriele/Documents/Arduino/librariescopy/SinricPro/src/SinricPro.h:13:0,

                 from include/PLH_Alexa_T_H.h:10,

                 from src/PLH_Alexa_T_H.cpp:8:

/Users/gabriele/Documents/Arduino/librariescopy/SinricPro/src/SinricProWebsocket.h: In member function 'void websocketListener::setExtraHeaders()':

/Users/gabriele/Documents/Arduino/librariescopy/SinricPro/src/SinricProWebsocket.h:84:46: error: no matching function for call to 'WiFiClass::macAddress()'

          headers += "mac:" + WiFi.macAddress() + "\r\n";

                                              ^

In file included from /Users/gabriele/Documents/Arduino/librariescopy/SinricPro/src/SinricProWebsocket.h:16:0,

                 from /Users/gabriele/Documents/Arduino/librariescopy/SinricPro/src/SinricPro.h:13,

                 from include/PLH_Alexa_T_H.h:10,

                 from src/PLH_Alexa_T_H.cpp:8:

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/WiFi.h:137:14: note: candidate: uint8_t* WiFiClass::macAddress(uint8_t*)

     uint8_t* macAddress(uint8_t* mac);

              ^

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/WiFi.h:137:14: note:   candidate expects 1 argument, 0 provided

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/utility/spi_drv.cpp: In static member function 'static char SpiDrv::spiTransfer(char)':

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/utility/spi_drv.cpp:36:78: warning: integer overflow in expression [-Woverflow]

 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }

                                                                              ^

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/utility/spi_drv.cpp:37:26: note: in expansion of macro 'DELAY_SPI'

 #define DELAY_TRANSFER() DELAY_SPI(10)

                          ^

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/utility/spi_drv.cpp:76:5: note: in expansion of macro 'DELAY_TRANSFER'

     DELAY_TRANSFER();

     ^

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/utility/spi_drv.cpp: In static member function 'static void SpiDrv::getParam(uint8_t*)':

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/utility/spi_drv.cpp:36:78: warning: integer overflow in expression [-Woverflow]

[... 1870 lines cut ...]
     virtual void begin(uint16_t port=0) =0;

                  ^

In file included from .pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnect.h:26:0,

                 from include/Zmotion_blev_conn_OTA_EEv1.h:33,

                 from src/Zmotion_blev_conn_OTA_EEv1.cpp:36:

/Users/gabriele/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h: In member function 'virtual size_t WebServer::_currentClientWrite_P(const char*, size_t)':

/Users/gabriele/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:144:83: error: 'class WiFiClient' has no member named 'write_P'

   virtual size_t _currentClientWrite_P(PGM_P b, size_t l) { return _currentClient.write_P( b, l ); }

                                                                                   ^

In file included from .pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoConnect.h:39:0,

                 from include/Zmotion_blev_conn_OTA_EEv1.h:33,

                 from src/Zmotion_blev_conn_OTA_EEv1.cpp:36:

.pio/libdeps/esp32doit-devkit-v1/AutoConnect/src/AutoCon

sgrizzi, 2021-01-04:
Sorry I think I sent you a mixedup output file.

This should be correct.

 

I made a copy of the /Arduino/libraries folder. I had two “Autoconnect” in there and removed both of them.

Now the version recognized is Autoconnect 1.2.2…. but I get even more errors….

Can you still give me some hints?

Rgds, Gabriele

 

 

> Executing task: pio run <

 

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)

-------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html

PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)

PACKAGES: 

 - framework-arduinoespressif32 3.10004.201016 (1.0.4) 

 - tool-esptoolpy 1.30000.201119 (3.0.0) 

 - toolchain-xtensa32 2.50200.80 (5.2.0)

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF Modes: Finder ~ chain+, Compatibility ~ soft

Found 65 compatible libraries

Scanning dependencies...

Dependency Graph

|-- <SinricPro> 2.7.6

|   |-- <ArduinoJson> 6.17.2

|   |-- <WiFi> 1.2.7

|   |   |-- <SPI> 1.0

|   |-- <WebSockets> 2.3.0

|-- <ArduinoJson> 6.17.2

|-- <WiFi> 1.2.7

|   |-- <SPI> 1.0

[... 702 lines cut ...]
     return !memcmp(config.bssid, WiFi.BSSID(item), sizeof(station_config_t::bssid));

                                                 ^

In file included from src/Zmotion_blev_conn_OTA_EEv1.cpp:32:0:

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/WiFi.h:173:14: note:   initializing argument 1 of 'uint8_t* WiFiClass::BSSID(uint8_t*)'

     uint8_t* BSSID(uint8_t* bssid);

              ^

*** [.pio/build/esp32doit-devkit-v1/src/PLH_web_handling.cpp.o] Error 1

*** [.pio/build/esp32doit-devkit-v1/src/Zmotion_blev_conn_OTA_EEv1.cpp.o] Error 1

================================================================ [FAILED] Took 6.31 seconds ================================================================

The terminal process "pio 'run'" terminated with exit code: 1.

 

Terminal will be reused by tasks, press any key to close it.

Date: Monday, 4 January 2021 at 22:48

Hieromon, 2021-01-04:
Umm...Copying files is not enough for PIO. I thought you needed to refresh the local registry of the PIO library, is it wrong?

sgrizzi, 2021-01-04:
Unfortunately it does not work…
I restored the old .ini file and the connection to the Arduino/Libraries to avoid any problem with using an alternative directory.

Updated all the libraries, and still I get Autoconnect 1.1.7 in the graph.

If I load explicitly Autoconnect 1.2.2 then I see it recognized in the graph but still the errors remain… Now it looks like something wrong/missing with WiFi management…

[env:esp32doit-devkit-v1]

platform = espressif32

board = esp32doit-devkit-v1

framework = arduino

lib_extra_dirs = ~/Documents/Arduino/libraries

lib_ldf_mode = chain+

lib_deps = hieromon/AutoConnect@^1.2.2

Any additional idea?

Rgds, Gabriele

 

 

> Executing task: pio run <

 

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)

-------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html

PLATFORM: Espressif 32 (2.1.0) > DOIT ESP32 DEVKIT V1

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)

PACKAGES: 

 - framework-arduinoespressif32 3.10004.201016 (1.0.4) 

 - tool-esptoolpy 1.30000.201119 (3.0.0) 

 - toolchain-xtensa32 2.50200.80 (5.2.0)

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF Modes: Finder ~ chain+, Compatibility ~ soft

Found 68 compatible libraries

Scanning dependencies...

Dependency Graph

|-- <AutoConnect> 1.2.2

|   |-- <PageBuilder> 1.4.2

|   |   |-- <WiFi> 1.2.7
[... 769 lines cut ...]
     return !memcmp(config.bssid, WiFi.BSSID(item), sizeof(station_config_t::bssid));

                                                 ^

In file included from src/Zmotion_blev_conn_OTA_EEv1.cpp:32:0:

.pio/libdeps/esp32doit-devkit-v1/WiFi/src/WiFi.h:173:14: note:   initializing argument 1 of 'uint8_t* WiFiClass::BSSID(uint8_t*)'

     uint8_t* BSSID(uint8_t* bssid);

              ^

*** [.pio/build/esp32doit-devkit-v1/src/PLH_web_handling.cpp.o] Error 1

*** [.pio/build/esp32doit-devkit-v1/src/Zmotion_blev_conn_OTA_EEv1.cpp.o] Error 1

================================================================ [FAILED] Took 6.45 seconds ================================================================

The terminal process "pio 'run'" terminated with exit code: 1.

 

Terminal will be reused by tasks, press any key to close it.

Date: Monday, 4 January 2021 at 23:33

Hieromon, 2021-01-04:
Where does `<WiFi> 1.2.7` come from?, is it correct?
Here is the Dependency Graph of my environment as follows:

```
Dependency Graph
|-- <WiFi> 1.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <AutoConnect> 1.2.3
|   |-- <PageBuilder> 1.5.0
|   |   |-- <WiFi> 1.0
|   |   |-- <WebServer> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <FS> 1.0
|   |   |-- <SPIFFS> 1.0
|   |   |   |-- <FS> 1.0
|   |-- <ArduinoJson> 6.17.2
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <EEPROM> 1.0.3
|   |-- <FS> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <Preferences> 1.0
|   |-- <Ticker> 1.1
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <HTTPUpdate> 1.3
|   |   |-- <WiFi> 1.0
|   |   |-- <HTTPClient> 1.2
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |-- <Update> 1.0
|   |-- <SPI> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <EEPROM> 1.0.3
|-- <PageBuilder> 1.5.0
|   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|-- <FS> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <Preferences> 1.0
|-- <Ticker> 1.1
|-- <ArduinoJson> 6.17.2
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <HTTPUpdate> 1.3
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|-- <Update> 1.0
```

sgrizzi, 2021-01-04:
Where does <WiFi> 1.2.7 come from?, is it correct?

I have no idea – initially I had version 1.0, then, after all updates/refresh/etc. this new version popped up!
Tomorrow morning I will compare your tree with mine carefully and then get back to you and let you know if I managed to fix the problem.

What I already see is that you are using autoconnect 1.2.3 and my version is 1.2.2; same with PageBuilder yours 1.5.0 and mine is 1.4.2 – but probably this in not where the problem is.

Thanks for your support!

Gabriele

Date: Tuesday, 5 January 2021 at 00:41

Hieromon, 2021-01-05:
> What I already see is that you are using autoconnect 1.2.3 and my version is 1.2.2; same with PageBuilder yours 1.5.0 and mine is 1.4.2 – but probably this in not where the problem is.

Yes, the new version has no relation to WiFi. It improves memory consumption and will be released soon.

sgrizzi, 2021-01-08:
Probably the several addition/deletions of files and libraries had corrupted the IDE.
I re-started a project from scratch, adding the various .cpp and .h files, and I added manually all the necessary libraries, removing the link to all the former Arduino libraries collection.
In this way I managed to get the project up and running again.
Thanks for your support!

closed 2021-01-08 by sgrizzi