Hieromon/AutoConnect #292: ESP32 - TTGO T5 E-INK

Anton2k, 2021-01-04:
Having a weird issue where, if my ESP32 looses power, and I give it power again it wont connect to the saved network. If I turn it off and back on again it will connect. It almost as if it will only connect every second time if that makes sense? I should mention power is killed while in the loop, that that makes any kind of difference?

Thanks

Hieromon, 2021-01-04:
Can you provide logs with AC_DEBUG enabled?
Follow the documentation, see here and here.

Anton2k, 2021-01-04:
@Hieromon The link's you posted don't take me anywhere, can you add them? thanks

Hieromon, 2021-01-04:
> Follow the documentation, see here and here.

I forgot the link. The link path you showed is my intention.
In particular, I need to know what is happening in the following situations:

> if my ESP32 looses power, and I give it power again it wont connect to the saved network.

Anton2k, 2021-01-04:
@Hieromon here is the output after enabling debugging, not sure it help's you much though.

**1st attempt after flashing**
```
[AC] Element<caption> not registered
[AC] caption<9> of /timezone created
[AC] *noname placed on /timezone
[AC] caption<9> of /timezone loaded
[AC] Element<timezone> not registered
[AC] timezone<6> of /timezone created
[AC] *noname placed on /timezone
[AC] timezone<6> of /timezone loaded
[AC] Element<newline> not registered
[AC] newline<2> of /timezone created
[AC] *noname placed on /timezone
[AC] newline<2> of /timezone loaded
[AC] Element<start> not registered
[AC] start<8> of /timezone created
[AC] *noname placed on /timezone
[AC] start<8> of /timezone loaded
[AC] /timezone on hands
[AC] Current:
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin() failed
[AC] autoReconnect, 4 network(s) found, VM380XXXX(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(VM380XXXX,xxxxxxxxxxxx).....established IP:192.168.0.62
[AC] http server started
WiFi connected: 192.168.0.62
```

**Second attempt after switch the ESP32 off and on**
```
[AC] Element<caption> not registered
[AC] caption<9> of /timezone created
[AC] *noname placed on /timezone
[AC] caption<9> of /timezone loaded
[AC] Element<timezone> not registered
[AC] timezone<6> of /timezone created
[AC] *noname placed on /timezone
[AC] timezone<6> of /timezone loaded
[AC] Element<newline> not registered
[AC] newline<2> of /timezone created
[AC] *noname placed on /timezone
[AC] newline<2> of /timezone loaded
[AC] Element<start> not registered
[AC] start<8> of /timezone created
[AC] *noname placed on /timezone
[AC] start<8> of /timezone loaded
[AC] /timezone on hands
[AC] Current:VM380XXXX
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin().....................................................................................................time out IP:0.0.0.0
[AC] autoReconnect, 4 network(s) found, VM380XXXX(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(VM380XXXX,xxxxxxxxxxxx) failed
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] SoftAP esp32ap/12345678 Ch(1) IP:172.217.28.1 
[AC] http server started
[AC] DNS server started
```

**Third attempt after switching the ESP32 off and on again**
```
[AC] Element<caption> not registered
[AC] caption<9> of /timezone created
[AC] *noname placed on /timezone
[AC] caption<9> of /timezone loaded
[AC] Element<timezone> not registered
[AC] timezone<6> of /timezone created
[AC] *noname placed on /timezone
[AC] timezone<6> of /timezone loaded
[AC] Element<newline> not registered
[AC] newline<2> of /timezone created
[AC] *noname placed on /timezone
[AC] newline<2> of /timezone loaded
[AC] Element<start> not registered
[AC] start<8> of /timezone created
[AC] *noname placed on /timezone
[AC] start<8> of /timezone loaded
[AC] /timezone on hands
[AC] Current:
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin() failed
[AC] autoReconnect, 4 network(s) found, VM380XXXX(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(VM380XXXX,xxxxxxxxxxxx)....established IP:192.168.0.62
[AC] http server started
WiFi connected: 192.168.0.62
```

And you get the idea, the cycle just repeats connecting only every other time, hope it helps and thanks.

Hieromon, 2021-01-04:
Please post your specified AutoConnectConfig settings. I'll try to reproduction the test with my environement.

Anton2k, 2021-01-04:
@Hieromon I'm not sure where I find those, just using the defaults as far as I know and using the Simple example sketch.

```
/*
  Simple.ino, Example for the AutoConnect library.
  Copyright (c) 2018, Hieromon Ikasamo
  https://github.com/Hieromon/AutoConnect

  This software is released under the MIT License.
  https://opensource.org/licenses/MIT
*/

#if defined(ARDUINO_ARCH_ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#elif defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#include <WebServer.h>
#endif
#include <time.h>
#include <AutoConnect.h>

static const char AUX_TIMEZONE[] PROGMEM = R"(
{
  "title": "TimeZone",
  "uri": "/timezone",
  "menu": true,
  "element": [
    {
      "name": "caption",
      "type": "ACText",
      "value": "Sets the time zone to get the current local time.",
      "style": "font-family:Arial;font-weight:bold;text-align:center;margin-bottom:10px;color:DarkSlateBlue"
    },
    {
      "name": "timezone",
      "type": "ACSelect",
      "label": "Select TZ name",
      "option": [],
      "selected": 10
    },
    {
      "name": "newline",
      "type": "ACElement",
      "value": "<br>"
    },
    {
      "name": "start",
      "type": "ACSubmit",
      "value": "OK",
      "uri": "/start"
    }
  ]
}
)";

typedef struct {
  const char* zone;
  const char* ntpServer;
  int8_t      tzoff;
} Timezone_t;

static const Timezone_t TZ[] = {
  { "Europe/London", "europe.pool.ntp.org", 0 },
  { "Europe/Berlin", "europe.pool.ntp.org", 1 },
  { "Europe/Helsinki", "europe.pool.ntp.org", 2 },
  { "Europe/Moscow", "europe.pool.ntp.org", 3 },
  { "Asia/Dubai", "asia.pool.ntp.org", 4 },
  { "Asia/Karachi", "asia.pool.ntp.org", 5 },
  { "Asia/Dhaka", "asia.pool.ntp.org", 6 },
  { "Asia/Jakarta", "asia.pool.ntp.org", 7 },
  { "Asia/Manila", "asia.pool.ntp.org", 8 },
  { "Asia/Tokyo", "asia.pool.ntp.org", 9 },
  { "Australia/Brisbane", "oceania.pool.ntp.org", 10 },
  { "Pacific/Noumea", "oceania.pool.ntp.org", 11 },
  { "Pacific/Auckland", "oceania.pool.ntp.org", 12 },
  { "Atlantic/Azores", "europe.pool.ntp.org", -1 },
  { "America/Noronha", "south-america.pool.ntp.org", -2 },
  { "America/Araguaina", "south-america.pool.ntp.org", -3 },
  { "America/Blanc-Sablon", "north-america.pool.ntp.org", -4},
  { "America/New_York", "north-america.pool.ntp.org", -5 },
  { "America/Chicago", "north-america.pool.ntp.org", -6 },
  { "America/Denver", "north-america.pool.ntp.org", -7 },
  { "America/Los_Angeles", "north-america.pool.ntp.org", -8 },
  { "America/Anchorage", "north-america.pool.ntp.org", -9 },
  { "Pacific/Honolulu", "north-america.pool.ntp.org", -10 },
  { "Pacific/Samoa", "oceania.pool.ntp.org", -11 }
};

#if defined(ARDUINO_ARCH_ESP8266)
ESP8266WebServer Server;
#elif defined(ARDUINO_ARCH_ESP32)
WebServer Server;
#endif

AutoConnect       Portal(Server);
AutoConnectConfig Config;       // Enable autoReconnect supported on v0.9.4
AutoConnectAux    Timezone;

void rootPage() {
  String  content =
    "<html>"
    "<head>"
    "<meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">"
    "<script type=\"text/javascript\">"
    "setTimeout(\"location.reload()\", 1000);"
    "</script>"
    "</head>"
    "<body>"
    "<h2 align=\"center\" style=\"color:blue;margin:20px;\">Hello, world</h2>"
    "<h3 align=\"center\" style=\"color:gray;margin:10px;\">{{DateTime}}</h3>"
    "<p style=\"text-align:center;\">Reload the page to update the time.</p>"
    "<p></p><p style=\"padding-top:15px;text-align:center\">" AUTOCONNECT_LINK(COG_24) "</p>"
    "</body>"
    "</html>";
  static const char *wd[7] = { "Sun","Mon","Tue","Wed","Thr","Fri","Sat" };
  struct tm *tm;
  time_t  t;
  char    dateTime[26];

  t = time(NULL);
  tm = localtime(&t);
  sprintf(dateTime, "%04d/%02d/%02d(%s) %02d:%02d:%02d.",
    tm->tm_year + 1900, tm->tm_mon + 1, tm->tm_mday,
    wd[tm->tm_wday],
    tm->tm_hour, tm->tm_min, tm->tm_sec);
  content.replace("{{DateTime}}", String(dateTime));
  Server.send(200, "text/html", content);
}

void startPage() {
  // Retrieve the value of AutoConnectElement with arg function of WebServer class.
  // Values are accessible with the element name.
  String  tz = Server.arg("timezone");

  for (uint8_t n = 0; n < sizeof(TZ) / sizeof(Timezone_t); n++) {
    String  tzName = String(TZ[n].zone);
    if (tz.equalsIgnoreCase(tzName)) {
      configTime(TZ[n].tzoff * 3600, 0, TZ[n].ntpServer);
      Serial.println("Time zone: " + tz);
      Serial.println("ntp server: " + String(TZ[n].ntpServer));
      break;
    }
  }

  // The /start page just constitutes timezone,
  // it redirects to the root page without the content response.
  Server.sendHeader("Location", String("http://") + Server.client().localIP().toString() + String("/"));
  Server.send(302, "text/plain", "");
  Server.client().flush();
  Server.client().stop();
}

void setup() {
  delay(5000);
  Serial.begin(115200);
  Serial.println("test setup");

  // Enable saved past credential by autoReconnect option,
  // even once it is disconnected.
  Config.autoReconnect = true;
  Portal.config(Config);

  // Load aux. page
  Timezone.load(AUX_TIMEZONE);
  // Retrieve the select element that holds the time zone code and
  // register the zone mnemonic in advance.
  AutoConnectSelect&  tz = Timezone["timezone"].as<AutoConnectSelect>();
  for (uint8_t n = 0; n < sizeof(TZ) / sizeof(Timezone_t); n++) {
    tz.add(String(TZ[n].zone));
  }

  Portal.join({ Timezone });        // Register aux. page

  // Behavior a root path of ESP8266WebServer.
  Server.on("/", rootPage);
  Server.on("/start", startPage);   // Set NTP server trigger handler

  // Establish a connection with an autoReconnect option.
  delay(5000);

  if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  }
}

void loop() {
  Serial.println("test loop 1");
  Portal.handleClient();
  Serial.println("test loop 2");
}
```

Hieromon, 2021-01-04:
In your environment, WiFi.begin by esp-idf is failing in the first place.
AutoConnect does not specify an argument in WiFi.begin if esp-idf already holds the current connection. Restoring the connection, in that case, is purely the role of esp-idf.

```
[AC] Current:VM380XXXX
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin().....................................................................................................time out IP:0.0.0.0
```

This is the result of reproducing your sketch in my module and environment. This is the connection status when the power is turned off and then turned on again after Config New.
As you can see, WiFi.begin without SSID parameter is regaining connectivity by esp-idf work itself.

```
[AC] Current:SPXAA2831416
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()......established IP:192.168.1.12
[AC] http server started
WiFi connected: 192.168.1.12
```
The cause seems to be somewhere other than AutoConnect.

Anton2k, 2021-01-05:
@Hieromon Is there a deeper level of debugging I can do on the ESP32 it's self? thanks

Anton2k, 2021-01-05:
@Hieromon Also here is the verbose output

```
[AC] Element<caption> not registered
[AC] caption<9> of /timezone created
[AC] *noname placed on /timezone
[AC] caption<9> of /timezone loaded
[AC] Element<timezone> not registered
[AC] timezone<6> of /timezone created
[AC] *noname placed on /timezone
[AC] timezone<6> of /timezone loaded
[AC] Element<newline> not registered
[AC] newline<2> of /timezone created
[AC] *noname placed on /timezone
[AC] newline<2> of /timezone loaded
[AC] Element<start> not registered
[AC] start<8> of /timezone created
[AC] *noname placed on /timezone
[AC] start<8> of /timezone loaded
[AC] /timezone on hands
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[AC] Current:VM380XXXX
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin().[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
....................................................................................................time out IP:0.0.0.0
[AC] autoReconnect[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
, 4 network(s) found, VM380XXXX(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(VM380XXXX,xxxxxxxxxxxx) failed
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] SoftAP esp32ap/12345678 Ch(1) IP:172.217.28.1 
[AC] http server started
[AC] DNS server started

```

Hieromon, 2021-01-05:
> I want to manually disconnect from the WiFi before powering off the ESP32, how would I do this in the Simple sketch,

You can use the Disconnect using AutoConnect menu navigation with the source code modification as below:

https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnect.cpp#L593

Change to:

```cpp
disconnectWiFi(true);
```

In addition, if you want to keep the connection AP data stored by esp-idf, you also need to modify the following  (AutoConnect's Disconnect menu erases the connected AP data stored in esp-idf by default):

https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnect.cpp#L1504

Change to:

```cpp
WiFi.disconnect(wifiOff, false);
```

Hieromon, 2021-01-05:
> `[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL`

The access point refused to connect caused the password is wrong... It is a phenomenon that the password requirement does not match between your AP and esp-idf, but it is impossible if we think normally.

Anton2k, 2021-01-05:
@Hieromon I have changed the 2 lines you mentioned above, but not sure how to call Disconnect in my sketch. would it just be

`Disconnect();`

Sorry still trying to figure this all out.

Hieromon, 2021-01-05:
No need to sketch. You can disconnect at any time by using Disconnect in the AutoConnect menu. Also the above fix will issue `WiFi.disconnect(true)`. 
Or, you want to be hard-coding in Sketch?

Anton2k, 2021-01-05:
> No need to sketch. You can disconnect at any time by using Disconnect in the AutoConnect menu. Also the above fix will issue `WiFi.disconnect(true)`.
> Or, you want to be hard-coding in Sketch?

Yes hard coded in sketch, my project will only wake for a minute or 2 every 24hrs to get some data.

So I added `WiFi.disconnect(true).` to the sketch as well as the 2 code changes you mentioned and it now connects every time! :) yay! So my theory is that, when I kill the power to the ESP32 while it is still connected to my wifi network, the connection is not killed cleanly, hence why I am getting these AUTH issues, or the ESP32 is retaining some kind of info somewhere which it does not let go of until a second reboot.

Hieromon, 2021-01-05:
> the ESP32 is retaining some kind of info somewhere which it does not let go of until a second reboot.

That's the specification of the ESP32 arduino core. WiFiSTAClass uses it for WiFi.reconnect () and WiFi.begin with no arguments. However, I don't think it has anything to do with this phenomenon. WiFi.begin with no arguments restores that information and attempts to connect to the AP.

> I kill the power to the ESP32 while it is still connected to my wifi network, the connection is not killed cleanly, hence why I am getting these AUTH issues

What is the reason for this? Is it the characteristics of a particular access point product?
I think the Auth error is rather due to the response on the AP side.

Anton2k, 2021-01-05:
> That's the specification of the ESP32 arduino core. WiFiSTAClass uses it for WiFi.reconnect () and WiFi.begin with no arguments. However, I don't think it has anything to do with this phenomenon. [...]

The project I am working on needs to use little to no power, the idea is the ESP32 connects to the internet, gets some data then powers off, I will use an RTC to power it back on. Meaning it will be drawing 0 power when it is not in use. I think the AP may have something to do with it, I'll need to test it out on my phone's hotspot or another router so see what results that yields. I'LL do some more testing tomorrow and get back to you on this, at the moment I think this work around should work well for my purposes.

Hieromon, 2021-01-05:
I understand, let wait for the results from you.
Maybe I need to add an optional setting for `WiFi.disconnet(true)` to the AutoConnect library.

Anton2k, 2021-01-05:
@Hieromon So I have done some testing today both on my router, another router and my phones hot spot. All return the exact same behaviour, I can only put it down to the ESP32 at this stage and when I kill the power the WiFi connection is not killed cleanly (whatever that means), all I know is that using `WiFi.disconnect();` allows me to kill the power and re-connect every time. For my project this is fine, the esp32 fires up, does some stuff then disconnects from the WiFi, but there is still a possibility that the power is killed to the device (as it will be battery powered) and if that happens it means having to to restart the ESP32 again on the first time it receives power.

Here is my thinking, store a variable in the flash memory of the ESP32, it stores only 1 of 2 values at and time (lets say the values are 0 and 1). 

- If the ESP32 boots and and connects to the WiFi successfully it changes the variable from 0 to 1.
- If the ESP32 looses power after connecting to the WiFi (my problem) then the variable will contain 1, if so ` ESP32.restart();`
- The restart should resolve any `202 - AUTH_FAIL` issues or whatever.

Don't know if my idea is any good or not, but `ESP.restart();` also seems to clean up anything the the power outage might have left behind.

Anton2k, 2021-01-06:
@Hieromon So another update... after a bit more digging I have discovered that putting in a delay at the very top of my setup() causes the ESP32 to auto connect each time, even if the power is killed after it has established a connection. I have managed to get the delay down to 35 seconds, any less than that at I get issues again.

So couple of things, WIFI. Disconnect(); stops the issues and so does adding in a 35 second delay before connecting to any wifi, this tells me that something somewhere is holding on to something that is preventing from connecting to the network, if I disconnect cleanly what ever is being held on to is released, if I dont it gets held, the delay allows what ever is being hung on to; to be reset or released, no idea if this could be an IP or something router side or something within the ESP32 it's self, there is a guy here https://github.com/espressif/arduino-esp32/issues/653#issuecomment-572387806 talking about the DHCP and stuff, I'm just now sure though. For the time being this delay will do!

Hieromon, 2021-01-06:
@Anton2k Thank you for the detailed survey results. I followed your guidance and traced the implementation of the arduino-esp32.

What the Arduino-esp32 core does with the `WiFi.disconnect()` with no arguments call is simply to issue an `esp_wifi_disconnect`.
If `esp_wifi_disconnect()` is sending a WIFI station-leave message to the AP, the AP will have the opportunity to free up resources for the station at that point. Also, if the 35 seconds wait is a timeout for the WiFi beacon, the AP detects the station has left, the AP can free up resources for the station at that point. Those things suggest that the AP will have the opportunity to clear the mismatch in the credentials exchanged between the AP and the WiFi device (ie. the AP will start negotiating again from the beginning). These are just my imaginations, but they are consistent with the workaround scenario you supplemented.

Even `autoReconnect` fails in the log of the second attempts you provided. And `autoReconnect` does not issue `WiFi.disconnect()` after the `1st-WiFi.begin` failure. This confirms the solution you have come up with.

```
[AC] Current:VM380XXXX
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin().....................................................................................................time out IP:0.0.0.0
[AC] autoReconnect, 4 network(s) found, VM380XXXX(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(VM380XXXX,xxxxxxxxxxxx) failed
```

I encountered this phenomenon for the first time. 
Since posting this issue, I've done ESP Power ON / OFF tests with different APs, but I still can't reproduce the issue. The ESP module restores connectivity with the AP with no delay.

I think there is something left on the AP side related to authentication. It's something other than an SSID and password. And that's something that is ignored (or adapted) if the negotiation is done from the beginning.
Country code? , I'm not sure. As far as I know, the country code is not often used for authentication, but it seems to be used by some routers.
(AutoConnect does not set the country code)

Anton2k, 2021-01-07:
@Hieromon At this point I am pretty sure it's a router side thing and it relates to DHCP. I think the ESP32 always tries to connect with the same IP to the access point. If the disconnect and reconnect happens quickly, depending on the router and manufacture, the IP will not be released in time from the router so that the ESP32 can connect.

After having a look in AutoConnectDefs.h I cam across this...

`// Time-out limitation when AutoConnect::begin [ms]
#ifndef AUTOCONNECT_TIMEOUT
#define AUTOCONNECT_TIMEOUT     30000
#endif // !AUTOCONNECT_TIMEOUT`

So by default the timeout is set to 30 seconds, which I can confirm does not work for me, I need at least 35 seconds. I think there should be a way to set the timeout via config, this would allow us to account for different routers. There is a chance that some other routers could be 60 seconds or more before release the IP from the DHCP.

Hieromon, 2021-01-07:
@Anton2k Thank you.
It is probably correct that the cause of the reconnection failure is the delay in releasing resources by a particular router. However, I was really weird about the `AUTH_FAIL` appeared in your log, so I proceeded with the investigation from a different perspective.

I also had read the topic you showed. It doesn't explain everything. And the workaround presented there is essentially the same as the method AutoConnect has already implemented as an autoReconnect option. The problem we encountered this time is even deeper.
Because even autoReconnect can't get teeth into this problem.

So, the investigation found important discoveries on topics similar to this issue. Its conversation in this thread is very long, suggesting that many contributors have been working on this issue for over a year. And I was able to have a scenario of this phenomenon from the following:

https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-705575707

[image]

> What I can see in wireshark is clearly that after wakeup from deepsleep with some routers (in my case a Fritz7362) the ESP32 does **NOT** send the Keys for encryp. I have another router where I have no problem at all with wifi and deep sleep it works any time.

The WiFi frame capture log by Wireshark is a fact. If that is the cause, there is also a reason for `AUTH_FAIL` in there.

Here, let's explore the logs we encountered again is the following:

```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[AC] Current:VM380XXXX
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin().[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
....................................................................................................time out IP:0.0.0.0
[AC] autoReconnect[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
, 4 network(s) found, VM380XXXX(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(VM380XXXX,xxxxxxxxxxxx) failed
```

The router has rejected the connection request from the second turned-on ESP32 that does not send the encryption key. It happens on the 1st-WiFi.begin() and the result is `AUTH_FAIL`.
Then, the autoReconnect obtains the SSID and password from the AutoConnectCredential to try to recover it. And WiFi.begin with SSID & PSK fails again. Perhaps it's a DHCP client start failure inside the `begin` method of the `WiFiSTA` class.

On the other hand, the AP is retaining the ESP32's MAC address and its assigned IP address. It is the same as the MAC address included in the DHCP-discover sent by the second WiFi begin. The AP will not assign a new IP and will not send a DHCP-offer to respond to the sent DHCP-discover. Then the EP32 DHCP client initiation fails.
And after a certain period of time, the AP will assume that the WiFi station has already left and will release the DHCP resources attached to the previous connection of the same IP.

Then, it will explain all the phenomena we encountered.

In any case, the current solution is, as you've found, a "delay method". It suggests a similar workaround in the topic above (for now). But I'm very sorry to disappoint you, I don't think extending `AUTOCONNECT_TIMEOUT` would not be a solution. (even if it's ad hoc)

> I think there should be a way to set the timeout via config, 

AutoConnect is already equipped with the setting. Refer to the [documentation](https://hieromon.github.io/AutoConnect/adconnection.html#timeout-settings-for-a-connection-attempt) and you can try it.

It is not enough that only extending the timeout for the 1st-WiFi.begin(). The proven workaround has appeared many times in the topics mentioned above and is called a "double-hitter".
I will start considering incorporating this method into AutoConnect and believe it is possible. In fact, the autoReconnect is close to a double-hitter. Just put a 30 seconds delay between the first and second WiFi.begin. The question is, how does AutoConnect properly determine when should insert delay?. It is not good even if made to appear the double hitter in all of the scenes.

Also, I think the contributors are getting closer to the heart of the matter.

Here,
https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-708704583

and Here.
https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-709641837

And Thank you again for your very committed cooperation. I'm glad you used AutoConnect.
I will release the patch for the workaround in AutoConnect as soon as possible, maybe in a week. I would like to have you test it. (Because the problem cannot be reproduced in my environment)

Anton2k, 2021-01-07:
@Hieromon Thanks again for trying to help me resolve this issues, I can understand some of above but not everything, I will need to read over it again and take time to digest it, but what you have said makes sense.

I will be more than happen to test any solution that you come up with and will keep a close eye on those other issues on the espressif github, thanks again and I look forward to hearing from you.

ありがとうございました

Hieromon, 2021-01-08:
@Anton2k Hello,
If you can, could you test the verification code below in your environment?
This is a candidate of the skeleton for a workaround based on the double-hitter strategy, which is considered to avoid waiting too long compared to the original double-hitter and to ensure backward compatibility with the current AutoConnect::begin behavior assisted with AutoConnectConfig::autoReconnect.

- Clear the ESP32 FALSH once before uploading the sketch. This will erase the past connection data saved by the ESP32.
- Please specify `Verbose` with `Core Debug Level` in Arduino IDE tool menu and get the log, and post to here. (`build_flags=-DCORE_DEBUG_LEVEL=5` for the `platformio.ini`)
- Repeat turning the power on and off in the same way as last time to see if this can be connected.

I hope this skeleton works, but it has no guarantee.  Even if this workaround goes wrong, I would like to consider the next best.

Thank you for your contribution.

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <esp32-hal-log.h>

// Change to suit your WiFi router
#define WIFI_SSID     "SSID_YOU_WISH"
#define WIFI_PASSWORD "PASSWORD_YOU_WISH"

void setup() {
  wl_status_t wst;
  unsigned long tm;

  delay(1000);
  Serial.begin(115200);
  Serial.println();

  esp_log_level_set("wifi", ESP_LOG_VERBOSE);
  esp_log_level_set("dhcpc", ESP_LOG_VERBOSE);

  WiFi.mode(WIFI_STA);
  Serial.print("connecting");
  wst = WiFi.begin();
  tm = millis();
  if (wst == WL_DISCONNECTED) {
    Serial.print(" - disconnected, wait for failure");
    while (millis() - tm < 35000) {
      wst = WiFi.status();
      if (wst == WL_CONNECT_FAILED || wst == WL_CONNECTED)
        break;
      delay(300);
      Serial.print('_');
    }
  }
  if (wst == WL_CONNECT_FAILED) {
    wst = WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.printf("\nAttempts reconnect(%d)", (int)wst);
  }
  while (millis() - tm < 30000) {
    if ((wst = WiFi.status()) == WL_CONNECTED)
      break;
    delay(300);
    Serial.print('.');
  }
  if (wst == WL_CONNECTED)
    Serial.printf("conencted IP:%s\n", WiFi.localIP().toString().c_str());
  else
    Serial.printf("failed:%d\n", (int)WiFi.status());
  Serial.println("setup processed end");
}

void loop() {
}
```

Anton2k, 2021-01-09:
@Hieromon This is what I get each time.

```
I (1037) wifi: wifi driver task: 3ffb57c4, prio:23, stack:3584, core=0
I (2109) wifi: wifi firmware version: 7997e4b
I (2110) wifi: config NVS flash: enabled
I (2110) wifi: config nano formating: disabled
I (2110) wifi: Init dynamic tx buffer num: 32
I (2114) wifi: Init data frame dynamic rx buffer num: 32
I (2119) wifi: Init management frame dynamic rx buffer num: 32
I (2125) wifi: Init management short buffer num: 32
I (2130) wifi: Init static rx buffer size: 1600
I (2134) wifi: Init static rx buffer num: 16
I (2138) wifi: Init dynamic rx buffer num: 32
I (2206) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
connecting - disconnected, wait for failure______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_______[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
_failed:1
setup processed end
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[... cut ...]
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
```

Hieromon, 2021-01-09:
@Anton2k Did you change SSID and password to suit your router?

```cpp
// Change to suit your WiFi router
#define WIFI_SSID     "SSID_YOU_WISH"
#define WIFI_PASSWORD "PASSWORD_YOU_WISH"
```

Anton2k, 2021-01-09:
@Hieromon Yea I have updated the details, here is how my sketch looks, this time it happens every time I boot the ESP32.

```
#include <Arduino.h>
#include <WiFi.h>
#include <esp32-hal-log.h>

// Change to suit your WiFi router
#define WIFI_SSID     "VM380XXXX"
#define WIFI_PASSWORD "xxxxxxxxxxxx"

void setup() {
  wl_status_t wst;
  unsigned long tm;

  delay(1000);
  Serial.begin(115200);
  Serial.println();

  esp_log_level_set("wifi", ESP_LOG_VERBOSE);
  esp_log_level_set("dhcpc", ESP_LOG_VERBOSE);

  WiFi.mode(WIFI_STA);
  Serial.print("connecting");
  wst = WiFi.begin();
  tm = millis();
  if (wst == WL_DISCONNECTED) {
    Serial.print(" - disconnected, wait for failure");
    while (millis() - tm < 35000) {
      wst = WiFi.status();
      if (wst == WL_CONNECT_FAILED || wst == WL_CONNECTED)
        break;
      delay(300);
      Serial.print('_');
    }
  }
  if (wst == WL_CONNECT_FAILED) {
    wst = WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.printf("\nAttempts reconnect(%d)", (int)wst);
  }
  while (millis() - tm < 30000) {
    if ((wst = WiFi.status()) == WL_CONNECTED)
      break;
    delay(300);
    Serial.print('.');
  }
  if (wst == WL_CONNECTED)
    Serial.printf("conencted IP:%s\n", WiFi.localIP().toString().c_str());
  else
    Serial.printf("failed:%d\n", (int)WiFi.status());
  Serial.println("setup processed end");
}

void loop() {
}
```

Hieromon, 2021-01-09:
@Anton2k, Ok. As I thought, the problem seems to be on the router. AUTH_FAIL may occur with or without parameters in WiFi.begin after power off without WiFi.disconnect and then power on.
Try this one. (Need to erase FLASH once)

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <esp32-hal-log.h>

// Change to suit your WiFi router
#define WIFI_SSID     "SPXAA2831416"
#define WIFI_PASSWORD "ae38546b444a55b"

void setup() {
  wl_status_t wst;
  unsigned long tm;

  delay(3000);
  Serial.begin(115200);
  Serial.println();

  esp_log_level_set("wifi", ESP_LOG_VERBOSE);
  esp_log_level_set("dhcpc", ESP_LOG_VERBOSE);

  WiFi.mode(WIFI_STA);
  Serial.print("connecting");
  wst = WiFi.begin();
  tm = millis();
  if (wst == WL_DISCONNECTED) {
    Serial.print(" - disconnected, wait for failure");
    while (millis() - tm < 30000) {
      wst = WiFi.status();
      if (wst == WL_CONNECT_FAILED || wst == WL_CONNECTED)
        break;
      delay(300);
      Serial.print('_');
    }
  }
  if (wst == WL_CONNECT_FAILED) {

    // Disconnection and delay encourages the router to release resources.
    // Please increase the delay time depending on your situation.
    WiFi.disconnect(true, false);
    delay(500);

    wst = WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.printf("\nAttempts reconnect(%d)", (int)wst);
  }
  while (millis() - tm < 30000) {
    if ((wst = WiFi.status()) == WL_CONNECTED)
      break;
    delay(300);
    Serial.print('.');
  }
  if (wst == WL_CONNECTED)
    Serial.printf("conencted IP:%s\n", WiFi.localIP().toString().c_str());
  else
    Serial.printf("failed:%d\n", (int)WiFi.status());
  Serial.println("setup processed end");
}

void loop() {
}
```

Anton2k, 2021-01-09:
@Hieromon With the new code and efter flash erase,

1st boot
```
I (3055) wifi: wifi driver task: 3ffb5880, prio:23, stack:3584, core=0
I (6155) wifi: wifi firmware version: 7997e4b
I (6155) wifi: config NVS flash: enabled
I (6155) wifi: config nano formating: disabled
I (6156) wifi: Init dynamic tx buffer num: 32
I (6160) wifi: Init data frame dynamic rx buffer num: 32
I (6165) wifi: Init management frame dynamic rx buffer num: 32
I (6170) wifi: Init management short buffer num: 32
I (6175) wifi: Init static rx buffer size: 1600
I (6180) wifi: Init static rx buffer num: 16
I (6183) wifi: Init dynamic rx buffer num: 32
I (6355) wifi: mode : softAP (24:6f:28:7a:5f:89)
I (6356) wifi: Total power save buffer number: 16
I (6356) wifi: Init max length of beacon: 752/752
I (6358) wifi: Init max length of beacon: 752/752
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
I (6374) wifi: mode : sta (24:6f:28:7a:5f:88)
connecting[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[E][WiFiSTA.cpp:219] begin(): connect failed!
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
I (6396) wifi: flush txq
I (6398) wifi: stop sw txq
I (6401) wifi: lmac stop hw txq
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
I (6911) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

Attempts reconnect(6)..I (7647) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
..I (8382) wifi: state: init -> auth (b0)
I (8385) wifi: state: auth -> init (8a0)
I (8385) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
...............................................................................................failed:4
setup processed end
```

2nd boot after killing power
```
I (3043) wifi: wifi driver task: 3ffb57c4, prio:23, stack:3584, core=0
I (6121) wifi: wifi firmware version: 7997e4b
I (6121) wifi: config NVS flash: enabled
I (6121) wifi: config nano formating: disabled
I (6122) wifi: Init dynamic tx buffer num: 32
I (6126) wifi: Init data frame dynamic rx buffer num: 32
I (6131) wifi: Init management frame dynamic rx buffer num: 32
I (6137) wifi: Init management short buffer num: 32
I (6142) wifi: Init static rx buffer size: 1600
I (6146) wifi: Init static rx buffer num: 16
I (6150) wifi: Init dynamic rx buffer num: 32
I (6234) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
connecting - disconnected, wait for failure__I (6974) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
__I (7709) wifi: state: init -> auth (b0)
I (7713) wifi: state: auth -> assoc (0)
I (7718) wifi: state: assoc -> run (10)
_I (7751) wifi: connected with VM3807585, channel 6, BW20
I (7755) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
_[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.0.62, MASK: 255.255.255.0, GW: 192.168.0.1
_conencted IP:192.168.0.62
[... 65 lines cut ...]
I (7678) wifi: state: auth -> init (8a0)
I (7678) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
_I (7717) wifi: flush txq
I (7717) wifi: stop sw txq
I (7718) wifi: lmac stop hw txq
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
I (8225) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

Attempts reconnect(6)I (8419) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
..I (9154) wifi: state: init -> auth (b0)
I (9160) wifi: state: auth -> assoc (0)
I (9165) wifi: state: assoc -> run (10)
.I (9198) wifi: connected with VM3807585, channel 6, BW20
I (9202) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.0.62, MASK: 255.255.255.0, GW: 192.168.0.1
.conencted IP:192.168.0.62
setup processed end
```

Hieromon, 2021-01-10:
@Anton2k It's a pretty tough opponent.
So, did you evaluate with a variety of delay times?

```cpp
// Disconnection and delay encourages the router to release resources.
// Please increase the delay time depending on your situation.
WiFi.disconnect(true, false);
delay(500);
```
Also, please verify how it behaves when you change the **second parameter** of `WiFi.disconnect` as above to `true`.

Anton2k, 2021-01-10:
@Hieromon Using
```
WiFi.disconnect(true);
delay(500);
```
And clearing the ESP32 flash here are the results.

1. Directly after flashing sketch (no power off or on just a "Hard resetting via RTS pin" that arduino IDE performs after uploading the sketch).
```
I (3055) wifi: wifi driver task: 3ffb5880, prio:23, stack:3584, core=0
I (6155) wifi: wifi firmware version: 7997e4b
I (6155) wifi: config NVS flash: enabled
I (6155) wifi: config nano formating: disabled
I (6156) wifi: Init dynamic tx buffer num: 32
I (6160) wifi: Init data frame dynamic rx buffer num: 32
I (6165) wifi: Init management frame dynamic rx buffer num: 32
I (6170) wifi: Init management short buffer num: 32
I (6175) wifi: Init static rx buffer size: 1600
I (6180) wifi: Init static rx buffer num: 16
I (6183) wifi: Init dynamic rx buffer num: 32
I (6348) wifi: mode : softAP (24:6f:28:7a:5f:89)
I (6349) wifi: Total power save buffer number: 16
I (6349) wifi: Init max length of beacon: 752/752
I (6350) wifi: Init max length of beacon: 752/752
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
I (6366) wifi: mode : sta (24:6f:28:7a:5f:88)
connecting[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[E][WiFiSTA.cpp:219] begin(): connect failed!
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
I (6389) wifi: flush txq
I (6391) wifi: stop sw txq
I (6394) wifi: lmac stop hw txq
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
I (6903) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

Attempts reconnect(6)..I (7639) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
..I (8374) wifi: state: init -> auth (b0)
I (8377) wifi: state: auth -> init (8a0)
I (8377) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
...............................................................................................failed:4
setup processed end
```

2. Switching the power to the ESP32 off then on
```
I (3042) wifi: wifi driver task: 3ffb57c4, prio:23, stack:3584, core=0
I (6119) wifi: wifi firmware version: 7997e4b
I (6119) wifi: config NVS flash: enabled
I (6119) wifi: config nano formating: disabled
I (6120) wifi: Init dynamic tx buffer num: 32
I (6124) wifi: Init data frame dynamic rx buffer num: 32
I (6129) wifi: Init management frame dynamic rx buffer num: 32
I (6135) wifi: Init management short buffer num: 32
I (6140) wifi: Init static rx buffer size: 1600
I (6144) wifi: Init static rx buffer num: 16
I (6148) wifi: Init dynamic rx buffer num: 32
I (6216) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
connecting - disconnected, wait for failure__I (6956) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
__I (7691) wifi: state: init -> auth (b0)
I (7695) wifi: state: auth -> assoc (0)
I (7700) wifi: state: assoc -> run (10)
I (7733) wifi: connected with VM38XXXX, channel 6, BW20
_I (7737) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.0.62, MASK: 255.255.255.0, GW: 192.168.0.1
_conencted IP:192.168.0.62
setup processed end
```

3. Power ESP32 off and on again for a second time, same results
```
I (3045) wifi: wifi driver task: 3ffb57c4, prio:23, stack:3584, core=0
I (6127) wifi: wifi firmware version: 7997e4b
I (6127) wifi: config NVS flash: enabled
I (6127) wifi: config nano formating: disabled
I (6128) wifi: Init dynamic tx buffer num: 32
I (6132) wifi: Init data frame dynamic rx buffer num: 32
I (6137) wifi: Init management frame dynamic rx buffer num: 32
I (6142) wifi: Init management short buffer num: 32
I (6147) wifi: Init static rx buffer size: 1600
I (6151) wifi: Init static rx buffer num: 16
I (6155) wifi: Init dynamic rx buffer num: 32
I (6224) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
connecting - disconnected, wait for failure__I (6964) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (6964) wifi: state: init -> auth (b0)
I (6967) wifi: state: auth -> init (8a0)
I (6967) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
_I (7141) wifi: flush txq
I (7141) wifi: stop sw txq
I (7142) wifi: lmac stop hw txq
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
I (7649) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

Attempts reconnect(6)I (7786) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
..I (8522) wifi: state: init -> auth (b0)
I (8525) wifi: state: auth -> assoc (0)
I (8531) wifi: state: assoc -> run (10)
I (8563) wifi: connected with VM380XXXX, channel 6, BW20
.I (8625) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.0.62, MASK: 255.255.255.0, GW: 192.168.0.1
.conencted IP:192.168.0.62
setup processed end
```

Anton2k, 2021-01-10:
@Hieromon Using

```
WiFi.disconnect(true, false);
delay(6500);
```

6500 is the lowest I can go, I tried 6400 and it failed first time after flashing. I was able to get it to connect first time with 6500 after flash, and every time there after. :)

1. Directly after uploading to esp32
```
I (3055) wifi: wifi driver task: 3ffb5880, prio:23, stack:3584, core=0
I (6155) wifi: wifi firmware version: 7997e4b
I (6155) wifi: config NVS flash: enabled
I (6155) wifi: config nano formating: disabled
I (6156) wifi: Init dynamic tx buffer num: 32
I (6160) wifi: Init data frame dynamic rx buffer num: 32
I (6165) wifi: Init management frame dynamic rx buffer num: 32
I (6170) wifi: Init management short buffer num: 32
I (6175) wifi: Init static rx buffer size: 1600
I (6180) wifi: Init static rx buffer num: 16
I (6183) wifi: Init dynamic rx buffer num: 32
I (6354) wifi: mode : softAP (24:6f:28:7a:5f:89)
I (6354) wifi: Total power save buffer number: 16
I (6354) wifi: Init max length of beacon: 752/752
I (6356) wifi: Init max length of beacon: 752/752
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
I (6372) wifi: mode : sta (24:6f:28:7a:5f:88)
connecting[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[E][WiFiSTA.cpp:219] begin(): connect failed!
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
I (6395) wifi: flush txq
I (6397) wifi: stop sw txq
I (6400) wifi: lmac stop hw txq
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
I (12909) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

Attempts reconnect(6)..I (13649) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
..I (14384) wifi: state: init -> auth (b0)
I (14387) wifi: state: auth -> assoc (0)
I (14393) wifi: state: assoc -> run (10)
I (14425) wifi: connected with VM380XXXX, channel 6, BW20
.I (14429) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
.[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.0.62, MASK: 255.255.255.0, GW: 192.168.0.1
.conencted IP:192.168.0.62
setup processed end
```

2. First power off on
```
I (3043) wifi: wifi driver task: 3ffb57c4, prio:23, stack:3584, core=0
I (6121) wifi: wifi firmware version: 7997e4b
I (6121) wifi: config NVS flash: enabled
I (6121) wifi: config nano formating: disabled
I (6122) wifi: Init dynamic tx buffer num: 32
I (6126) wifi: Init data frame dynamic rx buffer num: 32
I (6131) wifi: Init management frame dynamic rx buffer num: 32
I (6137) wifi: Init management short buffer num: 32
I (6142) wifi: Init static rx buffer size: 1600
I (6146) wifi: Init static rx buffer num: 16
I (6150) wifi: Init dynamic rx buffer num: 32
I (6219) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[... 46 lines cut ...]
I (7137) wifi: state: auth -> init (8a0)
I (7137) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
_I (7178) wifi: flush txq
I (7178) wifi: stop sw txq
I (7179) wifi: lmac stop hw txq
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
I (13686) wifi: mode : sta (24:6f:28:7a:5f:88)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

Attempts reconnect(6)I (13824) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
..I (14559) wifi: state: init -> auth (b0)
I (14563) wifi: state: auth -> assoc (0)
I (14569) wifi: state: assoc -> run (10)
I (14602) wifi: connected with VM380XXXX, channel 6, BW20
.I (14663) wifi: pm start, type: 1

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.0.62, MASK: 255.255.255.0, GW: 192.168.0.1
.conencted IP:192.168.0.62
setup processed end
```

Hieromon, 2021-01-11:
@Anton2k No matter how it is into stable, 65 sec is too long. This way cannot be applied as standard specification to the AutoConnect library.

According to the latest release candidate of the arduino-esp32 v1.0.5 (it's currently staged as RC4), `WiFiSTAClass::begin` is clearly modified regarding DHCP start compared to v1.0.4, and if v1.0.5 stabilizes, this problem can be expected to be resolved.
Therefore, I propose to you that the **workaround will be applied in compile time** with a particular directive like **`#define AUTOCONNECT_RECONNECT_WAITTIME`**.

In this case, I will release the library with `AUTOCONNECT_RECONNECT_WAITTIME` set to `0` in `AutoConnectDefs.h`. So, you need to edit the definition value of `AUTOCONNECT_RECONNECT_WAITTIME` included in `AutoConnectDefs.h` directly before building the Sketch or use a build system such as PlatformIO that can specify the `#define` directive as `-D` parameter at build time.

How about this?

#### For reference

- arduino-esp32 core v1.0.4

```cpp
wl_status_t WiFiSTAClass::begin()
{

    if(!WiFi.enableSTA(true)) {
        log_e("STA enable failed!");
        return WL_CONNECT_FAILED;
    }

    wifi_config_t current_conf;
    if(esp_wifi_get_config(WIFI_IF_STA, &current_conf) != ESP_OK || esp_wifi_set_config(WIFI_IF_STA, &current_conf) != ESP_OK) {
        log_e("config failed");
        return WL_CONNECT_FAILED;
    }

    if(!_useStaticIp) {
        if(tcpip_adapter_dhcpc_start(TCPIP_ADAPTER_IF_STA) == ESP_ERR_TCPIP_ADAPTER_DHCPC_START_FAILED){
            log_e("dhcp client start failed!");
            return WL_CONNECT_FAILED;
        }
    } else {
        tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA);
    }

    if(status() != WL_CONNECTED && esp_wifi_connect()){
        log_e("connect failed!");
        return WL_CONNECT_FAILED;
    }

    return status();
}
```

- arduino-esp32 core v1.0.5RC4

```cpp
wl_status_t WiFiSTAClass::begin()
{

    if(!WiFi.enableSTA(true)) {
        log_e("STA enable failed!");
        return WL_CONNECT_FAILED;
    }

    wifi_config_t current_conf;
    if(esp_wifi_get_config(ESP_IF_WIFI_STA, &current_conf) != ESP_OK || esp_wifi_set_config(ESP_IF_WIFI_STA, &current_conf) != ESP_OK) {
        log_e("config failed");
        return WL_CONNECT_FAILED;
    }

    if(!_useStaticIp && set_esp_interface_ip(ESP_IF_WIFI_STA) != ESP_OK) {
        log_e("set ip failed!");
        return WL_CONNECT_FAILED;
    }

    if(status() != WL_CONNECTED){
    	esp_err_t err = esp_wifi_connect();
    	if(err){
            log_e("connect failed! 0x%x", err);
            return WL_CONNECT_FAILED;
    	}
    }

    return status();
}
```

Anton2k, 2021-01-12:
@Hieromon I thinks it's 6.5 seconds, not 65 seconds. Yes that sounds like a good compromise until the main lib fixes the core of the problem, thanks again for your help :)

Hieromon, 2021-01-13:
@Anton2k, Thank you for more frequent results reporting.

I have staged a patched release around the issue.
You can evaluate a workaround using the [**patches/292**](https://github.com/Hieromon/AutoConnect/tree/patches/292) branch.
As soon as the evaluation gives good results, I will release it as v1.2.3.

#### Specifications for the workaround
- **AUTOCONNECT_RECONNECT_DELAY** makes a duration for a delay from the 1<sup>st</sup>-WiFi.begin failure to the start of autoReconnect. The default is 0, and a value &ge; 1 causes a delay.  
    To enable AUTOCONNECT_RECONNECT_DELAY, Whether edit the relevant part of AutoConnectDef.h appropriately or give the `-DAUTOCONNECT_RECONNECT_DELAY=6500` with the build system like PlatformIO.
    https://github.com/Hieromon/AutoConnect/blob/f2b8748b936b86c37925dd6c1450c159b9aeddb1/src/AutoConnectDefs.h#L154

- Its effect is only when AutoConnect::autoReconnect is enabled.

If you disagree with the above specifications, please suggest a correction. I hope the patch can resolve the issue.
Thank you.

txlim96, 2021-04-08:
Hi @Anton2k and @Hieromon, I am here because I faced the same problem when connecting to only certain router with certain internet service provider (to be specific, only when connecting to sim card router aka portable router, no issue with connecting to home wifi or mobile hotspot).

I saw someone suggested to try the method below in this [link,](https://www.esp32.com/viewtopic.php?t=15768) as they suggested this workaround can perform a hardware reset to the peripheral:
`#include "driver/periph_ctrl.h"
...
periph_module_reset(PERIPH_WIFI_MODULE);
delay(500);
WiFi.begin(ssid, passw);
...
`

Not sure whether that helps, was wondering could this be one of the solutions.

svdrummer, 2021-04-08:
What is the make and model of your portable hotspot? Seems the problem is
following your hotspot. Mine works on home, portable hotspot etc. Is your
password valid, correct number of digits, standard characters, not 5gig
band etc. Look further into your equipment as  thousands of people are not
having an issue. This is where I would start looking.

txlim96, 2021-04-08:
Hi @svdrummer, please bear with me as I am still new to this field. May I know how can I determine the make of the portable router? The model of the portable router should be: Huawei B315s-22 B315 (it is a simcard router, and I'm connecting to its 2.5GHz network).

My ESP32 is able to connect to the portable hostpot, and reconnect back (just that it won't reconnect back **after some time** (but manage to instantly reconnect back when I power cycle it). So I'm still searching for the root cause of this problem
FYI: it is quite difficult for me to collect the log, as this happens at the place where I set up the ESP32 (remote).

svdrummer, 2021-04-08:
This github is a library errors, faults, bug fixes and updates. It isn't
really a discussion and help page. The library author is generally pretty
good, but... Github is not a discussion group. Generally, a connection
check is done in the main loop. If a connection is lost then retry, else do
an ESP.reset()
I only mention this as you said you are new to this. Each post then gets
sent to thousands of email addresses.
I for one am getting flooded with emails from Autoconnect github.
I say this in the nicest possible way.

Hieromon, 2021-04-08:
@txlim96 I would recommend that you carry out a reproduction test you using a simple sketch. Do not include AutoConnect handling in the sketch. It would be logic like my previous comment, for example: https://github.com/Hieromon/AutoConnect/issues/292#issuecomment-756634645
**Note:** Please specify `Verbose` with `Core Debug Level` in Arduino IDE tool menu and get the log.

If you can reproduce the problem, then add the following code. It is following your advice.
```cpp
  #include <driver/periph_ctrl.h>
  ...
  esp_log_level_set("wifi", ESP_LOG_VERBOSE);
  esp_log_level_set("dhcpc", ESP_LOG_VERBOSE);

  // Insert the following 2 lines
  periph_module_reset(PERIPH_WIFI_MODULE);
  delay(500);

  WiFi.mode(WIFI_STA);
  Serial.print("connecting");
```

If your sketch can reconnect to your mobile router after disconnection once, you've found gold. I look forward to your results.
However, if the issue discussed here cannot be reproduced, `periph_module_reset` will not be our savior. Also, If you can't connect always with this test sketch, the cause is not AutoConnect.