Hieromon/AutoConnect #296: SYSTEM_EVENT_AP_STADISCONNECTED

veoreilly, 2021-01-10:
Hello, I recently purchased the ESP32-S2-WROOM module. Attempting to go through and verify things are still working with previous setups I have used on other ESP devices. 

When I attempt to compile the "Simple" (or any AutoConnect) example, I receive this error: 

Arduino\libraries\AutoConnect\src\AutoConnect.cpp:1455:21: error: 'SYSTEM_EVENT_AP_STADISCONNECTED' is not a member of 'arduino_event_id_t'
     }, WiFiEvent_t::SYSTEM_EVENT_AP_STADISCONNECTED);

Thoughts on this?

Hieromon, 2021-01-11:
Which version of an arduino-esp32 core are you using?

> 'SYSTEM_EVENT_AP_STADISCONNECTED' is not a member of 'arduino_event_id_t'

A datatype as `arduino_event_id_t` is not defined in the stable release of the arduino-esp32 core v1.0.4. Also, It was defined for the first time to support esp-idf 4.2.
The ESP32-S2 SoC, unlike the ESP32, is supported since [esp-idf v4.2](https://github.com/espressif/esp-idf/releases/tag/v4.2).

> ESP-IDF Release v4.2
> - Support for ESP32-S2 SoC

The stable release v1.0.4 Of the arduino-esp32 core is based on IDF v3.3 and is not compatible with ESP32-S2. AutoConnect does not support these yet.

See https://github.com/Hieromon/AutoConnect/issues/223

veoreilly, 2021-01-11:
I appreciate the response. Thank you for your time!
I will create my own.

roetske, 2021-04-26:
First of all great programming. 
Have same issue but do not understand which esp32 is compatible?
I have a esp32 -wroom. same issue. Which esp32 selection arduino board works with your code?
error is generated line 1455 = dependant arch if defined(ARDUINO_ARCH_ESP32)

roetske, 2021-04-26:
did temp comment on  AutoConnect.cpp action arch sp8266 in arch esp32 and compilation worked also code
void AutoConnect::_setReconnect(const AC_STARECONNECT_t order) {
#if defined(ARDUINO_ARCH_ESP32)
/**  if (order == AC_RECONNECT_SET) {
    _disconnectEventId = WiFi.onEvent([](WiFiEvent_t e, WiFiEventInfo_t info) {
      AC_DBG("STA lost connection:%d\n", info.disconnected.reason);
      AC_DBG("STA connection %s\n", WiFi.reconnect() ? "restored" : "failed");
    }, WiFiEvent_t::SYSTEM_EVENT_AP_STADISCONNECTED);
    AC_DBG("Event<%d> handler registered\n", static_cast<int>(WiFiEvent_t::SYSTEM_EVENT_AP_STADISCONNECTED));
  }
  else if (order == AC_RECONNECT_RESET) {
    if (_disconnectEventId) {
      WiFi.removeEvent(_disconnectEventId);
      AC_DBG("Event<%d> handler released\n", static_cast<int>(WiFiEvent_t::SYSTEM_EVENT_AP_STADISCONNECTED));
    }
  }
 */
  bool  strc = order == AC_RECONNECT_SET ? true : false;
  WiFi.setAutoReconnect(strc);
  AC_DBG("STA reconnection:%s\n", strc ? "EN" : "DIS");
#elif defined(ARDUINO_ARCH_ESP8266)
  bool  strc = order == AC_RECONNECT_SET ? true : false;
  WiFi.setAutoReconnect(strc);
  AC_DBG("STA reconnection:%s\n", strc ? "EN" : "DIS");
#endif

Hieromon, 2021-05-10:
@roetske ESP-WROOM32 will not generate an error if you select the right module.
https://www.espressif.com/en/products/modules

roetske, 2021-05-29:
thx for info.

LordIkol, 2021-08-11:
had the same issue with NodeMCU ESP-32S commenting out the section @roetske posted got it working for now

gellis12, 2021-09-05:
> The stable release v1.0.4 Of the arduino-esp32 core is based on IDF v3.3 and is not compatible with ESP32-S2.

Just a heads up, it looks like Espressif just released stable 2.0.0, which updated to ESP-IDF 4.4: https://github.com/espressif/arduino-esp32/commit/5502879a

I updated to the latest stable via the boards manager in Arduino, and got this same error with my ESP-WROOM-32 using the same board configuration that worked in previous versions. As above, @roetske's patch fixed it for me.

Hieromon, 2021-09-06:
I have prepared a tentative patch to avoid compilation errors in core 2.0.0. https://github.com/Hieromon/AutoConnect/commit/cf9dca7bb7be606a1ed75f478204a42c0a8da747
The latest AutoConnect library code that avoids compilation errors has been staged to [enhance/v130](https://github.com/Hieromon/AutoConnect/tree/enhance/v130)  branch.

However, this temporary measure may not resolve all breaking changes in core 2.0.0. Especially in ESP32S2, instability remains in the behavior of the WebServer class. Also, I have encountered a BIN size increase of about 10% in core 2.0.0. I've already started working on support for core 2.0.0 with AutoConnect, but it's taking longer than I expected.

LennartHennigs, 2021-09-12:
I have the same issue with an M5Stack-Core-ESP32

ChaitanyaYeole02, 2021-09-29:
I am having the same issue with NodeMCU32S what should I do?

ChaitanyaYeole02, 2021-09-29:
My deadline is tomorrow for pusing into production and I am not able to get even AutoConnect working as it was working before

Hieromon, 2021-09-29:
@LennartHennigs @ChaitanyaYeole02 I have already [mentioned](https://github.com/Hieromon/AutoConnect/issues/296#issuecomment-913331315) that I have staged the supported RC to a [branch](https://github.com/Hieromon/AutoConnect/tree/enhance/v130). It is about the support status of AutoConnect for ESP32 arduino core 2.0.0.
You have the opportunity to evaluate AutoConnect v1.3.0 with [PageBuilder v1.5.0](https://github.com/Hieromon/PageBuilder/tree/enhance/v150) based on the ESP32 core 2.0.0. Have you tried it?
If you still see this issue as a result based on its branch, you can report it according to the [FAQ](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) in the documentation.

closed 2021-09-30 by Hieromon

Neoris-Ru, 2022-01-20:
> Hello, I recently purchased the ESP32-S2-WROOM module. Attempting to go through and verify things are still working with previous setups I have used on other ESP devices.
>
> When I attempt to compile [...]

Change SYSTEM_EVENT_AP_STADISCONNECTED to ARDUINO_EVENT_WIFI_AP_STADISCONNECTED

PR #406 (2021-09-30): Enhance/v130
Fix #403
Fix #387
Fix #361 
Fix #356
Fix #331 
Fix #325
Fix #296 
Fix #223