Hieromon/AutoConnect #297: Can I embed AutoConnect into HTTPS GET request sketch?

advadv00, 2021-01-11:
I really appreciate your great job!

I want to embed AutoConnect into this HTTPS GET request sample sketch.

https://github.com/witnessmenow/arduino-sample-api-request/blob/master/ESP8266/HTTP_GET/HTTP_GET.ino

`
#include <WiFi.h>
#include <WiFiClientSecure.h>

//------- Replace the following! ------
char ssid[] = "SSID";       // your network SSID (name)
char password[] = "PASS";  // your network key

// For HTTPS requests
WiFiClientSecure client;

// Just the base of the URL you want to connect to
#define TEST_HOST "www.tindie.com"

void setup() {

  Serial.begin(115200);

  // Connect to the WiFI
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  
  // If you don't need to check the fingerprint
  client.setInsecure();

  // give the esp a breather
  yield();

  // Send HTTP request
  client.print(F("GET "));
  // This is the second half of a request (everything that comes after the base URL)
  client.print("/api/v1/order/?format=json&limit=1&username=brianlough&api_key=refwv43tgfvdfbe4444&offset=4");
  client.println(F(" HTTP/1.1"));

  //Headers
  client.print(F("Host: "));
  client.println(TEST_HOST);

  client.println(F("Cache-Control: no-cache"));

.
.
.

}
`

I tried many times but failed.
Is there any way to embed AutoConnect?

Hieromon, 2021-01-11:
If you're using WiFiClientSecure correctly, you probably don't need to be aware of it. Is your sketch working fine without AutoConnect?

The sketch below works in my environment.

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <AutoConnect.h>

AutoConnect portal;

bool Http_Get_access(String& host, String& url, String& argument) {
  BearSSL::WiFiClientSecure  client;
  client.setTimeout(500);
  client.setInsecure();
  
  const int httpPort = 443;
  const char* host2 = host.c_str();
  if (!client.connect(host2, httpPort)) {
    Serial.println("connection failed");
    return false;
  }
  client.print(String("GET ") + url + (argument.length() ? String("?") + argument : String()) + " HTTP/1.1\r\n" +
                      "Host: " + host + "\r\n" +
                      "User-Agent: ESP8266/1.0\r\n" +
                      "Connection: close\r\n\r\n");
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      client.stop();
      Serial.println("Client timeout");
      return false;
    }
  }
  while (client.available()) {
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  return true;
}

bool  firstOne;

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  portal.begin();
  firstOne = true;
}

void loop() {
  if (firstOne) {
    if (WiFi.status() == WL_CONNECTED) {
      firstOne = false;
      String  host = "news.yahoo.com";
      String  url = "/";
      String  argument = "";
      Http_Get_access(host, url, argument);
    }
  }

  portal.handleClient();
}
```

advadv00, 2021-01-21:
There is a crash with /mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA library....
when I use MatrixPanel double buffer, network connection is unstable.
after disabled double buffer, it work well.

closed 2021-01-25 by advadv00