Hieromon/AutoConnect #298: nvs_open failed - Credential initialization problem

Brunez3BD, 2021-01-17:
Hi, I'm experiencing a problem with the AutoConnect initialization. My setup is the following:
- Platform: PlatformIO
- board: esp32-wroom with 16MB flash (board json and partitions in zip attached)
- configuration: platformio.ini in zip attached
- sketch: I'm trying the library with the [most simple sketch](https://hieromon.github.io/AutoConnect/gettingstarted.html#lets-do-the-most-simple-sketch) in Getting Started page

I've uploaded the sketch with the following build flags (as reported 
[Autoconnect_problem_setup.zip](https://github.com/Hieromon/AutoConnect/files/5826361/Autoconnect_problem_setup.zip)
in platformio.ini file):
```
-D CORE_DEBUG_LEVEL=5
-D AC_DEBUG
```

And this is the serial output:
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[AC] Current:
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND 'AC_CREDT'
[AC] Preferences begin failed to import AC_CREDT
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[E][WiFiSTA.cpp:219] begin(): connect failed!      
[AC] WiFi.begin() failed
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0
[AC] SoftAP esp32ap/12345678 Ch(1) IP:172.217.28.1 
[AC] http server started
[AC] DNS server started
```

I'm using only the esp32-wroom module with external 3.3V supply (>500mA).
I've tried to debug and solve the problem, but debugging with the serial monitor is a huge pain in the ass. 
Maybe someone else experienced this problem or know how to fix it. 
Thank you!

Hieromon, 2021-01-18:
At first inspection, it looks similar to #289.
This initiation failure is intentional unless there are some errors in saving and loading the saved AutoConnect credentials. 
This failure occurs when the nvs key used by AutoConnect does not exist (that is, the nvs immediately after initialization). This tells AutoConnect that it is ready to assign a new AutoConnect Credential within the nvs.

In the current Preferences API, the only way to determine if the specified key area exists is to open it once. (As far as I understand) We can avoid initiation failure by handling ESP-IDF natively, but I try to avoid having a direct interface with ESP-IDF as much as possible.

Hieromon, 2021-10-09:
This behavior is intentional and is a recognition that does not interfere with the functionality of AutoConnect.
Please post again when you find a problem with the AutoConnect operation. Closed.

closed 2021-10-09 by Hieromon