Hieromon/AutoConnect #299: Fileselect element exist?

advadv00, 2021-01-25:
I want to list SPIFFS files, and delete a file by selecting it from the file list.
It seems to be a feature that is not currently implemented.
Do you have any plans to support fileselect feature?
Or... is there any way to implement it?

Hieromon, 2021-01-27:
AutoConnect library examples has already an example that resembles the sketch you envision.
Refer to https://github.com/Hieromon/AutoConnect/blob/master/examples/Credential/Credential.ino

Also, you can use [**AutoConnectRadio**](https://hieromon.github.io/AutoConnect/acelements.html#autoconnectradio) element to list the files and select which one to delete.
The elements you mentioned are not supported as a library as they are only effective for specific purposes and are less useful. Implementing equivalent functionality with Sketch is not too difficult.
(Please note that SPIFFS has deprecated if you use ESP8266)

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <LittleFS.h>
#include <AutoConnect.h>

const char  FILELIST_AUX[] PROGMEM = R"(
{
  "uri": "/list",
  "title": "Delete",
  "menu": true,
  "element": [
    {
      "name": "files",
      "type": "ACRadio",
      "label": "Files:",
      "arrange": "vertical"
    },
    {
      "name": "msg",
      "type": "ACText",
      "format": "%s files available"
    },
    {
      "name": "del",
      "type": "ACSubmit",
      "value": "Delete",
      "uri": "/del"
    }
  ]
}
)";

const char  FILEDEL_AUX[] PROGMEM = R"(
{
  "uri": "/del",
  "title": "Delete",
  "menu": false,
  "element": [
    {
      "name": "delfile",
      "type": "ACText"
    }
  ]
}
)";

ESP8266WebServer  server;
AutoConnect portal(server);
AutoConnectAux  auxList;
AutoConnectAux  auxDel;

String fileList(AutoConnectAux& aux, PageArgument& arg) {
  aux["files"].as<AutoConnectRadio>().empty();

  int n = 0;
  Dir dir = LittleFS.openDir("/");
  while (dir.next()) {
    aux["files"].as<AutoConnectRadio>().add(dir.fileName());
    n++;
  }
  if (n) {
    aux["msg"].value = String(n);
    aux["del"].enable = true;
  }
  else {
    aux["msg"].value = "No";
    aux["del"].enable = false;
  }
  return String();
}

String fileDel(AutoConnectAux& aux, PageArgument& arg) {
  String  fileName = auxList["files"].as<AutoConnectRadio>().value();
  if (fileName.length()) {
    if (LittleFS.remove(fileName.c_str()))
      aux["delfile"].value = fileName + String(" has deleted");
    else
      aux["delfile"].value = fileName + String(" can not delete");
  }
  else
    aux["delfile"].value = "Missing file to delete";

  Serial.println(aux["delfile"].value);
  return String();
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  // Responder of root page handled directly from WebServer class.
  server.on("/", []() {
    String content = "Place the root page with the sketch application.&ensp;";
    content += AUTOCONNECT_LINK(COG_24);
    server.send(200, "text/html", content);
  });

  if (!LittleFS.begin())
    Serial.println("LittleFS mount failed");

  auxList.load(FPSTR(FILELIST_AUX));
  auxList.on(fileList);
  auxDel.load(FPSTR(FILEDEL_AUX));
  auxDel.on(fileDel);
  portal.join({ auxList, auxDel});
  portal.begin();
}

void loop() {
  portal.handleClient();
}
```

#### NOTE
From now on, please post this kind of question to [**Discussions**](https://github.com/Hieromon/AutoConnect/discussions). Your question may provide useful help to other users.
The issue topic will eventually close, so other users may miss it.

Thank you.

Hieromon, 2021-10-09:
It close due to no report of the result.
Please use Discussions for such posts in the future.