Hieromon/AutoConnect #303: configure new ap blank page

imad-123, 2021-02-06:
please help me i get a blank page when i want to configure new ap on esp8266

Hieromon, 2021-02-08:
Please refer to [Turn on the debug log options](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options) of the documentation to get the log. The log is output to the serial monitor, so enable serial by `Serial.begin(baudrate)` in the `setup()` function.
If you can see the symptom similar described on "[Some Auto Connect page is cut off](https://hieromon.github.io/AutoConnect/faq.html#some-autoconnect-page-is-cut-off)", the heap of esp8266 is insufficient.

This issue may be resolved by updating AutoConnect to v1.2.3. Try a branch staged on the [enhance/v123](https://github.com/Hieromon/AutoConnect/tree/enhance/v123).
If you update AutoConnect to v1.2.3, you **need to update the [PageBuilder to v1.5.0](https://github.com/Hieromon/PageBuilder/tree/enhance/v150)** as well.
These versions are not yet available in the Arduino IDE Library Manager as they are pre-release. Download these as zip and place these into your library folder manually.

imad-123, 2021-02-08:
it works thanks mr. Hieromon

Hieromon, 2021-02-08:
Closed due to resolved.

closed 2021-02-08 by Hieromon