Hieromon/AutoConnect #304: How to keep ESP32 AutoConnect Captive Portal Always Running

nowzenith, 2021-02-07:
this is my code

nowzenith, 2021-02-07:
#include <WiFi.h>          // Replace with WiFi.h for ESP32
#include <WebServer.h>     // Replace with WebServer.h for ESP32
#include <AutoConnect.h>

WebServer Server;          // Replace with WebServer for ESP32
AutoConnect      Portal;
AutoConnectConfig Config;

void setup() {
  Serial.begin(115200);
  Serial.println();

  Config.autoReconnect = true;    // Attempt automatic reconnection.
  Config.reconnectInterval = 1;   // Seek interval time is 30[s].
  Config.autoRise = false;
  Config.immediateStart = false;
  Config.retainPortal = true;
  Portal.config(Config);
  Portal.begin();
}

void loop() {
  delay(1000);
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("wificonnect!!! !");
  }
  else {
    Serial.println("failed!!! !");
  }

  Portal.handleClient();
}

Hieromon, 2021-02-08:
I'm not sure what meaning you mentioned that is "**always**", but If your expectation is that "when you access the SSID launched as a SoftAP of ESP32 with AutoConnect, the AutoConnect page (it will be defined by AutoConnectConfig::bootUri) will **always pop up automatically**", AutoConnect will **not** always work that way.
When AutoConnect establishes a connection to any AP, it will shut down the DNS server, which is essential for automatic pop-ups. It won't restart until the WiFi connection is lost again. (If its DNS continues to work, all URL requests accessed by the sketch will be redirected to SoftAP)
Then, the auto pop-up will restart working when the ESP32 loses connection to any AP.

This behavior is described as a specification in [Launch the captive portal on-demand at losing WiFi](https://hieromon.github.io/AutoConnect/adcpcontrol.html#launch-the-captive-portal-on-demand-at-losing-wifi).

Note:
There is a verbose part in your code below. It is launching a Web Server instance twice.
```cpp
WebServer Server; // Replace with WebServer for ESP32
AutoConnect Portal;
```
Modify to:
```cpp
WebServer Server;
AutoConnect Portal(Server);
```
or
```cpp
AutoConnect Portal;
```

nowzenith, 2021-02-08:
Sorry i want webserver alway running because when esp32 connect network webserwer will close but i dont want to close it i want 
webserver to alway open

nowzenith, 2021-02-08:
i am bad at eng sorry

Hieromon, 2021-02-08:
>  when esp32 connect network webserwer will close

It's your perception incorrect. AutoConnect will start the WebServer internally, but does not stop it even after establishing a WiFi connection.
Refer to: [Inside AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html)

Koxx3, 2021-02-13:
I am looking for the same thing ;)
currently, once the ESP has switch to "wifi client", I got "Place the root page with the sketch ...".
I'd like to be able to keep the config page (especially to use the Autoconnect lib as a configuration portal with "elements").
Is it possible ?

Hieromon, 2021-02-15:
@Koxx3 Essentially, "Place the root page with the sketch ..." is a home page prepared by the user's sketch. (In general, its page will be the root page of your sketching application)

The following document describes how AutoConnect handles URIs, but this only addresses the occurrence of access to AutoConnect's URIs:
[Assign user sketch's home path](https://hieromon.github.io/AutoConnect/adnetwork.html#assign-user-sketchs-home-path)

I can't quite understand what you mentioned "keep the auto-connect portal/config page". However, if you want the AUX page to be displayed when accessing the ESP module's IP address (that is, its URI is `'/'`), you must redirect from the `'/'` to the URI of the AUX page. To redirect access to `'/'`, sketch the ESP32WebServer request handler as follows:

```cpp
// The following code is assuming the ESP32 as a platform.
WebServer server;
AutoConnect portal(server);

void handleRoot() {
  // Redirect to AUX page
  String auxPage = String("http://") + Server.client().localIP().toString() + String("/elements");
  server.sendHeader("Location", auxPage);
  server.send(302, "text/plain", "");
  server.client().stop();
}

void setup() {
  // If you want to limit redirects to /elements only when a WIFI connection is established,
  // register the handleRoot after the return value of portal.begin is true.
  server.on("/", handleRoot);
  portal.begin();
}

void loop() {
  portal.handleClient();
}
```

Hieromon, 2021-10-09:
Closed once due to no activity. It will reopen with a new post.

closed 2021-10-09 by Hieromon