Hieromon/AutoConnect #305: The OTA update doesn't work

Vasioky, 2021-02-08:
I've tried on 3 esp8266s, 2 nodeMCU and one esp-01. After the update, it restarts but doesn't come to live, doesn't connect to wifi, doesn't run portal and doesn't show any serial messages. I've tried also with ESP8266HTTPUpdateServer as described and it worked perfectly for all three modules.

Hieromon, 2021-02-08:
[Enable AC_DEBUG](https://hieromon.github.io/AutoConnect/adothers.html#debug-print) and get debug logs to see if AutoConnectOTA completed successfully, also the status of `@cp` process after updating. Check if the OTA area on the flash is sufficient as a matter you should check.
Since the flash size of esp-01 is **512K** bytes at most, there is a possibility that the OTA area is insufficient. The `@cp` process detects the lack of flash space. You claim that ESP8266HTTPUpdateServer works, but the ESP8266HTTPUpdateServer class also fails if the sketch size exceeds the OTA area.
The `@cp` process copies the new binary uploaded to the OTA area to the boot area. The completion of the operation can be confirmed by the following message (as `ld` following `@cp:0`):

```
[AC] Reset

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
@cp:0
ld
```

FYI (It is a quote from the [ESP8266 Arduino core documentation](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html?highlight=ota#troubleshooting)) :
> The most common causes of OTA failure are as follows:
> - not enough physical memory on the chip (e.g. ESP01 with 512K flash memory is not enough for OTA).
> - too much memory declared for the filesystem so new sketch will not fit between existing sketch and the filesystem – see [Update process - memory view](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html?highlight=ota#update-process-memory-view).
> - too little memory declared in Arduino IDE for your selected board (i.e. less than physical size).
> - not resetting the ESP module after initial upload using serial port.
>
> For more details regarding flash memory layout please check [File system](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html). For overview where new sketch is stored, how it is copied and how memory is organized for the purpose of OTA see [Update process - memory view](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html?highlight=ota#update-process-memory-view).

Vasioky, 2021-02-15:
Hey @Hieromon, thanks. It works

closed 2021-02-15 by Vasioky