Hieromon/AutoConnect #306: about SDcard

nowzenith, 2021-02-10:
i found some error when i use SD library and i don't know how to fix it

this is a error text

In file included from C:\Users\nowze\Desktop\temp2_v_1_3\temp2_v_1_3.ino:10:0:
C:\Users\nowze\Documents\Arduino\libraries\esp32-micro-sdcard/mySD.h:32:7: error: redefinition of 'class fs::File'
 class File : public Stream {
       ^
In file included from C:\Users\nowze\Documents\Arduino\libraries\AutoConnect-master\src/AutoConnectOTA.h:24:0,
                 from C:\Users\nowze\Documents\Arduino\libraries\AutoConnect-master\src/AutoConnect.h:39,
                 from C:\Users\nowze\Desktop\temp2_v_1_3\temp2_v_1_3.ino:6:
C:\Users\nowze\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src/FS.h:47:7: error: previous definition of 'class fs::File'
 class File : public Stream
       ^

Hieromon, 2021-02-11:
@nowzenith Would you please post all `#include` directive statements for your sketch that encountered this error?
I'll consider a workaround.

nowzenith, 2021-02-11:
#include <EEPROM.h>
#include <PubSubClient.h>
#include <WiFi.h>
#include <WebServer.h>
#include <time.h>
#include <AutoConnect.h>
#include <AutoConnectCredential.h>
#include <Wire.h>
#include <SPI.h>
#include <mySD.h>
#include <SD_Extension.h>
#include <FS.h>
#include <SPIFFS.h>

#include "DHT.h"

#include <Adafruit_GFX.h> 
#include <Adafruit_ST7735.h> 

#include <HTTPClient.h>

#include <Tone32.h>

nowzenith, 2021-02-11:
for download sdcard library
https://kmitlthailand-my.sharepoint.com/:u:/g/personal/63010881_kmitl_ac_th/Edpb8LMSaK5Ls4SvHkOFfuUBk_W2S06YG99xDBztK0wI2g?e=wupRAB

https://kmitlthailand-my.sharepoint.com/:u:/g/personal/63010881_kmitl_ac_th/ESlvaGkrRqdKmwkPQWYoljQBadnLfDHHQXHklIP4aqAaAQ?e=FRMdPS

Hieromon, 2021-02-15:
@nowzenith Since `mySD.h` defines the `File` class as global, it conflicts with `FS.h` provided by the ESP32 Arduino core as standard.

The main reason is that the `SD_Extension` class depends on `mySD.h`.
Does the `SD_Extension` library have compatible with the ESP32 Arduino core? In my view, if the sketch uses the filesystem with `FS.h` provided by the ESP32 Arduino core, including `SD_Extension.h` will cause a compilation error always in the sketch. It's not just about AutoConnect.

The filesystem of the ESP32 Arduino core is separated from the physical media, and the physical IO layer as a lower layer provides an API that can be accessed in the same procedure regardless of whether it is EEPROM or SD.
https://github.com/espressif/arduino-esp32/tree/master/libraries/SD

SD_Extension should be ported to follow this structure.

nowzenith, 2021-02-16:
Thank you so much

Hieromon, 2021-10-09:
Closed once due to no activity. It will reopen with a new post.

closed 2021-10-09 by Hieromon