Hieromon/AutoConnect #307: ESP wont connect back to stored WiFi

plusmartin, 2021-02-11:
Hello Hieromon, 

Im sorry to insist on this but I still cannot make the ESP autoconnect to a saved credential automatically.
The credential is there but I must select it manually every time I reset the ESP.
The only wifi that the ESP always connects to is that which I declare in the code:

const char *WIFI_SSID = "wifi1";
const char *WIFI_PASSWORD = "12345678";

But If this wifi is not there, the portal shows up (this is ok) but it does NOT connect to the saved wifis.

Please help me fix this insignificant issue.

If I cannot connect to the hard coded wifi I run this:

`void APmode()
{

  //disableCore0WDT();
  Serial.println("AP mode");
  Serial.println("Creating portal and trying to connect...");

  //Config.immediateStart = true;
  Config.autoReconnect = true;
  Config.hostName = "mydevice";
  Config.portalTimeout = 60000;
  Config.apid = "mydevice";
  //Config.apip = 192.168.0.1;
  Config.retainPortal = false; 
  Portal.config(Config);
  //esp_task_wdt_reset();

  // Establish a connection with an autoReconnect option.
  bool acEnable;
  acEnable = Portal.begin();

  if (acEnable)
  {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
    Portal.handleClient();
  }

  if (!acEnable)
  {
    Serial.println("portal end");

    delay(100);

    WiFi.reconnect();

  }
} // end AP mode`

plusmartin, 2021-02-11:
Also, 
I keep getting an insane amount of serial prints that say this:

[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9

How can I get rid of these messages? it makes my serial port impossible to read.

thank you

Hieromon, 2021-02-11:
@plusmartin Hello,
> The credential is there but I must select it manually every time I reset the ESP.

Hmm, that's strange behavior.
Would you like to work with me to find out the cause? It's a bit of a hassle, but I think it's necessary to investigate the cause of this phenomenon.

You are probably using the ESP32 as a platform. So, would you start by dumping the raw credentials stored with AutoConnect? 

1. Enable `AC_DEBUG`, `Core Debug Level=Verbose` and take a debug log
    Collect the AutoConnect debug log to see what is happening internally. In addition, set `Verbose` as the `Core Debug Level` at the sketch building. If the autoReconnect setting is enabled, it will give us a clue as to "Why aren't previous credentials used?"
    To enable `AC_DEBUG`, you can follow the procedure described in the documentation:
    https://hieromon.github.io/AutoConnect/adothers.html#debug-print
    To set the `Core Debug Level`, navigate the Arduino IDE and select it from `Tool` menu.
    
    **Please Note**:
    Follow the guidelines:**[Reports the issue to AutoConnect Github repository](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository)** when you post to an issue. It helps both of us to solve the problem early.

2. Dump AutoConnectCredential and diagnose saved credentials
    Upload and execute the following sketch. It will output the saved credentials by AutoConnect to the serial monitor.
    Then, you can validate the correctness of saved credentials according to refer to the data structure described in the documentation: [The credential entry](https://hieromon.github.io/AutoConnect/credit.html#the-credential-entry)
    
    The sketch to dump credentials is following:
    ```cpp
    #include <Arduino.h>
    #include <ctype.h>
    #include <WiFi.h>
    #include <WebServer.h>
    #include <AutoConnect.h>
    
    #define DUMP_WIDTH_COLUMNS    8
    #define DUMP_WIDTH_REMAIN(x)  (x & (DUMP_WIDTH_COLUMNS - 1))
    
    void setup() {
      delay(1000);
      Serial.begin(115200);
      Serial.println();
    
      AutoConnectCredential credit;
      Serial.printf("Credential entries:%d\n", credit.entries());
      for (uint8_t e = 0 ; e < credit.entries(); e++) {
        Serial.printf("\nentry #%d:", e);
        station_config_t  entry;
        if (credit.load(e, &entry)) {
          size_t  p = 0;
          while (p < sizeof(station_config_t)) {
            if (!DUMP_WIDTH_REMAIN(p))
              Serial.printf("\n%04x  ", p);
            Serial.printf("%02x ", (uint8_t)*(reinterpret_cast<uint8_t*>(&entry) + p++));
            if (!DUMP_WIDTH_REMAIN(p) || (p >= sizeof(station_config_t))) {
              int sp = DUMP_WIDTH_REMAIN(p) ? 3 * DUMP_WIDTH_REMAIN(p) + 1 : 1;
              Serial.printf("%*c", sp, ' ');
              int w = DUMP_WIDTH_COLUMNS - DUMP_WIDTH_REMAIN(p);
              for (uint8_t cp = p - w; w; cp++, w--)
                Serial.printf("%c", isprint((int)*(reinterpret_cast<uint8_t*>(&entry) + cp)) ? (char)*(reinterpret_cast<uint8_t*>(&entry) + cp) : '.');
            }
          }
          Serial.println();
        }
        else {
          Serial.printf(" loading failed\n");
        }
      }
    }
    
    void loop() {
    }
    ```

plusmartin, 2021-02-25:
@Hieromon hi!
I just installed from zero a new ESP32 running the Arduino framework on PlatformIO.
I installed the latest AutoConnect library and loaded the basic code from: https://hieromon.github.io/AutoConnect/gettingstarted.html

1. The ESP boots and generates the wifi, then I can successfully add a new wifi, OK.
2. If I reboot the ESP it reconnects automatically, OK.
3. If I remove the wifi, then reboot the ESP and wait  LESS  than around 30 sec it reconnects automatically, OK.
4. If I remove the wifi, then reboot the ESP and wait MORE than around 30 sec it does NOT reconnect automatically, ERROR.

[E][WiFiSTA.cpp:219] begin(): connect failed!

No matter how many times I reboot the ESP it will always give the same error and the only way to connect it to the wifi is by searching for "esp32ap", and selecting the wifi manually in Open "SSIDs".

This is the problem that I have. I only want that the ESP connects to the wifi automatically always.
I hope you can help me fix this issue.
Thank you!

plusmartin, 2021-03-03:
Hi @Hieromon ,
I hope you can give some guidance on the issue that I presented in my past post. I tried rebooting and now it wont reconnect (so task 2 of my previous post is now: ERROR).  thank you

svdrummer, 2021-03-03:
The library writer asked you to load some code and reply with the result.
Have you done that yet? We can't see the result, so it is hard to decode.
Bear in mind, tens of thousands of people use this library, so we are
trying to help with your configuration.
I notice you pass your SSID and password by reference, but we don't know
your string allocation size.

Hieromon, 2021-03-12:
@plusmartin

1. What exactly did you do with the "I removed wifi" you mentioned?
2. > wait MORE than around 30 sec    

    What does "wait" mean?

Unfortunately, your reporting phenomenon does not help problem diagnostics for me.
The first thing to doubt is whether the AutoConnectCredential is saved successfully. You can see that in the sketch I posted.

The next thing I want you to do is to enable AC_DEBUG and trace the logs.
The one-line message you provided, `[E] [WiFiSTA.cpp: 219] begin(): connect failed !`, will be output even if AutoConnect works properly. [Inside AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html), `WiFi.begin` is issued multiple times. The first WiFi.begin has no arguments, which is a technique that encourages the use of the previous connection information that Espressif IDF has stored independently. If IDF does not have the credentials at this time, WiFi.begin will fail and AutoConnect will detect the error and load a valid AutoConnect Credential and try to reconnect. AutoConnect will scan nearby WiFi to determine a valid AutoConnectCredential and will try to connect using it.
I need the AC_DEBUG and CORE_DEBUG_LEVEL logs to analyze if these sequences are working.

**P.S.**
It has been confirmed that reconnection is not successful on some routers. Please read the carefully the #292.