Hieromon/AutoConnect #308: Style aux form

Koxx3, 2021-02-13:
Is there any way to add styles to align elements correctly ?
i didn't find in the documentation 😅

 [image]

Hieromon, 2021-02-15:
Each element placed on a custom web page has a `div` element with a `.noorder` class as the parent element. You can qualify the style of the particular elements by using CSS selector with the `.nooder` class. 
You allow the sketch to override the style of selected elements by declaring the CSS using **ACStyle** element within your custom web page definition.

For example, the CSS to align the label columns of a custom web page is following:
```css
.noorder {
    width: 100%
}

.noorder label {
    display: inline-block;
    width: 10.5em;
    cursor: pointer;
    padding: 5px
}

.noorder input[type='number'] {
    width: 4em;
    margin: 3px 0.3em 0 0;
    padding: 3px
}
```

The JSON for a custom web page declaration to enable this CSS in a sketch using the ACStyle element looks like this:

```json
{
  "uri": "/elements",
  "title": "SmartElec config",
  "menu": true,
  "element": [
    {
      "name": "cs",
      "type": "ACStyle",
      "value": ".noorder{width:100%}.noorder label{display:inline-block;width:10.5em;cursor:pointer;padding:5px}.noorder input[type='number']{width:4em;margin:3px 0.3em 0 0;padding:3px}"
    },
    {
      "name": "cap",
      "type": "ACText",
      "value": "E-scooter characteristics",
      "style": "color:navy"
    },
    {
      "name": "wheel",
      "type": "ACInput",
      "label": "Wheel size",
      "apply": "number"
    },
    {
      "name": "motors",
      "type": "ACInput",
      "label": "Motor number of poles",
      "apply": "number"
    },
    {
      "name": "throttle",
      "type": "ACCheckbox",
      "label": "Throttle regeneration",
      "value": "throttle",
      "labelposition": "infront",
      "checked": true
    },
    {
      "name": "sep",
      "type": "ACElement",
      "value": "<hr style=\"height:1px;border-width:0;color:gray;background-color:#52a6ed\">",
      "posterior": "par"
    },
    {
      "name": "load",
      "type": "ACSubmit",
      "value": "Load",
      "uri": "/load"
    },
    {
      "name": "save",
      "type": "ACSubmit",
      "value": "Save",
      "uri": "/save"
    }
  ]  
}
```
```

Koxx3, 2021-02-15:
thanks a lot for this answer. i'll try that !

closed 2021-02-16 by Hieromon