Hieromon/AutoConnect #314: AutoConnect into web page

Jimmy76gr, 2021-02-17:
const char MAIN_page[] PROGMEM = R"=====(
<!DOCTYPE html>
<html>
<head>
	<title>ESP32 Weather Station</title>
	<meta name='viewport' content='width=device-width, initial-scale=1.0'>
	<link href='https://fonts.googleapis.com/css?family=Open+Sans:300,400,600' rel='stylesheet'>

<style>
html { font-family: 'Open Sans', sans-serif; display: block; margin: 0px auto; text-align: center;color: #44444;}
body{margin: 0px;}
h1 {margin: 30px auto 30px;} 
	.side-by-side{display: table-cell;vertical-align: middle;position: relative;}
	.text{font-weight: 600;font-size: 19px;width: 200px;}
	.reading{font-weight: 300;font-size: 30px;padding-right: 15px;}
	.temperature .reading{color: #F29C1F;}
	.humidity .reading{color: #3B97D3;}
	.pressure .reading{color: #26B99A;}
	.altitude .reading{color: #955BA5;}
	.superscript{font-size: 17px;font-weight: 600;position: absolute;top: 5px;}
	.data{padding: 10px;}
	.container{display: table;margin: 0 auto;}
	.icon{width:65px}
	
</style>

any idea how to add autoconect  
 <a> AUTOCONNECT_LINK(COG_24) </a>???
not works...

Hieromon, 2021-02-18:
No matter how you use an expression in a raw string literal, it will not be evaluated as an expression. It's the [C++ language standard](https://en.cppreference.com/w/cpp/language/string_literal).
You must terminate the literal temporarily to include the expression.

```cpp
const char PARA[] PROGMEM = R"(
<p></p><p style="padding-top:15px;text-align:center">)" AUTOCONNECT_LINK(COG_24) R"(</p>
)";
```

Jimmy76gr, 2021-02-19:
thank you, it works

closed 2021-02-19 by Jimmy76gr