Hieromon/AutoConnect #315: AC_AUTH_DIGEST causes Portal pages not be accessible after WiFi STA is connected.

v-c, 2021-02-27:
When using AC_AUTH_DIGEST with AC_AUTHSCOPE_PORTAL setting, it appears I can no longer access the _ac page. Deploy simple sketch below to reproduce the problem. When accessing the "_ac" page with a browser, keep getting the password prompt, but no matter what password is provided it is rejected.

WebServer Server;
AutoConnect Portal(Server);

void initAutoConnect(){
  AutoConnectConfig Config;
  Config.auth = AC_AUTH_DIGEST;
  Config.authScope = AC_AUTHSCOPE_PORTAL;
  Config.username = "Foo";
  Config.password = "Bar";
  Config.autoReconnect = false;
  Config.autoReset = true;
  Config.autoRise = true;
  Config.immediateStart = false;
  Config.ota = AC_OTA_BUILTIN;
  Config.autoSave = AC_SAVECREDENTIAL_AUTO;
  Config.apid = Hostname;
  Config.hostName = Config.apid;
  Config.ticker = true;
  Config.tickerPort = LED_BUILTIN;
  Config.tickerOn = LOW;
  Config.title = "TTGO Leak Detector";
  Config.principle = AC_PRINCIPLE_RSSI;
  Config.menuItems = AC_MENUITEM_CONFIGNEW | AC_MENUITEM_DISCONNECT | AC_MENUITEM_RESET | AC_MENUITEM_UPDATE | AC_MENUITEM_HOME;
  Portal.config(Config);
  Portal.begin();
}

// the setup function runs once when you press reset or power the board
void setup() {
  initAutoConnect();
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {
  Portal.handleClient();
}

v-c, 2021-02-28:
I did some further investigation and it appears Internet Explorer _can_ access the _ac page(!) but pretty much every other browser I tried (Chrome, Firefox, Edge) cannot.

Hieromon, 2021-03-16:
Is there a possibility of this? https://github.com/espressif/arduino-esp32/issues/3642

DIGEST authentication doesn't work well with ESP Arduino core 1.0.4.
I have confirmed this phenomenon and thought that it would be fixed in 1.0.5. (Noted in the [AutoConnect documentation](https://hieromon.github.io/AutoConnect/adauthentication.html#applying-http-authentication))
Haven't it fixed yet?

v-c, 2021-03-16:
I'm running PlatformIO with Espressif32 platform v3.1.0  released on 2021-02-26T13:38:32Z (latest version at the time of this comment) which does say "Updated Arduino framework to 1.0.5" and am still running into this problem. Perhaps you or someone else can confirm this configuration also has the unresolved problem with DIGEST auth?

v-c, 2021-03-16:
Also, the issue reported here is not only on iOS but on Android and Windows as well. Verified using browsers Chrome, Edge, Firefox. Note that if you are updating a sketch that goes from AUTH_BASIC to AUTH_DIGEST, you will need to view the _ac page in an incognito window after uploading the new sketch to the device. It appears that if you use a browser window that has already done AUTH_BASIC, the page is still accessible (though the new sketch has set the auth to AUTH_DIGEST)

Hieromon, 2021-03-16:
@v-c Can you validate the code below in your environment?

Although this problem has been created by the implementation of ESP8266WebServer, the ESP32 core has inherited it. Then in ESP8266, it was fixed. https://github.com/esp8266/Arduino/pull/5506

So if a problem can be reproduced, the problem that I have posted in the past there is a possibility of the remains left.

```cpp
#include <WiFi.h>
#include <ESPmDNS.h>
#include <ArduinoOTA.h>
#include <WebServer.h>

const char* ssid = "........";
const char* password = "........";

WebServer server(80);

const char* www_username = "admin";
const char* www_password = "esp32";
const char* www_realm = "Custom Auth Realm";
String authFailResponse = "Authentication Failed";

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("WiFi Connect Failed! Rebooting...");
    delay(1000);
    ESP.restart();
  }
  ArduinoOTA.begin();

  server.on("/", []() {
    if (!server.authenticate(www_username, www_password))
    {
      return server.requestAuthentication(DIGEST_AUTH, www_realm, authFailResponse);
    }
    server.send(200, "text/plain", "Login OK");
  });
  server.begin();

  Serial.print("Open http://");
  Serial.print(WiFi.localIP());
  Serial.println("/ in your browser to see it working");
}

void loop() {
  ArduinoOTA.handle();
  server.handleClient();
}
```

v-c, 2021-03-16:
@Hieromon - with the sketch you provide I see the same behavior as I see with Autoconnect's _ac pages. Only Internet Explorer can connect to the root page. Every other browser I tried (Chrome/Edge/Firefox) just keeps repeatedly bringing up the username/password dialog until I hit escape, at which point I get "Authentication Failed"

Hieromon, 2021-03-16:
@v-c Thank you for your testing.
It means that the WebServer class of the ESP32 Arduino core has been updated to 1.0.5 but has not yet been fixed.
Resume the issue I previously posted to the ESP32 Arduino core repository.

Hieromon, 2021-03-18:
@v-c 
I checked with ESP32 Arduino core 1.0.5, and authentication was succeeded with my environment.
Also, the fixes addressed for the ESP8266WebServer had properly merged into the ESP32 Arduino core too. https://github.com/espressif/arduino-esp32/pull/3329/commits/214e59d20efcb34138304eb36c07d60855b11e27

An issue that is 1.0.5 but is experiencing the same phenomenon as you are presented, so please trace there. https://github.com/espressif/arduino-esp32/issues/4943

Hieromon, 2021-03-27:
@v-c It seems to have been fixed in 1.0.6 https://github.com/espressif/arduino-esp32/commit/2d3c5763. could you confirm it?

v-c, 2021-03-28:
Thanks @Hieromon. I will confirm this issue is resolved once this change propagates to platformio/espressif.

v-c, 2021-03-31:
With the v3.2.0 of platformio/espressif32, which uses espressif/arduino-esp32 v1.6.0, this issue is resolved. Closing this out.

closed 2021-03-31 by v-c