Hieromon/AutoConnect #318: Not reconnecting to known AP after power lost for all

jcherrabi, 2021-03-06:
Hello there,
I've noticed that the AutoConnect switches to esp8266ap after it fails to connect to known router, which is OK but stays there even after the router comes back to life, and required a reboot of the module to reconnect... my question is how do i enable the module to continue to try to connect even after it goes to soft mode with esp8266ap... i think in this mode it should continue to scan for known wifi with saved credential and try to reconnect if successful then reboots and remove the esp8266ap... what do you think?

why we need this behavior?
well imagine you loose power in your house, and when the power comes back, obviously the module will reboot faster than the wifi router and hence will not be able to connect and kick in the esp8266ap mode and stays there forever even after the wifi router is back to online...

cheers,

plusmartin, 2021-03-06:
this is related to the issue that I posted a few entries below yours.

I have discovered that the ESP connects to wifi successfully IF it is available immediately after reboot. However, if the wifi only appears AFTER it launched the Portal, then it wont connect to it never again, even if rebooted.

In this state, if I reboot it I get this:
[E][WiFiSTA.cpp:219] begin(): connect failed!

The only way to get it back online is to manually enter the captive portal and select the wifi in the SSID list of credentials. So the credential is there but it never tries to connect to it. 
I have reported this issue several months ago.

Im considering extracting the credentials (I dont know where and how they are stored) and save them elsewhere where I can use them in a WiFi.begin(WIFI_SSID, WIFI_PASSWORD); but this defeates the entire purpose of using Autoconnect.

Hieromon, 2021-03-16:
@jcherrabi 
> config.reconnectInterval = 10 means 10x30 (the default) = 300 tries or is it one try every 300 sec?

That means trying to reconnect once every 300 seconds within AutoConnect::handleClient loop.

Hieromon, 2021-10-09:
Closed once due to no activity. It will reopen with a new post.

closed 2021-10-09 by Hieromon

zener79, 2021-12-31:
Hello there,

zener79, 2021-12-31:
Hello there,
I'm facing the same issue. After a power loss my router takes some time to reboot and my ESP reboots in SoftAP mode because of the `Portal.begin()` on my `setup()` but it doesn't reconnect to my WiFi after the router completes the reboot.

From what I understand `reconnectInterval` will only help if you already are inside the `loop()` thanks to `Portal.handleClient()`, but not when you are stuck inside the `Portal.begin()`.

The desired behavior should be the ESP to start in SoftAP mode and then to connect to a known WiFi network when available.
Is this possible? 

Thanks