Hieromon/AutoConnect #323: Sketch execution during the captive portal loop

Koxx3, 2021-03-15:
Hello,

i've seen the explanation to keep the sketch running while waiting for wifi connection.
(cf last chapter here : https://hieromon.github.io/AutoConnect/adcpcontrol.html#disable-the-captive-portal)
This solution is complicated to use to keep the main loop running on big projects with real time constrains.
Is there any other way like running the portal loop in a FreeRTOs task ? if yes, what are the difficulties ?

Koxx3, 2021-03-15:
here is my current implementation :
https://github.com/Koxx3/SmartController_SmartDisplay_ESP32/blob/master/src/WifiSettingsPortal/WifiSettingsPortal.cpp

To be clear, I'd like my main loop running whatever the autoconnect/portal state ;)

Hieromon, 2021-03-16:
I would like to ask you about the basic strategy it has for the code you have presented.
Your AUX page contains JQuery, but if the ESP module doesn't have a connection with any AP, there is no internet transparency. So, will the AUX page work properly in the captive portal state?

And I don't quite understand what you mean.
> This solution is complicated to use to keep the main loop running on big projects with real time constrains.
Is there any other way like running the portal loop in a FreeRTOs task ? if yes, what are the difficulties ?

Do you want to execute a captive portal loop by AutoConnect and sketch-specific processing concurrently?

closed 2021-03-16 by Hieromon