Hieromon/AutoConnect #329: Autoconnect page routes are accessible without authentication irrespective of AutoConnectConfig Config.auth setting.

v-c, 2021-03-16:
When connected as STA, the autoconnect routes linked to menu items disconnect and reset are accessible without authentication irrespective of AutoConnectConfig Config.auth setting. Ideally, these routes would require authentication based on the setting. Nobody should be able to reset my device without me giving them access to do it.

Reproduction: Use basic sketch with this config:

  AutoConnectConfig Config;
  Config.auth = AC_AUTH_BASIC;
  Config.authScope = AC_AUTHSCOPE_PORTAL;
  Config.username = AC_USERNAME;
  Config.password = AC_PASSWORD;

Now, in an incognito browser window, visit http://{local_ip}/_ac/reset

Notice that you are not asked for the username password and are still able to reset the device. Likewise with the http://{local_ip}/_ac/disc route.

Hieromon, 2021-03-17:
@v-c
In v1.2.2 implementation, the authentication range you mentioned is an intentional specification.
The URIs that follow the AutoConnectConfig::auth settings are:
https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnectDefs.h#L93
https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnectDefs.h#L112
https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnectDefs.h#L113
https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnectDefs.h#L115

On the other hand, the following URI does not pass the authentication process:

https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnectDefs.h#L116
https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnectDefs.h#L117

Its criteria for authentication are very simple and are determined by the reqAuth flag and is a fork implemented inside the `AutoConnect::_setupPage` function in `AutoConnectPage.cpp`. Maybe you've confirmed that, and of course you can modify it yourself.

https://github.com/Hieromon/AutoConnect/blob/19938b82309a1bf7d4934103ca130fbe26997e03/src/AutoConnectPage.cpp#L1399

Initially, I implemented it exactly as you mentioned, but after some testing, I changed my mind. Authentication does not work under the captive portal (it will always deny). In other words, the module cannot be reset after establishing a WiFi connection.
There are pros and cons regarding them.

v-c, 2021-03-17:
Thanks for the explanation, especially discussion in #171. I am closing this as it appears the behavior is by design.

closed 2021-03-17 by v-c