Hieromon/AutoConnect #331: When a WiFiClientSecure client are used in a sketch along with AutoConnect library the autoconnect pages are not accessible

v-c, 2021-03-20:
When a WiFiClientSecure client is used in a sketch along with AutoConnect library  the autoconnect pages are not accessible. 

Here's a simple sketch that demonstrates the problem. If you comment out the #define useSecureMQTT and load the sketch, then the _ac pages are accessible again. Note that this behavior has been reproduced on an ESP8266. 

// comment the following line to disable use of Secure MQTT connections
#define useSecureMQTT

#define mqttServer "mqtt.thingspeak.com"
#define MQTT_PASS "__REDACTED__"
#define CHANNEL_ID 000000
#define CHANNEL_API_KEY "__REDACTED__"
#define THINGSPEAK_CERT_FINGERPRINT "27 18 92 dd a4 26 c3 07 09 b9 7a e6 c5 21 b9 5b 48 f7 16 e1"
#define AC_USERNAME "foo"
#define AC_PASSWORD "bar"

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <PubSubClient.h>
#include <AutoConnect.h>

#ifdef useSecureMQTT
  #include <WiFiClientSecure.h>
  #define mqttPort 8883
  WiFiClientSecure client;
#else
  #include <WiFiClient.h>
  #define mqttPort 1883
  WiFiClient client;
#endif

const char mqttPass[] = MQTT_PASS;
const long writeChannelID = CHANNEL_ID;
const char writeAPIKey[] = CHANNEL_API_KEY;
const String Hostname = "NodeMCUv1";
const String topicString = "channels/" + String(writeChannelID) + "/publish/" + writeAPIKey;
bool ledState = false;
const char* thingspeakServerFingerprint = THINGSPEAK_CERT_FINGERPRINT;

ESP8266WebServer Server;
AutoConnect Portal(Server);
long lastWriteMillis = 0;
long lastDisplayUpdateMillis = 0;

PubSubClient mqttClient(client);

#include "main.h"

void rootPage(){
  String index = "<!DOCTYPE html> <html>\n" \
                  "<body>\n" \
                  "Hello, World!\n" \
                  "</body>\n" \
                  "</html>\n";
  char content_type[] = "text/html";   
  Server.send(200,content_type,index);
}

void mqttConnect() {
  // Loop until connected.
  while (!mqttClient.connected())
  {
      String clientID = Hostname + String(random(0xFFFF), HEX);
      // Connect to the MQTT broker.
      if ( !mqttClient.connect( clientID.c_str(), (Hostname.substring(0,4)).c_str(), mqttPass ) )
      {
          // See https://pubsubclient.knolleary.net/api.html#state for the failure code explanation.
          String mqttConnectError = "MQTT connect failed, rc = ";
          mqttConnectError += String(mqttClient.state()) + ". Retrying in 5s.";
          Serial.println(mqttConnectError);
          delay(5000);
      }
  }
}

void mqttPublishRSSIToThingSpeak() {
    if (!mqttClient.connected()) {
      mqttConnect();      
    }
    // Create a topic string and publish data to ThingSpeak channel feed.
    mqttClient.publish(topicString.c_str(), ("field1=" + String(WiFi.RSSI())).c_str());
}

void initAutoConnect(){
  AutoConnectConfig Config;
  Config.auth = AC_AUTH_NONE;
  Config.authScope = AC_AUTHSCOPE_PORTAL;
  Config.username = AC_USERNAME;
  Config.password = AC_PASSWORD;
  Config.autoReconnect = false;
  Config.autoRise = true;
  Config.immediateStart = false;
  Config.ota = AC_OTA_BUILTIN;
  Config.autoSave = AC_SAVECREDENTIAL_AUTO;
  Config.apid = Hostname;
  Config.hostName = Config.apid;
  Config.ticker = true;
  Config.tickerPort = LED_BUILTIN;
  Config.tickerOn = LOW;
  Config.principle = AC_PRINCIPLE_RSSI;
  Config.menuItems = AC_MENUITEM_CONFIGNEW | AC_MENUITEM_DISCONNECT | AC_MENUITEM_RESET | AC_MENUITEM_UPDATE | AC_MENUITEM_HOME;
  Portal.config(Config);
  Portal.begin();
}

void handleWiFiStateChange(WiFiEvent_t event) {
  switch (event) {
    // Disable AP when we are connected to a STA 
    case WIFI_EVENT_STAMODE_GOT_IP:
      WiFi.softAPdisconnect(false);
      WiFi.enableAP(false);
      break;
  }
}

// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  WiFi.onEvent(handleWiFiStateChange);
  initAutoConnect();
  #ifdef useSecureMQTT
    client.setFingerprint(thingspeakServerFingerprint);
  #endif
  mqttClient.setServer(mqttServer, mqttPort);
  Server.on("/", rootPage);
}

// the loop function runs over and over again forever
void loop() {
  Portal.handleClient();
  if (WiFi.isConnected()) {
    // Only do MQTT loop if WiFi is connected
    mqttClient.loop(); 
    // Periodically update RSSI on ThingSpeak channel 
    if ( (lastWriteMillis == 0) || (abs(millis() - lastWriteMillis) > 1*15*1000) ) {
      mqttPublishRSSIToThingSpeak();
      lastWriteMillis = millis();
      digitalWrite(LED_BUILTIN, ledState);
      ledState = !(ledState);
    }
  }
}

v-c, 2021-03-20:
I tried a similar sketch on the ESP32 and it works fine there. I believe the issue may be the underlying libraries, specifically WiFiClientSecure and its interaction with AutoConnect. I am fine if you want to document the issue and close it if you believe it is not something entirely up to AutoConnect to fix.

Hieromon, 2021-03-22:
If the encountered phenomenon with **"the autoconnect pages are not accessible"** is **"accessing /_ac results in a blank page response"**, it may not be a problem due to the WiFiClientSecure.
https://hieromon.github.io/AutoConnect/faq.html#some-autoconnect-page-is-cut-off

Please take the debug log with specifying **AC_DEBUG** and **PB_DEBUG**, then check if the following PageBuilder failure has occurred.
```
[PB] Failed building, free heap:<Size of free heap>
```
If you have seen the failure log as above, it's due to a simply insufficient heap.
This issue has resolved in [AutoConnect v1.2.3](https://github.com/Hieromon/AutoConnect/tree/enhance/v123) and [PageBuiler v1.5.0](https://github.com/Hieromon/PageBuilder/tree/enhance/v150). AutoConnect authentication works with the sketch you provided, at least in my environment with AutoConnect v1.2.3 and PageBuilder v1.5.0.
However, those versions have not been released yet.

v-c, 2021-03-22:
Thanks @Hieromon. I will wait until AutoConnect v1.2.3 and PageBuilder v1.5.0 are released to test and close out this issue if it is no longer reproducible.

Hieromon, 2021-03-22:
@v-c Ok, If possible, you can pre-validate using RC version staged in each branch.
PageBuilder v1.5.0 has already been tested. The release of AutoConnect v1.2.3 will be skipped. The next valid version is v.1.3.0, and its release will still take some time.

PR #406 (2021-09-30): Enhance/v130
Fix #403
Fix #387
Fix #361 
Fix #356
Fix #331 
Fix #325
Fix #296 
Fix #223
closed 2021-09-30 by Hieromon