Hieromon/AutoConnect #335: Exemple mqttRSSI_NA it's not working

DanielSamp4, 2021-03-22:
Primeiramente, parabens pelo trabalho,

Mas os exemplos de mqtt da biblioteca, nao estão conseguindo se conectar com o servidor thingspeak.

[image]
[image]
[image]

I dont know, but i'am putting all the api key's right, but it's not working, kepp printing Connection failed:-2, and this -2 its from PubSubClient::state() and -2 indicates that connection failed.

v-c, 2021-03-22:
The server for ThingSpeak MQTT is mqtt.thingspeak.com.

closed 2021-03-23 by DanielSamp4

Hieromon, 2021-03-23:
Added a supplementary explanation to correctly set the parameters that give to mqttRSSI.ino for connecting to the MQTT broker serviced by ThingSpeak.
https://hieromon.github.io/AutoConnect/howtoembed.html#sketch-publishes-messages