Hieromon/AutoConnect #352: Memory leak

Koxx3, 2021-04-07:
Hello,

I observe what seems to be a memory leak.

Each time I reload on my ACElements page, I loose few kB ;)

[image]

Koxx3, 2021-04-07:
Process to reproduce : switch between pages and use `esp_get_free_heap_size`.

Koxx3, 2021-04-07:
Have you ever observed such trouble ?

Garbage collector seems to recover a part of the memory, but not all : 

```

23:05:46.840 > wifi portal open !
23:05:53.961 > RAM left 43564
23:06:04.138 > RAM left 43424
23:06:14.305 > RAM left 43424
23:06:24.477 > RAM left 43424
23:06:34.641 > RAM left 43424
23:06:44.811 > RAM left 43424
23:06:51.494 > portal where =
23:06:51.496 > load begin
23:06:51.501 > load end
23:06:55.022 > RAM left 41548
23:07:05.185 > RAM left 41548
23:07:15.360 > RAM left 41548
23:07:19.931 > portal where =
23:07:19.933 > load begin
23:07:19.938 > load end
23:07:25.547 > RAM left 41236
23:07:35.716 > RAM left 41236
23:07:45.880 > RAM left 41236
23:07:52.275 > portal where =
23:07:52.278 > load begin
23:07:52.282 > load end
23:07:56.076 > RAM left 40932
23:08:06.244 > RAM left 40932
23:08:16.405 > RAM left 40932
23:08:26.571 > RAM left 40932
23:08:36.731 > RAM left 40932
23:08:46.904 > RAM left 40932
23:08:57.070 > RAM left 41140
23:09:07.239 > RAM left 41140
23:09:17.402 > RAM left 39428
23:09:27.553 > RAM left 41352
23:09:37.718 > RAM left 41352
23:09:47.880 > RAM left 41352
23:09:58.056 > RAM left 39884
23:10:06.062 > calib page
23:10:06.063 > calib : no arg match
23:10:07.460 > flash page
23:10:08.246 > RAM left 40896
23:10:08.895 > portal where =
23:10:08.896 > load begin
23:10:08.902 > load end
23:10:18.455 > RAM left 40264
23:10:28.620 > RAM left 40264
23:10:28.620 > load openSSID
23:10:38.812 > RAM left 38148
23:10:48.975 > RAM left 39860
23:10:59.150 > RAM left 39860
23:11:09.306 > RAM left 39860
23:11:19.468 > RAM left 39860
23:11:29.640 > RAM left 39860
23:11:39.806 > RAM left 39860
23:11:49.970 > RAM left 39860
23:12:00.136 > RAM left 39860
23:12:10.303 > RAM left 40488
23:12:20.479 > RAM left 40700
23:12:30.643 > RAM left 40700
23:12:40.820 > RAM left 40940
23:12:50.992 > RAM left 40940
23:13:01.156 > RAM left 40940
23:13:11.334 > RAM left 40940
23:13:21.507 > RAM left 40940
23:13:31.669 > RAM left 40940

```

Koxx3, 2021-04-07:
even loading alternatively standard "Configure new AP" and "Open SSIDs" pages seems to produce the same effect.

Hieromon, 2021-04-08:
Is it really a leak?
To me, it looks like a string buffer reservation by the WString class.

Koxx3, 2021-04-08:
oh ok ! I didn't know that !

Hieromon, 2022-01-01:
There has been no activity since the IDF version was updated, so close it once. If the same phenomenon happens with the latest version of the core, it will reopen.

closed 2022-01-01 by Hieromon