Hieromon/AutoConnect #353: Sketch wont compile after update from 1.1.7 to 1.2.*

diogo24m, 2021-04-14:
I have a Sketch that worked fine with AutoConnect on version 1.1.7, but after updating to any 1.2 versions it started to give this error on compile:

```
C:\Users\diogo\Documents\Arduino\libraries\AutoConnect\src\AutoConnectPage.cpp: In member function 'void AutoConnect::_authentication(bool, HTTPAuthMethod)':
C:\Users\diogo\Documents\Arduino\libraries\AutoConnect\src\AutoConnectPage.cpp:1614:18: error: 'class PageBuilder' has no member named 'authentication'
   _responsePage->authentication(user, password, method, AUTOCONNECT_AUTH_REALM, fails);
                  ^
```

My board is ESP32 and I'm using **WiFiClientSecure** library instead of **Wifi**

Hieromon, 2021-04-15:
Which PageBuilder version you used?
The version of PageBuilder required for AutoConnect v1.2 and later is mentioned in the documentation.
https://hieromon.github.io/AutoConnect/index.html#required-libraries

diogo24m, 2021-04-15:
You were right.
I didnt notice that in the documentation.

closed 2021-04-15 by diogo24m