Hieromon/AutoConnect #356: Element limitations per custom page?

chr1s692, 2021-04-26:
hi,
i'm using your library to develop my own small portal for my iot device. :) 
These days i wanted to extend my custom page with some more elements, but now the page only shown as empty page. If i reduce the amount of elements, all elements are shown correctly again. Is there a limitation of elements on a custom page?

here some snippets of my code and the serial console:

```
ACStyle   (p_style, "label+input,label+select{position:sticky;left:120px;width:230px!important;box-sizing:border-box;}");
ACElement (newline, "<hr>");

ACText    (p_cap_device, "Device:", "font-family:serif;color:#4682b4;");
ACInput   (p_device_type, "", "Type", "^{a-zA-Z0-9]*$", "Type of your device (e.g. light, sensor)"); 
ACInput   (p_device_name, "", "Name", "^(([a-zA-Z0-9]|[a-zA-Z0-9][a-zA-Z0-9\\-]*[a-zA-Z0-9])\\.)*([A-Za-z0-9]|[A-Za-z0-9][A-Za-z0-9\\-]*[A-Za-z0-9])$", "Name of your device (hostname)"); 
ACCheckbox(p_device_unique, "Unique AP", "Use unique Access Point ID", true);
ACInput   (p_device_user, "", "User");
ACInput   (p_device_pass, "", "Pass", "", "Password of portal user", AC_Tag_BR, AC_Input_Password);

ACText    (p_cap_mqtt, "MQTT:", "font-family:serif;color:#4682b4;");
ACInput   (p_mqtt_server, "", "Server", "^(([a-zA-Z0-9]|[a-zA-Z0-9][a-zA-Z0-9\\-]*[a-zA-Z0-9])\\.)*([A-Za-z0-9]|[A-Za-z0-9][A-Za-z0-9\\-]*[A-Za-z0-9])$", "MQTT broker server");
ACInput   (p_mqtt_user, "", "User");
ACInput   (p_mqtt_password, "", "Pass", "", "Password of MQTT user", AC_Tag_BR, AC_Input_Password);

ACInput   (p_mqtt_port, "1883", "Port", "^[0-9]*$");
ACInput   (p_mqtt_topic, "", "Topic", "^[a-zA-Z0-9/]*$", "MQTT (parent) topic");
ACInput   (p_mqtt_updateperiod, "10", "Period", "^[0-9]*$", "Update Period in seconds");

ACText    (p_cap_leds, "LEDs:", "font-family:serif;color:#4682b4;");
ACInput   (p_leds_datapin, "3", "Data Pin", "^[0-9]*$");
ACInput   (p_leds_number, "10", "Number", "^[0-9]*$");

ACText    (p_cap_sensor, "Sensor:", "font-family:serif;color:#4682b4;");
ACCheckbox(p_sensor_bme_enabled, "sensor_bme_enabled", "Enable bme sensor", false);
ACInput   (p_sensor_bme_updateperiod, "60", "Period", "^[0-9]*$", "Update Period in seconds");
ACCheckbox(p_sensor_bh1750_enabled, "sensor_bh1750_enabled", "Enable bh1750 sensor", false);
ACInput   (p_sensor_bh1750_updateperiod, "1", "Period", "^[0-9]*$", "Update Period in seconds");
ACCheckbox(p_sensor_dht_enabled, "sensor_dht_enabled", "Enable dht sensor", false);
ACInput   (p_sensor_dht_updateperiod, "60", "Period", "^[0-9]*$", "Update Period in seconds");
ACInput   (p_sensor_dht_datapin, "4", "Data Pin", "^[0-9]*$");

//ACCheckbox(p_device_unique, "Unique AP", "Use unique Access Point ID", true);

ACSubmit(save, "Save", AUX_CONFIG_SAVE_URI);
ACSubmit(discard, "Discard", "/_ac");
```

working page definition:
```
AutoConnectAux aux_config(AUX_CONFIG_URI, "Config", true, {
  p_style,
  p_cap_device,
  p_device_type,
  p_device_name,
  p_device_user,
  p_device_pass,
  p_device_unique,
  newline,
  p_cap_mqtt,
  p_mqtt_server,
  p_mqtt_user,
  p_mqtt_password,
  p_mqtt_port,
  p_mqtt_topic,
  p_mqtt_updateperiod,
  newline,
  p_cap_leds,
  p_leds_datapin,
  p_leds_number,
  newline,
  save,
  discard
});
```

non working page definition:
```
AutoConnectAux aux_config(AUX_CONFIG_URI, "Config", true, {
  p_style,
  p_cap_device,
  p_device_type,
  p_device_name,
  p_device_user,
  p_device_pass,
  p_device_unique,
  newline,
  p_cap_mqtt,
  p_mqtt_server,
  p_mqtt_user,
  p_mqtt_password,
  p_mqtt_port,
  p_mqtt_topic,
  p_mqtt_updateperiod,
  newline,
  p_cap_leds,
  p_leds_datapin,
  p_leds_number,
  newline,
  p_cap_sensor,
  p_sensor_bme_enabled,
  p_sensor_bme_updateperiod,
  p_sensor_bh1750_enabled,
  p_sensor_bh1750_updateperiod,
  p_sensor_dht_enabled,
  p_sensor_dht_updateperiod,
  newline,
  save,
  discard
});
```

loading function (copied some days ago from one of your examples):
```
String loadParams(AutoConnectAux& aux, PageArgument& arg) {
  (void)(args);
  
  File param = SPIFFS.open(CONFIG_FILE, "r");
  if (param) {
    if (aux.loadElement(param)) {  
      Serial.println("PORTAL - config loaded");
    }
    else
      Serial.println("PORTAL - config failed to load");
    param.close();
  }
  else {
    Serial.println("PORTAL - config open failed");
#ifdef ARDUINO_ARCH_ESP32
    Serial.println("If you get error as 'SPIFFS: mount failed, -10025', Please modify with 'SPIFFS.begin(true)'.");
#endif
  }
  return String("");
}
```

Sampozzo, 2021-05-05:
Hello I have the same issue on my portal.
I have a page on the portal that do a loop on the sd card file system and create a table on the portal to list the files.
With 100 files it works, with 150 files i have a blank page (only the top portal menu is shown).
No errors in console.

It seems there is a sort of timeout but i can't understand how to increase it.

Have you found a solution?

Hieromon, 2021-05-10:
The root cause of this phenomenon seems to be similar to https://github.com/Hieromon/AutoConnect/discussions/342.
To see if it's a known issue, compile with [PB_DEBUG](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options) enabled and rerun.

If you receive the message described in the documentation at [here](https://hieromon.github.io/AutoConnect/faq.html#some-autoconnect-page-is-cut-off) from serial console, you can use [enhance/v130](https://github.com/Hieromon/AutoConnect/tree/enhance/v130) branch to see if the problem is resolved. (For AutoConnect v1.3.0, [PageBuilder v1.5.0](https://github.com/Hieromon/PageBuilder/tree/enhance/v150) is also required)

PR #406 (2021-09-30): Enhance/v130
Fix #403
Fix #387
Fix #361 
Fix #356
Fix #331 
Fix #325
Fix #296 
Fix #223
closed 2021-09-30 by Hieromon