Hieromon/AutoConnect #360: LoadProhibited when calling portal.end()

cremus, 2021-05-10:
### System information:

Windows 10 Pro (20H2, build: 19042.964)
Arduino IDE 1.8.13
AutoConnect 1.2.2

### Issue

I am trying to implement a functionality where the ESP32-WROOM-32D is reading a button, called the OTA button. This button determines the state of the AutoConnect portal. The portal is not connecting to WiFi, the portal is used to set a variable using a Aux page. When the button is high, portal.begin() is called, when the button is low, portal.end() is called. Sometimes, when portal.end() is called, the following error code shows and the ESP32 reboots.

#### The error code:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40167b28  PS      : 0x00060430  A0      : 0x800d9904  A1      : 0x3ffb1cb0  
A2      : 0x00008100  A3      : 0x3ffcff3c  A4      : 0x00000003  A5      : 0x00000000  
A6      : 0x3ffd0788  A7      : 0x3ffc6cbc  A8      : 0x00008100  A9      : 0x800dba10  
A10     : 0x3ffb1cbc  A11     : 0x006e6962  A12     : 0x3f401dcc  A13     : 0x3ffd07c0  
A14     : 0x3ffd0798  A15     : 0x0000000a  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000081ac  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40167b28:0x3ffb1cb0 0x400d9901:0x3ffb1cd0 0x400e1d7b:0x3ffb1cf0 0x400dba0d:0x3ffb1d40 0x400dbbc4:0x3ffb1e10 0x400dc0ea:0x3ffb1e30 0x400dc10f:0x3ffb1f70 0x400d1f2f:0x3ffb1f90 0x400eb881:0x3ffb1fb0 0x40089b06:0x3ffb1fd0
```

#### AutoConnect config used:
```
  config.ota = AC_OTA_BUILTIN;
  config.apid = "TEST-" + String((uint32_t)(ESP.getEfuseMac() >> 32), HEX);
  config.psk = 12345678;
  config.title = "Towermaster";
  config.menuItems = AC_MENUITEM_UPDATE | AC_MENUITEM_RESET;
  config.portalTimeout = 1;
  config.immediateStart = true;
  config.retainPortal = true;
  portal.disableMenu(AC_MENUITEM_CONFIGNEW | AC_MENUITEM_DISCONNECT | AC_MENUITEM_OPENSSIDS | AC_MENUITEM_HOME);
  portal.config(config);
```

#### Code to read button and change portal status:
```
void loop() {

  int ota = digitalRead(OTA);   // 1 = OFF, 0 = ON

  if (ota == 0) {
    if (!acEnable) {
      Serial.println("enabling portal...");
      portal.begin();
      Serial.println("portal enabled");
      acEnable = true;
    }
    portal.handleClient();
  } else {
    if (acEnable) {
      Serial.println("disabling portal...);
      portal.end();
      Serial.println("portal disabled");
      acEnable = false;
    }
  }
}
```

Hieromon, 2021-05-13:
I need to you show me the condition of the WebServer instance allocation to diagnose the issue. How did the sketch declare for WebServer, `AutoConnect portal` or `WebSrerver server; AutoConnect portal(server);`?

cremus, 2021-05-18:
Sorry for the late reply. I declared WebSserver and autoconnect like this:

```
#include <WebServer.h>
typedef WebServer WiFiWebServer;

#include <AutoConnect.h>

WiFiWebServer server;
AutoConnect portal(server);
```

Bighoneypot, 2021-05-18:
@cremus try to reset ESP32 with this https://www.espressif.com/sites/default/files/tools/flash_download_tool_v3.8.7.zip 

I with another esp32 some time ago I solved it in this way.

cremus, 2021-05-21:
> 
> 
> @cremus try to reset ESP32 with this https://www.espressif.com/sites/default/files/tools/flash_download_tool_v3.8.7.zip
> 
> I with another esp32 some time ago I solved it in this way.

Thank you for trying to help @Bighoneypot. However, I do not understand why this could solve my problem with AutoConnect. Nonetheless, I've tried resetting it with Espressifs tool, but without any result.

Hieromon, 2021-05-22:
@cremus I was able to reproduce the crash based on your posted logic sequence.
However, your sketch logic is not appropriate for handling the ESP32 WebServer class.
Assuming that the switch connected to the `OTA` pin is an alternate type (although your sketch loop logic won't work in the momentary switch anyway), please try making the following changes:

1. Insert `WiFi.mode(WIFI_STA)` in `setup()`, WiFi must be started before the Arduino core's loop task thread (otherwise the TCP stack will not start properly)
2. Avoid `portal.handleClient` when `acEnable` is false.

cremus, 2021-05-22:
@Hieromon Thank you for the suggestions. I will try that coming Tuesday (25-05-2021). I will let you know if anything changed.

cremus, 2021-05-26:
@Hieromon Unfortunately, it did not resolve the issue. I've tried adding the `WiFi.mode(WIFI_STA) at various places in the setup, without luck. I've also removed the `portal.handleClient` when acEnable is false.

Hieromon, 2021-05-26:
@cremus The following code is working in my environment:
```cpp
void loop() {
  int ota = digitalRead(22);
  if (ota == 0) {
    if (!acEnabled) {
      Serial.println("OTA trigger detected");
      Serial.println("enableing portal...");
      portal.begin();
      Serial.println("server started");
      acEnabled = true;
    }
  }
  else {
    if (acEnabled) {
      Serial.println("disabling portal...");
      server.stop();
      Serial.println("portal disabled");
      acEnabled = false;
    }
  }
  if (acEnabled)
    portal.handleClient();
}
```
However, it just doesn't crash and we can't reconnect to the webserver. Because the TCP stack has already been freed. As I said before, I don't think your sketch code is suitable for handling the WebServer class.

Hieromon, 2022-12-14:
No response, Closing.

closed 2022-12-14 by Hieromon