Hieromon/AutoConnect #361: Running example, update, compile error

svdrummer, 2021-05-16:
Hello all,
Is there any traps to running compiling on esp8266?
I am trying the example for "update" When i try to compile i get compile errors saying it wont compile for (whatever esp8266 i have set)
compiles okay for esp32 modules.

svdrummer, 2021-05-17:
In file included from C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.h:30,
                 from C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\examples\Update\Update.ino:20:
C:\Users\cruis\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:182:16: error: 'virtual bool PageBuilder::canHandle(HTTPMethod, String)' marked 'override', but does not override
  182 |   virtual bool canHandle(HTTPMethod requestMethod, String requestUri) override;
      |                ^~~~~~~~~
C:\Users\cruis\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:183:16: error: 'virtual bool PageBuilder::canUpload(String)' marked 'override', but does not override
  183 |   virtual bool canUpload(String uri) override;
      |                ^~~~~~~~~
C:\Users\cruis\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:184:8: error: 'bool PageBuilder::handle(WebServerClass&, HTTPMethod, String)' marked 'override', but does not override
  184 |   bool handle(WebServerClass& server, HTTPMethod requestMethod, String requestUri) override;
      |        ^~~~~~
C:\Users\cruis\Documents\Arduino\libraries\PageBuilder\src/PageBuilder.h:185:16: error: 'virtual void PageBuilder::upload(WebServerClass&, String, HTTPUpload&)' marked 'override', but does not override
  185 |   virtual void upload(WebServerClass& server, String requestUri, HTTPUpload& upload) override;
      |                ^~~~~~
C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\examples\Update\Update.ino: In function 'String onSetup(AutoConnectAux&, PageArgument&)':
C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\examples\Update\Update.ino:114:32: warning: unused parameter 'aux' [-Wunused-parameter]
  114 | String onSetup(AutoConnectAux& aux, PageArgument& arg) {
      |                ~~~~~~~~~~~~~~~~^~~
C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\examples\Update\Update.ino:114:51: warning: unused parameter 'arg' [-Wunused-parameter]
  114 | String onSetup(AutoConnectAux& aux, PageArgument& arg) {
      |                                     ~~~~~~~~~~~~~~^~~
C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\examples\Update\Update.ino: In function 'String onApply(AutoConnectAux&, PageArgument&)':
C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\examples\Update\Update.ino:124:32: warning: unused parameter 'aux' [-Wunused-parameter]
  124 | String onApply(AutoConnectAux& aux, PageArgument& arg) {
      |                ~~~~~~~~~~~~~~~~^~~
C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\examples\Update\Update.ino:124:51: warning: unused parameter 'arg' [-Wunused-parameter]
  124 | String onApply(AutoConnectAux& aux, PageArgument& arg) {
      |                                     ~~~~~~~~~~~~~~^~~
Multiple libraries were found for "SD.h"
 Used: C:\Users\cruis\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.0\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
 Not used: C:\Users\cruis\Documents\Arduino\libraries\SD
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\cruis\Documents\Arduino\libraries\ArduinoJson
 Not used: C:\Users\cruis\Documents\Arduino\libraries\arduino_863954
 Not used: C:\Users\cruis\Documents\Arduino\libraries\ArduinoJson-old

Hieromon, 2021-05-17:
> Is there any traps to running compiling on esp8266?

No such option exists in the latest versions of AutoConnect and PageBuilder. In a properly built and maintained environment, the AutoConnect library bundled examples has been confirmed to compile successfully on **ESP8266 Arduino Core 2.7.4**. Refs: https://travis-ci.org/github/Hieromon/AutoConnect/builds/749314436

It looks like you're already using an ESP8266 3.0.0 core, which was just released two days ago, but I haven't verified it yet. ESP8266 3.0.0 has [*many breaking changes*](https://github.com/esp8266/Arduino/releases/tag/3.0.0) and **lost backward compatibility**. It will take some time for many other libraries to catch up with the latest version of ESP8266 arduino core, not only AutoConnect.

Hieromon, 2021-05-22:
The [AutoConnect v1.3.0 branch](https://github.com/Hieromon/AutoConnect/tree/enhance/v130) is useful as a temporary measure until AutoConnect has completely verified ESP8266 arduino core 3.0.0. That branch is a candidate for the next version of AutoConnect and currently works with ESP8266 core 3.0.0. You also need to upgrade your library to [PageBuilder v1.5.0 RC](https://github.com/Hieromon/PageBuilder/tree/enhance/v150) for AutoConnect v1.3.0 RC. PageBuilder since v1.4.3 is compatible with ESP8266 Core 3.0.0 and is backward compatible.

I have not finished the ESP8266 core 3.0.0 verification at still many use cases, but you can proceed with the operation check of your sketch on the latest Arduino core for ESP8266 using the [AutoConnect v1.3.0 RC](https://github.com/Hieromon/AutoConnect/tree/enhance/v130) with [PageBuilder v1.5.0 RC](https://github.com/Hieromon/PageBuilder/tree/enhance/v150).

If you find a problem working with esp8266 3.0.0, please create an issue topic. I will try to get rid of it by the release of AutoConnect v1.3.0.
Thank you for your contribution.

rbandres98, 2021-06-18:
Same problem here. I solved this using AutoConnect v1.2.3 (from [enhance/v123 branch](https://github.com/Hieromon/AutoConnect/tree/enhance/v123)) and PageBuilder v1.5.0 (from [enhance/v150 branch](https://github.com/Hieromon/PageBuilder/tree/enhance/v150))

> In file included from C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\src/AutoConnect.h:30,
> from C:\Users\cruis\Documents\Arduino\libraries\AutoConnect\examples\Update\Update.ino:20: [...]

Hieromon, 2021-06-25:
@rbandres98 It is no meaning. I’m still working on esp8266 core 3.0.0 and not sure why you used the v123 branch.  It's an internal release.

rbandres98, 2021-06-25:
> @rbandres98 It is no meaning. I’m still working on esp8266 core 3.0.0 and not sure why you used the v123 branch. It's an internal release.

@Hieromon when I tested the library, I used different versions of AutoConnect and PageBuilder but my ESP8266 (Lolin WEMOS D1 R1) do not show portal content in some paths. For example, the Information Page ("/_ac") cannot be reached and device show [this content](https://github.com/Hieromon/AutoConnect/issues/362#issue-900648549) instead. Is by this reason that I'm using internal release. 

I don't know if I was wrong in any procecss, I will try again with release version.

cassc, 2021-08-17:
Similar problem here, had to use older version of ESP8266 core in platformio.ini

```
platform = espressif8266@2.3.0
```

PR #406 (2021-09-30): Enhance/v130
Fix #403
Fix #387
Fix #361 
Fix #356
Fix #331 
Fix #325
Fix #296 
Fix #223
closed 2021-09-30 by Hieromon