Hieromon/AutoConnect #373: Problems with Autoconnect hanging on startup

DanMan32, 2021-07-05:
I know this has been brought up before but I still am not able to get it right.
Autoconnect still seems to hang on autoconnect::begin if it isn't able to connect to the WIFi AP it last connected to.
I had it working once where it immediately dropped out of begin but not this last time.

Here's my config (solarpool is the HelloWorld renamed):
  solarPoolPage.on(onSolarPool);      // Register the attribute overwrite handler.
  portal.join(solarPoolPage);     // Join the hello page.
  config.ota = AC_OTA_BUILTIN;
  config.autoReset = false;     // Not reset the module even by intentional disconnection using AutoConnect menu.
  config.autoReconnect = true;  // Reconnect to known access points.
  config.reconnectInterval = 6; // Reconnection attempting interval is 3[min].
  config.retainPortal = true;   // Keep the captive portal open.
  config.portalTimeout = 1 ;
  portal.config(config);
  portal.begin();

Another related issue:
As I develop my project, I am either at my house or my dad's house.   Autoconnect has STA credentials for the WiFi at both sites stored and I can go to the  Open SSIDs page to connect to the current one.
How do I get it to connect to the one I am located and not just the last one connected?  This is when it hangs on Begin
BTW there should be a way to remove a stored SSID.

This probably should be on another issue, but can you document the non-JSON elements ACText, ACRadio, etc?
I see the corollary between these and AutoconnectElements.   The HelloWorld example uses the former

Hieromon, 2021-07-06:
Does your hang mean that control doesn't escape from AutoConnect::begin?
Please provide the minimum code that can reproduce the phenomenon. It also includes the loop() {}.

Also,

> can you document the non-JSON elements ACText, ACRadio, etc?

What specific description does it need to have? What's missing from this currently published document as [here](https://hieromon.github.io/AutoConnect/apielements.html) and [here](https://hieromon.github.io/AutoConnect/acelements.html#how-to-coding-for-the-elements)?

DanMan32, 2021-07-09:
It looks like it does release Begin if it doesn't know where to connect to and the Captive Web Portal isn't connected but takes 30 seconds which has it seem that it is hung forever.  But now that I know it takes this long, I'll just count to 40 (extra 10 for good measure.
My code displays on OLED display the current WiFi IP which shows as 0.0.0.0 since there's no SSID defined.
If one does not go into the Captive Web Portal, my loop does it's thing but eventually stops.   Perhaps that's because it enabled the Captive again for another 30 seconds?
It's hard to reproduce because once it is within the range of a stored SSID, it works fine and it is not practical to turn off my WiFi.

Sometimes it looks like it hasn't locked in on my WiFi when I am in a setup phase, as my display shows 0.0.0.0 that it received from WiFi.LocalIP() but if I reboot, it locks in on my WiFi and displays the IP.

As for the web documentation, I have found what I am looking for, but wasn't where I expected it to be.   It's actually in the JSON section that defines the AC functions.

Hieromon, 2021-07-19:
> If one does not go into the Captive Web Portal, my loop does it's thing but eventually stops. Perhaps that's because it enabled the Captive again for another 30 seconds?

Yes, it is. Because you set AutoConnectConfig::autoReconnect.

Can you define the behavior for your expected WiFi connection a little more clearly?
It seems to me that you are confusing what you should do with the setup() and the reconnect behavior within the loop(). If the behavior you expect isn't possible with the current AutoConnect, I'm willing to enhance the library as much as possible.

So, this topic move to the discussions.

closed 2021-07-19 by Hieromon