Hieromon/AutoConnect #383: get text but no captal portal

xciser77, 2021-07-23:
installed the packaged on a nodemcu v3. It uploads fine and I can connect to the AP name, but the portal does 'load' .
but its not a portal. I get the text below:

!OTP tltteAtCnetsaitctlfn-aiyHleiaAilsn-ei;otsz:6x-stx-ieajs:0%-ekttx-ieajs:0%-o-s-otsotiggasae-ektfn-mohn:nilae}oymri:;adn:}bs-ae{agn02p 2x.aepnl*lbl:o(bn)dslyiln-lc;it:.e;etainrgt.aepnl*.ls{it:uofn-ie09mmri-et1p;etainlf}nu{mzapaac:oe-ektapaac:oefn-ie09mmri:p uo.a{iiiiyclas}lptre{iiiiyvsbe.a:agt.vra{pct:.;rniin03}lptre mdlbto{pct:;rniin03}oelptp0lf:;it:0%hih:0%psto:ie;pct:;akrud#0;-ne:00.oa_utnbre-ais1p;akrud#603clr#ffcpdig2p 0xtx-lg:etrtx-eoainnn;etrsaig1xfn-egtbl;ipa:nieboktp4%lf:0;it:0;oiinfxdoaiy0zidx10}al{odrclas:olpebre-pcn:;odr1xsld#d;oo:44bcgon-oo:ffmri-otm2p}al.notbeif>fo,al.notedwdh10;odrclr#CDDtbeif>ha{akrudclr#CDDtbeif>ha>rt{oo:fft,hpdig1p 2xtedbcgon-oo:fff;odrbto:p oi ddtedt>hfn-egt40tx-lg:ettotbre-o:p oi ddtoytoyt:t-hl(d)bcgon-oo:fftoyt>dtott>dfn-egt30fn-ie.8mtoyt:t-hl(vn{akrudclr#777tbeif bd>rnhcidee)bcgon-oo:EFF}l-ie{it:0%psto:ie;o:;et0zidx10;o-hdw01x3xrb(,0 ,01) p p ga0 ,0 .4}l-ugrsa,l-ugrsa:bfr,l-ugrsa:atrdslybokhih:p;it:6xtasto:.ses}l-bcekd.bmn i.bbre pnbcgon-oo:rnprn}l-bcekd.bmn i.bbre pn:eoe.bc:hce~l-eul l-ugrsa:atrmri-o:}l-edrdslyfe;lxdrcinrwjsiycnetsaebtenainiescne;egt5p}l-eurgt.bbre{agnlf:uo.bbadfn-ie16mpdig1p 4x1p 4x.bmn{i-egt5p;rniin06 aewdh10}l-aiaindslyfe;lxdrcinclm;itsyenn;adn-et0mri:}l-eua.bie {etdcrto:oeclrihrtcro:one}l-tmhih:8x.bie {adn:8x2p 8x2p;ipa:lc}l-ugrpdig1p 4x1p 4xpsto:eaiecro:one}l-ugrsa:bfr,l-ugrsa:atrcnet';oiinaslt}l-ugrsa:bfr{agntp-p}l-ugrsa:atrmri-o:p}l-bdslynn}l-bnt:hce).bmn{vrlwhde;egt5p}l-bcekd.bmn{rniinhih .ses;egt10hoefo:uo.rponpsto:eaiehih:uomnhih:8x.rponhvru{oiinrltv;ipa:lc;i-it:0%.rpona:fe{oiinaslt;otn:'rgt1p;o:5xbre-it:p p ;odrclrtasaetbre-tl:oi}dodw>ldslybokoefo-:idnls-tl:oepdig0.rponu l-tmmnwdh10;egt2p;adn:p 0x5x4p}dodw>l.bie {i-egt2p;iehih:9xpdig0@ei cenad(i-it:6p).bnvgto{lxfo:o;utf-otn:lxed}l-ugrdslynn;.bc:o(cekd~l-euoefo:iil;.bc:hce~l-euhih:8x}l-eu.bie{odrtp0}l-eurgt.bhae{agnrgtat;.rponhih:8x}dodw:oe>lpsto:bouelf:;o:8xpdig0}dodw>ldslynn;.rponu l-tmpdig5x1p;.rponu l-tmawiesaenwa;}l-bcekd.bmn l-ugrdlpnsa:bfr{rnfr:oae25e).bc:hce+l-eu.bbre-bsi pn:fe{rnfr:oae-2dg}l-eumtra,l-eumtra dodw lbcgon-oo:233;oo:ff.bmn-aeil.cie.bmn-aeil.bie:oe{akrudclr#77f.bmn-aeil.bbre pn.bmn-aeil.bbre pn:eoe.bmn-aeil.bbre pn:fe{akrudclr#f}/tl>/edhae d"b ls=l-ie"dvcas"bmn bmn-ih bmn-aeil>u ls=l-aiain>l ls=l-edr>ahe=/a"cas"bbad>uoonc/ae>/il ls=l-tm>ahe=/a/pn>pnSIs/>/il ls=l-tm d"ee"< rf"rl"Rst.l ls=l-tm>ahe=/>OE/>/i/i>dvcas"a"i=rl"< rf"rst ls=oelp>/>dvcas"oa_utn>h>ahe=/a/ee"cas"oa_utn>EE/i>/i>/edrtbecas"no tl=bre:oe"t>t>salse oncin/dt>t>oe/dt>t>P/d/rt>I ne)/dt>t>untmst>I ne)/

xciser77, 2021-07-29:
was using the 3.0.0 nodemcu, downgraded to 2.7 and it works

closed 2021-07-29 by xciser77

Hieromon, 2021-07-30:
@xciser77 AutoConnect v1.3.0 to support ESP8266 Arduino core 3.0.0 or later will be released soon. Thank you for your patience.
#365