Hieromon/AutoConnect #386: ESP32 GPIO's pins are getting LOW states periodically

opanyki314, 2021-07-30:
Hello.
During using Autoconnect after several minutes all ESP32 GPIO's pins are getting LOW states periodically (approx interval ... minutes). Even if I use autoconnect only. Therefore I can't use UART since Autoconnect periodically sets the "rts_io_num" to LOW  in a function "uart_set_pin" (uart.h from the arduinoespressif32 framework).
How can I use Autoconnect with the UART and GPIO's pins?
Before:
[image]
After:
[image]

Hieromon, 2021-08-02:
There are only two cases where AutoConnect operates GPIO directly:
- Use AutoConnectConfig::ticker = true
- Use AutoConnectConfig::ota = AC_OTA_BUILTIN

Other than that, AutoConnect doesn't deliberately manipulate GPIOs. If you captured unintended GPIO changes, it is due to the Arduino core.

opanyki314, 2021-08-02:
Apparently it is due to the Arduino core. I didn't set anything of this. And if I set Config.ticker = false and Config.ota = AC_OTA_EXTRA nothing changes.

Hieromon, 2021-08-03:
@opanyki314 Thank you for the diagnostics. I also start problem analysis.
In the ESP32 Arduino core, analog pin and WiFi component have been considered in the past. I haven't encountered any reports of similar symptoms in digital IO so far, but some ESP-IDF API calling sequences may cause unintended interference with IO pins.
Refs: https://github.com/espressif/arduino-esp32/issues/102

Let me create and verify a test sketch that allows native WIFI drive without AutoConnect. If there are new discoveries will report here.