Hieromon/AutoConnect #394: arduino lolinD32 can not auto connect wifi at the same time

dalnel, 2021-09-01:
I have 5 lolinD32 after I setting wifi it can auto connect to wifi  in 1 by 1, but when I  power on at the same time(5 lolionD32), it just 3 lolinD32 can auto connect to my wifi, the other need set again in the web site.

I try 3 lolinD32 power on at the same time, they all can connect to my wifi.

I also ask this question in here : https://stackoverflow.com/questions/68993857/arduino-lolind32-can-not-connect-wifi-at-the-same-time

Hieromon, 2021-09-04:
I have a mixed connection of up to 8 ESP32s and ESP8266s by utilized well-balanced channels with stable radio channels, but I haven't encountered the problem you've shown. It is not related to the cause of AutoConnect, but to the ESP32 limits. Neither ESP8266 nor ESP32 have the ability to avoid channel collisions when negotiating with APs. It's up to the user.

I recommend that you try only using WiF:begin with the AutoConenct library removed to identify the problem. The combination of trials will require both DHCP and static IP. If the problem reproducibility is different for DHCP only and static IP only, then there is a problem with your access point's DHCP resource allocation. In this way, proceed with the analysis of where the cause lies. Then, if you have any doubts about the behavior of AutoConnect, please indicate the AutoConnect API where you think it is relevant.

If you cannot identify the conditions associated with the problem event by your own analysis, please post your problem to Discussons. You can expect solutions from more users. However, that is not always the case.

dalnel, 2021-09-07:
> 我使用平衡良好的頻道和穩定的無線電信號，可混合連接多達 8 個 ESP32 和 ESP8266s，但我沒有遇到您顯示的問題。它與自動連接的原因無關，但與 ESP32 限制有關。ESP8266 和 ESP32 在與 AP 談判時都無法避免通道碰撞。這要由用戶決定。
> 此外，您還可以使[`AD_DEBUG`](https://hieromon.github.io/AutoConnect/adothers.html#debug-print)標誌從串行監視器獲取連接失敗時所發生情況的線索。請參閱[文件](https://hieromon.github.io/AutoConnect/faq.html#still-not-stable-with-my-sketch)。
> 
> 我建議您嘗試只使用WiF：從刪除自動會議庫開始，以識別問題。試驗的組合將同時需要 DHCP 和靜態 IP。如果問題可重複性僅針對 DHCP 和僅靜態 IP 不同，則訪問點的 DHCP 資源分配存在問題。這樣，繼續分析原因所在。然後，如果您對自動連接的行為有任何疑問，請指出您認為相關的自動連接 API。
> 
> 如果您無法通過自己的分析識別與問題事件相關的條件，請將您的問題發佈到討論或 Espressif 論壇。您可以期待更多用戶的解決方案。然而，情況並非總是如此。

thank you replay, I all esp32 are used static IP to test
BTW I try to use debug get log now. after i will report my log.

Hieromon, 2021-09-07:
Have you verified with a test sketch that eliminates AutoConnect and drives WiFi.begin directly? That's the first thing you should do. Isolate whether the problem is related to your WiFi environment.
Then set CORE DEBUG LEVEL to Verbose with the sketch build to detect WiFi library and TCP stack errors. That is the first clue.

dalnel, 2021-09-08:
I finish to test I can not copy my error this time....
It is all work good
so I think is my fault when I set the static IP...
Maybe it's an IP conflict .

Hieromon, 2021-09-09:
Good for you that the cause was clarified. Closed.

closed 2021-09-09 by Hieromon