Hieromon/AutoConnect #399: Error after upload file on Upload page with ESP32

nikanet, 2021-09-14:
I use ESP32 and LittleFS. After upload file like .css or images on Upload AC page all files uploaded by Platformio before no longer available via http. On ESP8266 everything is fine but on ESP32 have this issue. Please help.

Hieromon, 2021-09-30:
@nikanet At the time you posted this issue, the ESP32 LittleFS was integrated into core 2.0.0. AutoConnect has supported ESP32 LittleFS since v130. And I released it after your post.
Which branch did you use?
Did you correctly declare the definition macro to [make AutoConnect use the ESP32 LittleFS](https://hieromon.github.io/AutoConnect/filesystem.html#selecting-appropriate-filesystem)?
Also, FileUpload example is hard-coded to use SPIFFS for ESP32. Did you point out this after understanding the contents of the sketch?

Hieromon, 2021-09-30:
This topic may be an issue based on the wrong composition, so it will be moved to the discussion once. It will be returned to ISSUE when the AutoConnect issue becomes apparent.

closed 2021-09-30 by Hieromon