Hieromon/AutoConnect #413: Update Dont Work For AI THinker board

afshinjian, 2021-10-09:
Hi Hieromon
Thank you for doing a very good project.
The update section does not work for the AI Thinker ESP32 cam model.
My code is too long 

[rest2-OTA-autoconnect.zip](https://github.com/Hieromon/AutoConnect/files/7316197/rest2-OTA-autoconnect.zip)

Hieromon, 2021-10-10:
See [here](https://hieromon.github.io/AutoConnect/adothers.html#debug-print) also [FAQ](https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options), and get the **AC_DEBUG** log. I don't review the whole of your code, but I can parse the AC_DEBUG log.
First, verify with you that your sketch meets the requirements for OTA.
In addition, [this area](https://arduino.stackexchange.com/questions/75198/why-doesnt-ota-work-with-the-ai-thinker-esp32-cam-board) or [this REPO](https://github.com/wjsanek/wjsanek) may help you.

By the way, the Web Server required to satisfy all the functions of ESP32-CAM is different from the ESP32 WebServer library. AutoConnect does not officially support it. It may be applicable if you hack it, but it's not included in my plan.

Hieromon, 2021-10-11:
@afshinjian I'm glad you have taken a step toward the solution. I recognized that the "**The update section doesn't work**" you said was **AutoConnect's OTA**. Because [AutoConnect OTA](https://hieromon.github.io/AutoConnect/otabrowser.html#updates-with-the-web-browserupdated-wv115) will show "**Update**" in the menu. On this premise, I have a few questions for you.

1. Partition scheme
Refs: [**AutoConnect FAQ**](https://hieromon.github.io/AutoConnect/faq.html#how-much-memory-does-autoconnect-consume)
You were told that choosing a partition schema was a hint, so I don' know you were choosing the default.csv from the beginning.     But that shouldn't matter directly related to OTA. Because if the compiled binary size exceeds the default size, a link error will occur. If it is not an error, the space required for OTA is reserved. So you can't get a sketch beyond the partition. But you are saying "update section doesn't work". This means that you already have a compiled sketch binary of a size that can accommodate the default partition.    
The definitions below are the `Default` and `Minimal SPIFFS` schemas. The `app0` is the flash area for the sketch binary. This area will be boot. OTA binary will be staged to `app1`. And after reset, it will be copied to `app0`. This is how ESP8266 and ESP32 OTA work. The `app0` and `app1` have the same value in both of the two partition definitions. That is, if you are using the Arduino IDE to create a sketch binary that fits in `app0`, there is already an OTA area (`app1`) to reserved it.

- Default partition scheme
    https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv
    ```
    # Name,   Type, SubType, Offset,  Size, Flags
    nvs,      data, nvs,     0x9000,  0x5000,
    otadata,  data, ota,     0xe000,  0x2000,
    app0,     app,  ota_0,   0x10000, 0x140000,
    app1,     app,  ota_1,   0x150000,0x140000,
    spiffs,   data, spiffs,  0x290000,0x170000,
    ```

- Minimal SPIFFS
    https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv
    ```
    # Name,   Type, SubType, Offset,  Size, Flags
    nvs,      data, nvs,     0x9000,  0x5000,
    otadata,  data, ota,     0xe000,  0x2000,
    app0,     app,  ota_0,   0x10000, 0x1E0000,
    app1,     app,  ota_1,   0x1F0000,0x1E0000,
    spiffs,   data, spiffs,  0x3D0000,0x30000,
    ```
    
    Therefore, the problem you encounter is not directly related to the partition schema. am I wrong?

2. GPIO0 to flash the sketch binary
Controlling the IO0 pin is required to switch to ESP32 flash mode. Also, we need to trigger the EN (reset) pin to boot automatically after flushing. A typical ESP32 programmer with USB serial, such as the ESP-Prog, controls these two signals with RS-232C DTR / RTS signals. This trick allows flash from a USB host and post-flash reset and serial communication. The ESP32-DevKit has this circuit recommended by [Espressif](https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html#automatic-downloading-function), but the ESP32-CAM standalone module lacks it.

- The flash programing Circuit
    https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html#automatic-downloading-function
    [image]

 - Ai-Thinker ESP32-CAM Schematic
    https://docs.ai-thinker.com/_media/esp32/docs/esp32_cam_sch.pdf
    [image]

    I know there is the [ESP32-CAM AI-Thinker MB Programmer](https://makeradvisor.com/tools/esp32-cam-mb-usb-programmer/) that can supplement the flash programming circuit of the ESP32-CAM, but I don't know if you are using it.
    Refs: [Upload Code to ESP32-CAM AI-Thinker using ESP32-CAM-MB USB Programmer](https://randomnerdtutorials.com/upload-code-esp32-cam-mb-usb/)
    However, the signal control required for this flash programming has nothing to do with OTA. It's only needed for serial programming. So if your "The update section does not work" is OTA, then the GPIO0 connection with GND has nothing to do with the problem you encountered.

So, Could you describe your problem clearly and specifically? Does your "**update**" mean the serial flash programming? (that's Arduino IDE menu Sketch > **Upload**)