Hieromon/AutoConnect #425: Password no longer set correctly

fanzypantz, 2021-11-17:
For some reason, out of nowhere I can't use the new password, nor the default 12345678. I rolled back pretty far to test it out but to no avail. I'm unsure as to why this is happening as I don't think I recently changed that part of the code.

Here is my current code, the setup and loop is called from the main.cpp file:

```c++
#include <Arduino.h>
#include <portal.h>
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
#include <main.h>
#include <Secrets.h>
#include <ESPmDNS.h>
#include <FS.h>
#include <SPIFFS.h>

#define FORMAT_ON_FAIL true
#define GET_CHIPID()    ((uint16_t)(ESP.getEfuseMac()>>32))
using WiFiWebServer = WebServer;
fs::SPIFFSFS& FlashFS = SPIFFS;

// Portal for Autoconnect to WIFI
WiFiWebServer Server;
AutoConnect Portal(Server);
AutoConnectConfig Config("Q-CloudAccessPoint", "administrator");

const char* PARAM_FILE      = "/param.json";
const char* AUX_SETTING_URI = "/app_settings";
const char* AUX_SAVE_URI    = "/app_save";
const char* AUX_CLEAR_URI   = "/app_clear";

String  serverName;
String  channelId;
String  userKey;
String  apiKey;
unsigned long publishInterval = 0;

static const char PAGE_Q_CLOUD[] PROGMEM = R"(
{ "title": "Q-Cloud", "uri": "/", "menu": false, "element": [
    { "name": "caption", "type": "ACText", "value": "<h2>Welcome to Q-Cloud IoT Device</h2>",  "style": "text-align:center;color:#2f4f4f;padding:10px;" },
    { "name": "content", "type": "ACText", "value": "This is some content." } ]
}
)";  

// JSON definition of AutoConnectAux.
// Multiple AutoConnectAux can be defined in the JSON array.
// In this example, JSON is hard-coded to make it easier to understand
// the AutoConnectAux API. In practice, it will be an external content
// which separated from the sketch, as the mqtt_RSSI_FS example shows.
static const char AUX_app_settings[] PROGMEM = R"raw(
[
  {
    "title": "Connection settings",
    "uri": "/app_settings",
    "menu": true,
    "element": [
      {
        "name": "style",
        "type": "ACStyle",
        "value": "label+input,label+select{position:sticky;left:120px;width:230px!important;box-sizing:border-box;}"
      },
      {
        "name": "header",
        "type": "ACText",
        "value": "<h2>MQTT broker settings</h2>",
        "style": "text-align:center;color:#2f4f4f;padding:10px;"
      },
      {
        "name": "mqttserver",
        "type": "ACInput",
        "label": "Server",
        "pattern": "^(([a-zA-Z0-9]|[a-zA-Z0-9][a-zA-Z0-9\\-]*[a-zA-Z0-9])\\.)*([A-Za-z0-9]|[A-Za-z0-9][A-Za-z0-9\\-]*[A-Za-z0-9])$",
        "placeholder": "MQTT broker server"
      },
      {
        "name": "channelid",
        "type": "ACInput",
        "label": "Channel ID",
        "pattern": "^[0-9]{6}$"
      },
      {
        "name": "userkey",
        "type": "ACInput",
        "label": "User Key"
      },
      {
        "name": "apikey",
        "type": "ACInput",
        "label": "API Key"
      },
      {
        "name": "newline",
        "type": "ACElement",
        "value": "<hr>"
      },
      {
        "name": "period",
        "type": "ACRadio",
        "value": [
          "30 sec.",
          "60 sec.",
          "180 sec."
        ],
        "label": "Update period",
        "arrange": "vertical"
      },
      {
        "name": "save",
        "type": "ACSubmit",
        "value": "Save&amp;Start",
        "uri": "/app_save"
      },
      {
        "name": "discard",
        "type": "ACSubmit",
        "value": "Discard",
        "uri": "/"
      }
    ]
  },
  {
    "title": "Connection settings",
    "uri": "/app_save",
    "menu": false,
    "element": [
      {
        "name": "caption",
        "type": "ACText",
        "value": "<h4>Parameters saved as:</h4>",
        "style": "text-align:center;color:#2f4f4f;padding:10px;"
      },
      {
        "name": "mqttserver",
        "type": "ACText",
        "format": "Server: %s",
        "posterior": "br"
      },
      {
        "name": "channelid",
        "type": "ACText",
        "format": "Channel ID: %s",
        "posterior": "br"
      },
      {
        "name": "userkey",
        "type": "ACText",
        "format": "User Key: %s",
        "posterior": "br"
      },
      {
        "name": "apikey",
        "type": "ACText",
        "format": "API Key: %s",
        "posterior": "br"
      },
      {
        "name": "period",
        "type": "ACText",
        "format": "Update period: %s sec.",
        "posterior": "br"
      },
      {
        "name": "clear",
        "type": "ACSubmit",
        "value": "Clear channel",
        "uri": "/app_clear"
      }
    ]
  }
]
)raw";

void getParams(AutoConnectAux& aux) {
  serverName = aux[F("mqttserver")].value;
  serverName.trim();
  channelId = aux[F("channelid")].value;
  channelId.trim();
  userKey = aux[F("userkey")].value;
  userKey.trim();
  apiKey = aux[F("apikey")].value;
  apiKey.trim();
  AutoConnectRadio& period = aux[F("period")].as<AutoConnectRadio>();
  publishInterval = period.value().substring(0, 2).toInt() * 1000;
}

// Load parameters saved with saveParams from SPIFFS into the
// elements defined in /app_settings JSON.
String loadParams(AutoConnectAux& aux, PageArgument& args) {
  (void)(args);
  Serial.print(PARAM_FILE);
  File param = FlashFS.open(PARAM_FILE, "r");
  if (param) {
    if (aux.loadElement(param)) {
      getParams(aux);
      Serial.println(" loaded");
    }
    else
      Serial.println(" failed to load");
    param.close();
  }
  else
    Serial.println(" open failed");
  return String("");
}

// Save the value of each element entered by '/app_settings' to the
// parameter file. The saveParams as below is a callback function of
// /app_save. When invoking this handler, the input value of each
// element is already stored in '/app_settings'.
// In the Sketch, you can output to stream its elements specified by name.
String saveParams(AutoConnectAux& aux, PageArgument& args) {
  // The 'where()' function returns the AutoConnectAux that caused
  // the transition to this page.
  AutoConnectAux&   app_settings = *Portal.aux(Portal.where());
  getParams(app_settings);

  // The entered value is owned by AutoConnectAux of /app_settings.
  // To retrieve the elements of /app_settings, it is necessary to get
  // the AutoConnectAux object of /app_settings.
  File param = FlashFS.open(PARAM_FILE, "w");
  app_settings.saveElement(param, {"mqttserver", "channelid", "userkey", "apikey", "period"});
  param.close();

  // Echo back saved parameters to AutoConnectAux page.
  AutoConnectInput& mqttserver = app_settings[F("mqttserver")].as<AutoConnectInput>();
  aux[F("mqttserver")].value = serverName + String(mqttserver.isValid() ? " (OK)" : " (ERR)");
  aux[F("channelid")].value = channelId;
  aux[F("userkey")].value = userKey;
  aux[F("apikey")].value = apiKey;
  aux[F("period")].value = String(publishInterval / 1000);

  return String();
}

void setupAutoconnect() {

  FlashFS.begin(FORMAT_ON_FAIL);

  if (Portal.load(FPSTR(AUX_app_settings))) {
    PageArgument  args;
    AutoConnectAux& app_settings = *Portal.aux(AUX_SETTING_URI);
    loadParams(app_settings, args);
    Portal.on(AUX_SETTING_URI, loadParams);
    Portal.on(AUX_SAVE_URI, saveParams);
  }
  else {
    Serial.println("load error");
  }  

  // Wifi portal
  Config.apid = "Q-Cloud IoT Portal";
  Config.ota = AC_OTA_BUILTIN;
  Config.auth = AC_AUTH_DIGEST;
  Config.authScope = AC_AUTHSCOPE_AUX;
  Config.username = LOGIN_USERNAME;
  Config.password = LOGIN_PASSWORD;
  Config.portalTimeout = 5 * 60 * 1000; // 5 minutes
  Config.autoReconnect = true;
  Portal.config(Config);

  Portal.load(FPSTR(PAGE_Q_CLOUD));
  
  if(Portal.begin()) {     
      Serial.println("HTTP server: " + WiFi.localIP().toString());
  } else {
      Serial.println("Autoconnect server connection failed");
  }
}

void autoconnectLoop() {
    // Wifi portal
    Portal.handleClient();   
}
```

fanzypantz, 2021-11-17:
I think I got the AutoConnectConfig(const char* ap, const char* password) thing done wrong. How would you set this correctly?

Hieromon, 2021-11-18:
@fanzypantz I tried to reproduce the issue using your code. I've changed the following in your code to allow self-execution testing:

- Commented out missing headers. `portal.h`, `main.h` and `Secrets.h`
- Added const char with `LOGIN_USERNAME`, `LOGIN_PASSWORD`
- Inserted two lines, `Serial.begin(115200);` and `Serial.println();`, just below `void setup() {`
- Changed function name to `setup()` and `loop()`
- Changed captive portal expiry timeout to 60 sec with AutoConnectConfig::portalTimeout

I tested it with Arudino IDE latest version with ESP32 core 2.0.1 and had reproduce the issue. But the problem I encountered is a little different from what you have declared.

SoftAP launched after the *1st-WiFi.begin* fails, it accepts the username and password specified in the sketch correctly. SoftAP remains after the captive portal is shut down after the timeout period expires. It is the specification. But, it broadcasts the default SSID of WIFI_AP like `ESP-4E4FCD`. The phenomenon will occur when the following conditions are met:

- Use ESP32 arduino core 2.0.0 or later
- Use the captive portal timeout period with portalTimeout
- Captive portal starts once but stops due to timeout setting
- At this time, SoftAP will continue, but the SSID will be reset to the default. Therefore, it will have a display name like ESP-4E4FCD, unlike the SSID specified in the sketch. The password specified in the sketch will also be reset.

This issue does not occur with ESP32 Arduino Core 1.0.6, so it seems lose backward compatibility. AutoConnect may not be following the breaking changes in Core 2.0.0.

If the problem you encounter does not meet the above criteria, please post detailed logs with `AC_DEBUG` and `CORE_DEBUG_LEVEL=5` according to [Debug Print](https://hieromon.github.io/AutoConnect/adothers.html#debug-print) in the documentation. I will start investigation while tracking your logs.

Thank you for your contribution.

svdrummer, 2021-11-18:
If I read this correctly, Autoconnect does not work with ezp32 v2 core?

Hieromon, 2021-11-18:
Which part?, What do you think?
First of all, is the phenomenon you encountered the same as what I found out? Answer that first, please.

I had tested for many use cases before releasing AutoConnect. However, it is not possible to cover all cases.
If you want AutoConnect to behave as you expect, first clearly explain what you expect.

Your sketch wants to keep SoftAP even after the captive portal has a timeout, but doesn't let it pop up the portal screen automatically. Your sketch suggests it. Is this your intention?

If so, AutoConnect will not behave as you would expect because ESP32 Core 2.0 has lost backward compatibility. On the other hand, if you want to keep the automatic popup even if the portal times out, use AutoConnect::retainPortal. Instructions on how to use this setting are documented.

Hieromon, 2021-11-18:
@fanzypantz The failure associated with this issue is not a backward compatibility loss, but a security failure.
If ESP32 stops SoftAP while WIFI_AP mode is enabled, SoftAP will fail back and will lose SSID and password. In this state, anybody can connect to SoftAP without a password. A similar phenomenon has been reported in the EMS-ESP32 repo. https://github.com/emsesp/EMS-ESP32/issues/191 It is consistent with the phenomenon mentioned in this topic. And it doesn't seem to be due to Arduino Core 2.0, but to the deeper EDP-IDF. https://github.com/espressif/arduino-esp32/issues/5038

The situation is very dangerous. I have to fix this problem immediately.
So, I have a proposal for you. If your expected behavior is to reopen the captive portal even after the captive portal has temporarily closed due to a timeout and rushed into the loop function, consider using [AutoConnectConfig::retainPortal](https://hieromon.github.io/AutoConnect/adcpcontrol.html#launch-the-captive-portal-on-demand-at-losing-wifi). For example, add a following line to your sketch.

```cpp
...
Config.username = LOGIN_USERNAME;
Config.password = LOGIN_PASSWORD;
Config.portalTimeout = 5 * 60 * 1000;
Config.retainPortal = true; // <- inset here
Config.autoReconnect = true;
Portal.config(Config);
...
```

If it is convenient for you, I'll change the AutoConnect spec. Stops WIFI_AP mode continuation due to captive portal timeout. This fix will prevent the SSID from being broadcast. The ESP32 will be isolated from WiFi access unless the AutoConnectConfig::retainPortal or [AutoConnectConfig::reconnectInterval](https://hieromon.github.io/AutoConnect/adconnection.html#automatic-reconnect-background) is set. However, this fix improves security. 

This bug is due to ESP-IDF, but the situation is dangerous and needs to be fixed immediately. I would appreciate it if you could reply as soon as possible. Thank you.

Hieromon, 2021-11-18:
I prepared the fix with [enhance/v132](https://github.com/Hieromon/AutoConnect/tree/enhance/v132) branch.

fanzypantz, 2021-11-19:
We've been out of internet at work so been preoccupied. Will come back to you again once we got that sorted and I can test it again. 

From what I can see, I should be on 1.0.6 if I can read the build trace correct:

```
PLATFORM: Espressif 32 (3.3.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 50 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <eModbus> 1.2.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|-- <MCCI LoRaWAN LMIC library> 4.0.0+sha.93108f0
|   |-- <SPI> 1.0
|-- <LoRa Serialization> 3.0.1+sha.e47a230
|-- <AutoConnect> 1.3.0
|   |-- <PageBuilder> 1.5.0
|   |   |-- <FS> 1.0
|   |   |-- <SPIFFS> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <WebServer> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.5
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <EEPROM> 1.0.3
|   |-- <Preferences> 1.0
|   |-- <Update> 1.0
|   |-- <Ticker> 1.1
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <HTTPUpdate> 1.3
|   |   |-- <HTTPClient> 1.2
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <WiFiClientSecure> 1.0
|   |   |   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|-- <PCF8574> 0.3.2
|   |-- <Wire> 1.0.1
|-- <AsyncMqttClient> 0.9.0
|   |-- <AsyncTCP> 1.1.1
|-- <ArduinoJson> 6.18.5
|-- <Ai Esp32 Rotary Encoder> 1.4.0
|-- <SPI> 1.0
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS>  #1.0
```

Hieromon, 2021-11-19:
@fanzypantz I tested the sketch along with the [above comment](https://github.com/Hieromon/AutoConnect/issues/425#issuecomment-972724117) in 1.0.6 and it accepts the password correctly. I cannot reproduce your encountered phenomenon with my environment.
Please get the trace as I already mentioned. You have to do this first. Your problem is probably different from the obstacles I have found. I'll start releasing my fix without waiting for the resolution of this topic.

fanzypantz, 2021-11-19:
@Hieromon I can no longer replicate the issue, which isn't very helpful. I didn't update any libraries in use yet. `<AutoConnect> 1.3.0`

The only difference meanwhile on the controller is that I changed `AutoConnectConfig Config("Q-CloudAccessPoint", "administrator");` to `AutoConnectConfig Config;` for a bit to test the settings menu in the portal. To try and replicate it again I went back to the exact point as in the first comment, no changes (stashed changes in git). Erasing the flash before trying to replicate the issue. 

I no longer have the issue, which is not comforting at all. :sweat: I'll be a bit busy but I'll take the controller home and see if I can replicate it again during the weekend.

Hieromon, 2021-11-19:
@fanzypantz You mention that the password is neither the specified value nor the default. It suggests memory corruption, but the phenomenon I encountered is due to the deeper layer of the more radical esp-idf. The obstacles you have experienced are clearly different from the problems I encountered this time. Therefore, it is possible that the password loss phenomenon has not yet been resolved.
In order to reproduce the problem, it is recommended to try by excluding other libraries and using  AutoConnect only. It will be close to the code fix I made (see comments above). Don't forget to enable AC_DBUG and CORE_DEBUG_LEVEL. These logs help problem analysis and reduce the time it takes to find the cause. I have a formal occupation too just like you, and I can't spend too much time developing libraries. Time is important.

PR #437 (2022-01-01): Enhance/v132
Fix #425
closed 2022-01-01 by Hieromon