Hieromon/AutoConnect #436: ACSubmit.onclick() throws error

konradmoesch, 2021-12-23:
I tried adding a custom web page for NTP configuration.
When clicking on an AutoConnectSubmit element, an error is thrown in the javascript console:
`Uncaught TypeError: _bu(...).submit is not a function`
I also tried using the example from the docs, which seems to be broken, too.

Hieromon, 2021-12-24:
Below is the code of the `_bu()` generated by AutoConnect. If this script is corrupted, it means your sketch code is incorrect, read the [FAQ](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) and post the information needed to diagnose the problem.
```javascript
    <script>
      function _bu(url) {
        var uri = document.createElement('input');
        uri.setAttribute('type', 'hidden');
        uri.setAttribute('name', '_acuri');
        uri.setAttribute('value', '&#47;');
        var fm = document.getElementById('_aux');
        fm.appendChild(uri);
        fm.action = url;
        return fm;
      }
      function _sa(url) {
        _bu(url).submit();
      }
      function _ma(el, pos) {
        var mag = pos == 'p' ? el.previousElementSibling : el.nextElementSibling;
        mag.innerText = el.value;
      }
    </script>
```

Hieromon, 2021-12-31:
@konradmoesch I can't identify which example you are referring to, but I checked the examples I have in my library just in case. They all work. If you need any help, please go to Discussions.

closed 2021-12-31 by Hieromon