Hieromon/AutoConnect #438: Not reconnecting to known AP after power lost for all

zener79, 2022-01-01:
Hello there,
I'm facing the same issue already reported several time, but without a definitive solution. 

After a power lost the router takes some time to reboot and my ESP reboots faster in SoftAP mode because of the `Portal.begin()` on my `setup()` but it doesn't reconnect to my WiFi after the router completes the reboot.
A manual reboot of the ESP fixes the issue.

From what I understand `reconnectInterval` will only help if you already are inside the `loop()` thanks to `Portal.handleClient()`, but not when you are stuck inside the `Portal.begin()` in the `setup()` function.

The desired behavior should be the ESP to start in SoftAP mode and then to connect to an already known WiFi network when becomes available again.
Is this possible? 

Thanks

Hieromon, 2022-01-01:
It seems to me that just adjusting the AutoConnectConfig::beginTimeout and portalTimeout values will solve the your problem.
How does the ESP32 know that the WiFi router is restarting (that is, it can connect after a while) in the first place?

> same issue already reported several time

Which issue is it?
In the past, there have been some requests to trigger reconnection while moving the ESP module (i.e., leaving the current connected AP and reconnecting in response to approaching a known AP), and we have settled on the current specification to handle such use cases.

zener79, 2022-01-01:
Hi @Hieromon , and many thanks for the prompt answer on 1st January (happy new year!!!)

Regarding `beginTimeout` it is not clear to me if raising the value (for example) to 180 seconds, it will take 180 secs for the SoftAp to be enabled. In this case it will be not acceptable.

`portalTimeout` is not an option since I don't want the captive portal to be timed-out if there is not an known WiFi network available.

> How does the ESP32 know that the WiFi router is restarting (that is, it can connect after a while) in the first place?

It can not know this. I would expect the device to try to reconnect to a known WiFi in the `begin()` in the same way it does in the `handleClient()` with a `reconnectInterval` different from 0

> Which issue is it?

Look here:
https://github.com/Hieromon/AutoConnect/issues/318
https://github.com/Hieromon/AutoConnect/issues/247
https://github.com/Hieromon/AutoConnect/issues/153
https://github.com/Hieromon/AutoConnect/issues/193
https://github.com/Hieromon/AutoConnect/issues/118

and probably some others.

Hieromon, 2022-01-01:
@zener79 Hmmm, do you expect the `WiFi.begin` timeout to be automatically extended according to the situation without launching the captive portal? 

If the first WiFi begin fails and the captive portal opens, AutoConnect currently offers the following ways to escape from the captive portal:
- portalTimeout
- whileCaptivePortal exit
- Attempts reconnect with background the handleClient loop.

I think that both of the issues you have presented are likely to be addressed by the above features. After all, how does AutoConnect, which only works passively, know what caused the loss of the access point connection? (i.e. automatically extending the connection timeout)

> The desired behavior should be the ESP to start in SoftAP mode and then to connect to an already known WiFi network when becomes available again,

I think this can be implemented by the user own code using whileCaptivePortal exit.
I'm trying to figure out if your request is consistent with the current specification, but the conditions for reconnection are still unclear.

zener79, 2022-01-01:
@Hieromon maybe I'm missing something or maybe I don't know enough about the AutoConnect internals, but  this is how I would expect the device to work in the following scenarios:

**1. Not previous known networks is present when booting the device, and not previous known networks are broadcasted afterwards:** (this already works properly)

The device is not able to connect to any WiFi so it will start the SoftAp/CaptivePortal mode as soon as possible and stays in the `begin()`.

**2. A previous known networks is present when booting the device:**  (this already works properly)

The device connects to the WiFi and skips the SoftAp/CaptivePortal mode exiting the  `begin()`

**3. Not previous known networks is present when booting the device, but a previous known network is broadcasted afterwards (for example 3 minutes or 5 minutes):**

The device is not able to connect to any WiFi so it will start the SoftAp/Captive Portal as soon as possible and stays in the `begin()`. After 3 minutes, when a previous WiFi network is detected (this should be implemented inside the `begin()`), it exits the the `begin()` and close the SoftAp/CaptivePortal.

If point 3 can be already be obtained by the user own code using whileCaptivePortal exit, it should probably be documented as I really think this is the default expected behavior for most of the users.

Hieromon, 2022-01-01:
@zener79 Hmm, like this? I think we can implement it with whileCaptivePortal exit...

1. Declare a flag for AP detection as `inDetection = false`.
2. Prepare `detectAP` function as a `whileCaptivePortal` exit routine.
    - Measure the elapsed time since the first whileCaptivePortal exit call.
    - If the elapsed time does not exceed about 10 seconds then return `true`.
    - If `inDetection` is true then check to see if the `WiFi.scanNetworks` is finished.
        - If `WiFi.scanNetworks` is not finished then return `true`.
        - Match detected SSID with AutoConnectCredential.
        - If a matching SSID is found then return `false` to exit the captive portal loop.
        - return `true`.
    - Start `WiFi.scanNetworks` with async mode, and set `inDetection` to `true`.
    - return `true`.
3. Rush into the `portal.begin`

Do I need to incorporate it into the AutoConnect library standard function? I'm wary of conflicting functionality with other reconnect specifications. In particular, I think the relationship with `portalTime` setting will get complicated.

How do you think?

Hieromon, 2022-01-02:
@zener79 I have implemented the above proposal as follows, so could you consider this?
This is the case of ESP32, but the basic method is the same even with ESP8266.
You'll need to adjust the two wait times `scanInterval ` and `AutoConnecctConfig::reconnectInterval`, maybe.

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>

AutoConnect portal;
AutoConnectConfig config;

unsigned long tmDetection;
const unsigned long scanInterval = 10 * 1000;

// Known AP detection while in the captive portal
bool detectAP(void) {
  // Probing the results of the background WiFi scan.
  int16_t  ns = WiFi.scanComplete();

  if (ns == WIFI_SCAN_RUNNING) {
    // The scan is still running, nothing to do.
  }
  else if (ns == WIFI_SCAN_FAILED) {
    // The scan will start in the background where it has not started.
    // But it will wait for scanInterval from the last scan
    if (millis() - tmDetection > scanInterval) {
      WiFi.disconnect();
      WiFi.scanNetworks(true, true, false);
    }
  }
  else {
    // Now that the scan is finished, you can start checking if any known APs
    // have been detected.
    Serial.printf("scanNetworks:%d\n", ns);

    int16_t  scanResult = 0;
    while (scanResult < ns) {
      AutoConnectCredential cred;
      station_config_t  staConfig;

      if (cred.load(WiFi.SSID(scanResult++).c_str(), &staConfig) >= 0) {
        // Successfully detected a known AP, aborting captive portal,
        // Relying on autoReconnect and reconnectInterval to reconnect.
        Serial.printf("AP %s ready\n", (char*)staConfig.ssid);
        WiFi.scanDelete();
        return false;
      }
    }

    Serial.println("No found known AP");
    WiFi.scanDelete();
    tmDetection = millis();
  }

  return true;
}

void setup() {
  Serial.begin(115200);
  Serial.println();

  config.autoReconnect = true;
  config.reconnectInterval = 2;

  portal.onDetect([](IPAddress& apip) {
    Serial.println("Captive portal started, Start detection in the background");
    tmDetection = millis();
    return true;
  });
  portal.whileCaptivePortal(detectAP);

  portal.config(config);
  portal.begin();
}

void loop() {
  portal.handleClient();
}
```