Hieromon/AutoConnect #441: Weird behaviour after update from v1.1.7

zener79, 2022-01-02:
Hello,
I'm running on v1.1.7 from more than a year and have a working sketch in production. Before updating the library I enabled the AC_DEBUG and here is the output:

```
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] /setup on hands
[AC] /delconn on hands
[AC] /delconnexecute on hands
[AC] /setupexecute on hands
[AC] Current:
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting....................................................................................................time out IP:0.0.0.0
[AC] 8 network(s) found
[AC] autoReconnect loaded:Benve(RECENT)
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(Benve,*********)
[AC] Connecting..established IP:192.168.0.104
[AC] http server started
```

I just found that the `WiFi.begin()` is timing out, but `autoReconnect` fixes the issue immediately after, so I never noticed this.

Now... updating the library to version > 1.2.0 things are getting weird.
When I boot the device I see, in an exactly alternating manner, this

```
[AC] /setup on hands
[AC] /delconn on hands
[AC] /delconnexecute on hands
[AC] /setupexecute on hands
[AC] Set persistance
[AC] Current:Benve
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()....................................................................................................timeout
[AC] autoReconnect, 8 network(s) found, Benve(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(Benve,*********) failed
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] SoftAP MySSID/12345678 Ch(1) IP:172.217.28.1 
```
or this:
 
 ```
[AC] /setup on hands
[AC] /delconn on hands
[AC] /delconnexecute on hands
[AC] /setupexecute on hands
[AC] Set persistance
[AC] Current:
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin() failed
[AC] autoReconnect, 8 network(s) found, Benve(RECENT) loaded
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(Benve,*********)..established IP:192.168.0.104
[AC] http server started

 ```
 switching version at each reboot.

Hieromon, 2022-01-03:
Make sure you use the latest version of the core and dependent libraries. Also, don't forget that there is a gap in the ESP32 core version between Arduino IDE and PlatformIO.
Also, if you need help, please don't just post "it doesn't work", post the information needed to diagnose the problem. You can find what you need in the documentation [FAQ](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository).
Please remember that this is a community for open source development. Not a user support center.

zener79, 2022-01-03:
Ok, sorry for the misunderstanding and many thanks for your effort in this project.
I'm perfectly fine on running v1.1.7 so I'm not asking for support.

I simply discovered a weird behavior switching from 1.1.7 to the last AutoConnect version (1.3.2) but I found the issue started from 1.2.0 so I assumed it might be worth investigating.

You can see a test PlatformIO repo here:
https://github.com/zener79/AC_DEBUG

By simply switching the AutoConnect version in platformio.ini you should be able to reproduce the issue.

Hieromon, 2022-01-03:
@zener79 
> espressif32@3.0.0
> PageBuilder@1.4.2
> On versions >= 1.2.0
> WiFi.begin() in autoReconnect fails

Huh? Is this issue a continuation of the background reconnect in while captive portal that I suggested in Discussons #440 ?

zener79, 2022-01-03:
Not exactly.
I was running 1.1.7 fine, except for the background reconnection issue while captive portal.

You provided an excellent solution using `whileCaptivePortal` in https://github.com/Hieromon/AutoConnect/discussions/440. 

Since `whileCaptivePortal` is not supported in 1.1.7, I updated the library and faced this weird behavior.

Hieromon, 2022-01-03:
@zener79 Ok, I understood.

1. So please validate with the following version combination. I used this combination to test Discussons#440.
- AutoConnect 1.3.2
- PageBuilder 1.5.2
- Espressif32 3.4.0 on PlatfromIO w/ ESP-IDF 4.3.1
- ESP32 Arduino Core 2.0.2 w/ Arduino IDE 1.8.19
- Espressif8266 on PlatfromIO w/ ESP8266 Arduino Core 3.0.2

2. Regarding the phenomenon that the first WiFi.begin alternates between immediate failure and timeout every time it restarts.
This is not happening in my environment with the version combinations mentioned above.

As you know, the SDK side of ESP-IDF stores the already connected access point information as the currently enabled station configuration. AutoConnect reads the information on the SDK side to determine if it needs to wait until the timeout, and if the information stored by the SDK is empty, AutoConnect will immediately fail to wait until the timeout. I think the phenomenon you encountered is an effect of this change.

However, as you can see from the log below, background reconnection during captive portal works regardless of the WiFi.begin failure factor.  The first is an immediate failure, and the second is after a reboot, waiting for the access point saved by the SDK and failing. Both work a background reconnection while the captive portal.

```
--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

[AC] Set persistance
[AC] Current:
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[E][WiFiSTA.cpp:221] begin(): connect failed!
[AC] WiFi.begin() failed
[AC] autoReconnect, 5 network(s) found failed
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0
[AC] SoftAP esp32ap/12345678 Ch(1) IP:172.217.28.1
Captive portal started, Start detection in the background
[AC] WebServer allocated
[AC] http server started
[AC] DNS server started
scanNetworks:5
No found known AP
scanNetworks:3
No found known AP
scanNetworks:6
No found known AP
scanNetworks:2
No found known AP
scanNetworks:5
No found known AP
scanNetworks:3
No found known AP
scanNetworks:3
No found known AP
scanNetworks:2
No found known AP
scanNetworks:3
No found known AP
scanNetworks:4
No found known AP
scanNetworks:2
No found known AP
scanNetworks:6
No found known AP
scanNetworks:3
No found known AP
scanNetworks:5
AP aiPon ready
[AC] autoReconnect running
[AC] 3 network(s) found
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(aiPon,67zjz1f9x685b) ch(1)...................established IP:172.20.10.8
[AC] DNS server stopped
[AC] Event<17> handler registered
[AC] aiPon credential saved
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

[AC] Set persistance
[AC] Current:aiPon
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin().....................................................................................................timeout
[AC] autoReconnect, 4 network(s) found failed
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0
[AC] SoftAP esp32ap/12345678 Ch(1) IP:172.217.28.1
Captive portal started, Start detection in the background
[AC] WebServer allocated
[AC] http server started
[AC] DNS server started
scanNetworks:4
No found known AP
scanNetworks:7
AP aiPon ready
[AC] autoReconnect running
[AC] 5 network(s) found
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin(aiPon,67zjz1f9x685b) ch(1)....................established IP:172.20.10.8
[AC] DNS server stopped
[AC] Event<17> handler registered
[AC] aiPon credential saved
```

 If you encounter the same problem again along the lines of the above combination, then we need to narrow down the conditions under which the problem occurs.

Hieromon, 2022-01-29:
It will be closed once there is no progress in the activity.
Please use the latest version of AutoConnect and then post again if the symptoms recur. At that time it will be opened again.

closed 2022-01-29 by Hieromon