Hieromon/AutoConnect #446: Improper work when WiFi repeater in WiFi network

Warlib1975, 2022-01-07:
When a WiFi repeater works in a WiFi network Autoconnect shows the repeater and the main WiFi AP with the same names. In this case, when I enter WiFi password Autoconnect thinks a lot of time and then returns the page with the error.

[image]

It happens when connecting via Android and Windows 11. 

As soon as I turn off the repeater everything starts working properly.  

Windows/Android/iOS works properly with a repeater in the WiFi network showing only one AP.

Hieromon, 2022-01-07:
I am also running a WiFi repeater with the same SSID, but I have not experienced the phenomenon you encountered.
Can you enable [AC_DEBUG](https://hieromon.github.io/AutoConnect/adothers.html#debug-print) and get a trace log? I'll diagnose what's going on AutoConnect::begin inside.
Also, attach the information needed for the diagnosis as indicated in the [FAQ](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository), please. I don't know if you are using ESP8266 or ESP32, and I also need the core version and AutoConnect version.

> error: "Web page not available."

This is not enough to diagnose the problem; either the WiFi connection is successful and the redirect to the success page is failing, or the WiFi connection is failing and the redirect to the failure page is also failing. Depending on that, there are different factors causing your problem.

> As soon as I remember, the same behavior when in the WiFi network works several WiFi access points with the same name. Windows/Android/iOS shows only a single AP.

If the BSSID is different, AutoConnect will recognize it as a different AP. This behavior is a specification for detecting SSIDs that can be reconnected automatically. WiFi.scanNetworks returns the name of the hidden SSID as blank, so forcing it to use the BSSID to check against the stored credentials.
https://hieromon.github.io/AutoConnect/adconnection.html#match-with-known-access-points-by-ssid

> Windows/Android/iOS works properly with a repeater in the WiFi network showing only one AP. It seems they are shown only the AP with the highest level of the signal.

Not exactly. For iOS, the WiFi network selection specification in iOS is more complex. iOS dynamically changes SSID selection depending on the score of the user's operation. 
Refs: https://support.apple.com/en-bw/HT202831

If you are using ESP32, the above behavior you mentioned is the same for ESP32 Arduino core 1.0.6 or later. AutoConnect does not intervene in this behavior, it is all ESP32 core WiFi library behavior. if SSID and password are explicitly specified, AutoConnect will execute WiFi.begin with just the given SSID and password, without using BSSID.
Refs:
https://github.com/espressif/arduino-esp32/commit/3253de87
https://hieromon.github.io/AutoConnect/adconnection.html#connects-depending-on-the-wifi-signal-strength

Anyway, please submit an AC_DEBUG trace and I'll try to diagnose the problem. There may be some factors that I have overlooked.

Warlib1975, 2022-01-07:
Hi, 
here is the screenshot of the web interface where the name of the original AP is shown (MoM) and repeater AP:
[image]

Hieromon, 2022-01-07:
It's normal AutoConnect behavior.

- For Windows 10
[image]

- For iOS 14.8
[image]

My concern is not the multiple listing of the same SSID. Even in the above environment, AutoConnect should be able to successfully connect using `WiFi.begin(SSID, PW)` if the SSID and PW are explicitly stated.

You mentioned that
> "Web page not available."

Did the ESP module have a successful WiFi connection when it happened? If not, could it be signal interference from the repeaters occupying the same channel? Neither ESP32 nor ESP8266 has a roaming function.

So do you take the AC_DEBUG trace?

Warlib1975, 2022-01-08:
Thank you for the fast replay. I'll recheck the problem and can't understand the circumstances. After turning off and on repeater, and resetting ESP32 WiFi password I can connect to the Autoconnect web page as usual. I'll do additional tests, maybe the problem was in the repeater or something with ESP32.

Hieromon, 2022-01-08:
I agree with your assigned 2.4Ghz band channel placement 1,9,13. Its arrangement is the least susceptible to interference. If you are using DHCP, I recommend that you build with a core debug level of 5 and check if the ESP32 can get an IP address after connecting.
Since I can't see the situation, all I can advise is the following:

1. About Wi-Fi repeater
Typically, the DHCP offer from the DHCP server is unicast. But some repeaters have a DHCP offer as a broadcast message. At this time, the ESP32 side may have missed the IP address offered by the repeater. It will be logged in the EDP-IDF, so you can determine what is happening.

2. Check voltage level being pressurized to GOIO0
It should be in the range 3V to 3.3V. If more pressure is applied, the reach of Wi-Fi will be narrowed.

I'm sure you have already checked the above, but just in case you haven't, I'll tell you.
This topic will remain open for a while until there is a progress report on the situation.

Warlib1975, 2022-02-12:
Hi. I haven't met with the situation again. I suppose the issue could be closed.
Thank you for the fast response.

closed 2022-02-12 by Warlib1975