Hieromon/AutoConnect #447: Disconnect page +++

ESP4Ever, 2022-01-19:
Hello!

You did a great job creating such an amazing lib, thank you!
I have few comments on its functionality:
- "Disconnect" page is a VERY dangerous thing for a remote-installed devices as it makes them completely unrecoverable remotely!
It even hasn't a confirmation as for "Reset" + it doesn't redirect to a home page prior to execution causing a second disconnect when the browser is reopened! Is there any way to customize it?

- The LED ticker option stays in fast-blinking mode after connection to WiFi is established until reset/restart

Hieromon, 2022-01-22:
@ESP4Ever Thank you for your interest in AutoConnect.

> "Disconnect" page is a VERY dangerous thing for remote-installed devices as it makes them completely unrecoverable remotely! It even hasn't a confirmation as for "Reset" + it doesn't redirect to a home page prior to execution causing a second disconnect when the browser is reopened! Is there any way to customize it?

I generally agree with your opinion. That's something I was unsure about early on in the design of AutoConnect.
We may need some exit routine for when a reset request occurs. 
For example:

```cpp
AutoConnect portal;

// An exit when a reset request is generated
int resetExit() {
  // Determine the criteria of resettability based on the execution status of the sketch.
  // Possible return values include rushing into to reset, redirect to a specific page, or ignore reset.
  // In that case, additional specifications such as allowing the redirect destination to be specified 
  // in AutoConnectConfig would be necessary.
  return N;
}

void setup() {
  portal.onReset(resetExit);  // We can trap the reset operation.
  portal.begin();
}
```

If you can suggest the desired behavior, I would be willing to work on the above feature enhancements.
Can you help me?

> The LED ticker option stays in fast-blinking mode after connection to WiFi is established until reset/restart

This behavior is strange. Please let me know your AutoConnectConfig settings and ESP module type. I will start to diagnose the problem.

Hieromon, 2022-01-29:
@ESP4Ever I have verified the Ticker option again with the ESP8266, and it does not cause the problem you encountered. It works as [specified](https://hieromon.github.io/AutoConnect/adothers.html#ticker-for-wifi-status).

Also, regarding Disconnect, I considered a Disconnect exit but decided that the use case for that exit is currently rare. The reasons are as follows:

1. AutoConnectConfig has an [option setting](https://hieromon.github.io/AutoConnect/menu.html#applying-the-active-menu-items) to dynamically enable and disable the built-in menu. Therefore, ESP4Ever's point can be avoided with the AutoConnect option.
2. It also has a [setting](https://hieromon.github.io/AutoConnect/adconnection.html#automatic-reconnect-background) that if you lose WiFi connection while running the sketch, it will recover it as much as possible.
3. **Reset** is prompted, and if you don't want to Reset, you can avoid Reset by tapping (clicking) outside the area of the button.

Therefore, the points made on this topic are inadequate, and this is not an issue, so it is closed. You can check the documentation and if you still have questions, please go to the Discussion.

Thank you for your contribution.

closed 2022-01-29 by Hieromon