Hieromon/AutoConnect #449: Redirect to Station/DHCP IP after Successful Connection (_ac/success)

KungFuHaiku, 2022-01-21:
Hi there,

First up, thanks for such a feature-packed library; it's my first foray into ESP32 programming and I've managed to get everything up and running.

My question is whether it's possible, after making a successful connection, to redirect to the newly acquired IP for the ESP32's webserver (so, redirecting from the softAP 172.217.28.1 to whatever my DHCP has assigned to the ESP32, e.g. 192.168.0.29).  This is where the main sketch's GUI will reside.

I notice that after a successful connection on my iPhone, the resulting URL is _.../_ac/success_.  I was thinking of perhaps introducing some jQuery/JavaScript to this page in order to pick up the DOM element in the table, set it as a new URL for an arbitrary element, and then perform a simple redirect.

Is this something that's already available in the API or configuration?  If not, could I possibly ask where I would modify/inject additional HTML into the _/_ac/success_ page (or even on the main AutoConnect _/_ac_ page).

Many thanks.

closed 2022-01-22 by KungFuHaiku

Hieromon, 2022-01-22:
@KungFuHaiku I don't know why you closed this topic, but just in case, I'll describe the followings:

1. Change `/_ac/success`, which is the redirection destination for successful connections
    You can change it at the build time. Replace the `AUTOCONNECT_URI_SUCCESS` macro definition.
https://github.com/Hieromon/AutoConnect/blob/be996858e024cb5d58c9d4ef81e1d3965261bff4/src/AutoConnectDefs.h#L138
    By using PlatformIO, you can replace it without modifying the master library code. Describe it into platformio.ini.
    ```ini
    build_flgas = -DAUTOCONNECT_URI_SUCCESS='"/YOUR_PAGE_PATH"'
    ````
2.  Navigate the connected IP address which assigned by DHCP
    I can implement it, but doing so will make you unable to reach the ESP module from your iPhone.
    Your iPhone resides in the SoftAP network created by the ESP module. It is a different network segment from the AP that the ESP module is connected to. This means that your iPhone needs to leave the ESP module's AP and instantly switch to the AP connected by the ESP module. iOS can't do that trick. There will be a time lag of at least a few seconds.

KungFuHaiku, 2022-01-24:
Hi Hieromon,

Thanks for your prompt response.  I closed the issue as I realised I should've posted it under Discussions; I'm new (as a user) to Github too, so apologies for that.

As to your solutions, I'm just starting to understand the caveats with switching between WiFi connections.  When the **_ac/success** page is shown, I was wrongly assuming that the connection to the newly-discovered WiFi network (station) would be active.  I now see that this is not the case as smartphones do not support multiple simultaneous connections.

So, I guess I need to find the most graceful method to do the handover from SoftAP IP to Network IP.  I'll look into your suggestions.

Thanks again.

ラッセル

Hieromon, 2022-01-24:
@KungFuHaiku I know exactly what you want. In the early days of AutoConnect design, I experimented a lot with trying to navigate a device that opened a captive portal to the access point that the ESP module was connected to after the ESP module switched to station mode.
This can be done in a pseudo manner by redirecting the request arriving at SOFTAP to the STA side inside the ESP module. The STA and SOFTAP of the ESP module are independent and do not interfere with each other, so the bridge between requests to the SOFTAP and requests to the STA must be done by the application program. It is already a gateway.

But the next problem arises.
1. The code size for the pseudo-gateway is too large to fit on the ESP8266 with AutoConnect.
2. It directly intervenes in the TCP stack of the ESP32 or ESP8266 core, so the processing time of the interrupt handler will be excessive and may be trapped by the watchdog timer.

So this problem remains unresolved to this day.
The ESP module SOFTAP is mainly designed to fit into the [ESP-NOW](https://www.espressif.com/en/news/ESP-NOW) and [ESP-MESH](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-wifi-mesh.html) low-power wireless mesh networks. Therefore, it is not fully functional as a pseudo-access point to open a captive portal.

Your question may be true for other users as well. I would like to move this topic to [Discussions](https://github.com/Hieromon/AutoConnect/discussions) as I want other users to know the characteristics of ESP8266 and ESP32. Maybe we can expect other users to show us some good solutions. What do you think?

KungFuHaiku, 2022-01-24:
@Hieromon That sounds like a great idea.  I imagine it'll help to explain some basic concepts for those new to the world of developing WiFi apps.  It can be a bit overwhelming with such a huge number of libraries out there that essentially do the same thing, and it tends to pull you into a rabbit-hole.

Via your library (and excellent documentation) I've also learned about captive portals, creating a simple web server, and SPIFFS.  I've subsequently worked out how to serve up static content from the ESP32, along with HTML pages that use AJAX to avoid page reloads (and circumvent the compatibility issues with AsyncWebServer).

Your out-of-box solution got me up and running very quickly and acted as a gateway (pun intended) to further investigation and learning.  However, there were a couple of fundamental concepts/pitfalls that I couldn't wrap my head around, so yes, I'm all up for putting it out there and seeing what comes back.

Hieromon, 2022-01-25:
@KungFuHaiku Ok, this topic will be continued in Discussions.