Hieromon/AutoConnect #451: Problem in start page configuration mqttRSSI_FS.ino

Bighoneypot, 2022-01-24:
Gday, I try to use mqttRSSI_FS.ino, but in configuration page there is a error.

Mqtt settings page have a problem.

[image]

Bighoneypot, 2022-01-24:
@Hieromon 
I found this

[image]

Hieromon, 2022-01-24:
@Bighoneypot The `mqttRRI_FS.ino` has been tested and verified. Usually, such problems do not occur. Either the `para.json` in the `data` folder is corrupted, or the value you entered does not fit as a JSON string.

Bighoneypot, 2022-01-24:
@Hieromon 
Dear, my mistake, this is mqttRSSI_NA

Hieromon, 2022-01-24:
@Bighoneypot Well you mean that initial values is not set. I knew this, and thought it wouldn't be a big problem, so I left it at that. But as you've encountered, it may destroy the HTML.
I will fix it. Thank you for finding the problem, also your contribution.

Bighoneypot, 2022-01-24:
@Hieromon 
It is a pleasure for me to contribute to this work. thank you

Hieromon, 2022-01-25:
@Bighoneypot I fixed the issue and commit to the master. Can you please verify in your environment that the fix is correct?
I have not released it yet, so in your verification, please download the master branch and extract it to your Arduino library folder.
I will release the latest HEAD as v1.3.3 once it is confirmed to work correctly.

Bighoneypot, 2022-01-25:
@Hieromon 
The problem of html code corruption for this moment seems to work.
I tried with an esp8266, I upload the firmware and write all the data in the captive portal, but if I disconnect the USB cable and reconnect it, the board does not have the channel configuration saved, but only the parameters of the Wifi connection. The Thingspeak channel configuration is cleared.

Hieromon, 2022-01-25:
@Bighoneypot Thank you for your testing. I will try to reproduce the phenomenon on my end, so please tell me more about your procedure. I have confirmed that the following steps work fine. What is the difference between your procedure and mine?

1. Use esptools to erase all flash are of the ESP8266 once. This will clear all of the EEPROM area, as well as AutoConnect's stored credentials and Thingspeak's channel settings.
2. Build and upload the `mqttRSSI_NA.ino` sketch.
3. AutoConnect does not have any known credentials, so the captive portal starts. Then open the captive portal from cell phone and navigate to `MQTT Setting` from the AutoConnect menu.
4. It will open the *MQTT broker settings* page where each setting value is blank. Enter the configuration values and tap `Start`.
5. It will open the confirmation page, then tap `Clear channel`.
6. Back to the chart page of Thingspeak, the ESP8266 is not connected to any access point yet, so you cannot see the chart.
7. Tap the cog icon to go to the AutoConnect menu. Select Configure new AP from the menu to start connecting to the access point.
8. After the ESP8266 has successfully connected to the access point, unplug the USB cable.
9. Plug in the USB cable again, the ESP8266 will auto-reset and boot.
10. The ESP8266 will automatically connect to the access point and start MQTT publishing.
11. Access the IP address of the ESP8266 from the cell phone's browser.
12. Tap the cog icon and navigate to the AutoConnect menu.
13. Go to `MQTT Setting` menu, open the *MQTT broker settings*. The entire settings are still there.

The `mqttRSSI_NA.ino` saves Thingspeak channel settings in the EEPROM area. That area usually overlaps with the AutoConnect credentials storage area. Therefore, for sketches that use the EEPROM area independently, it is necessary to notify AutoConnect of the existence of the EEPROM area for storing the sketch-specific data by setting [AutoConnectConfig::boundaryOffset](https://hieromon.github.io/AutoConnect/adcredential.html#move-the-saving-area-of-eeprom-for-the-credentials).
The mqttRSSI_NA.ino specifies the [AutoConnectConfig::boundaryOffset](https://hieromon.github.io/AutoConnect/adcredential.html#move-the-saving-area-of-eeprom-for-the-credentials), but in sketches lacking it, AutoConnect writes the credentials from the beginning of the EEPROM area, so the user's EEPROM data will lost.

Hieromon, 2022-01-25:
@Bighoneypot I don't think deep sleep is relevant. Reading from EEPROM is performed before the `portal.begin()`, so if ESP8266 can connect to WiFi, then `portal.begin()` has been completed successfully, and it can read the channel settings from EEPROM successfully.

https://github.com/Hieromon/AutoConnect/blob/87e510a22f18db69674ca28c0de9faeda3c6b46f/examples/mqttRSSI_NA/mqttRSSI_NA.ino#L289-L293

Bighoneypot, 2022-01-25:
> @Bighoneypot Thank you for your testing. I will try to reproduce the phenomenon on my end, so please tell me more about your procedure. I have confirmed that the following steps work fine. What is [...]
I already use esptool from cmd line(python esptool.py -p com6 erase_flash and  andpython esptool.py -p com6 erase_region 0x3FC000 0x4000) and from ESPTool
> 2. Build and upload the `mqttRSSI_NA.ino` sketch.
Ok
> 3. AutoConnect does not have any known credentials, so the captive portal starts. Then open the captive portal from cell phone and navigate to `MQTT Setting` from the AutoConnect menu.
Ok
> 4. It will open the _MQTT broker settings_ page where each setting value is blank. Enter the configuration values and tap `Start`.
Ok
> 5. It will open the confirmation page, then tap `Clear channel`.
Ok
> 6. Back to the chart page of Thingspeak, the ESP8266 is not connected to any access point yet, so you cannot see the chart.
Ok
> 7. Tap the cog icon to go to the AutoConnect menu. Select Configure new AP from the menu to start connecting to the access point.
Ok
> 8. After the ESP8266 has successfully connected to the access point, unplug the USB cable.
Ok
> 9. Plug in the USB cable again, the ESP8266 will auto-reset and boot.
Ok
> 10. The ESP8266 will automatically connect to the access point and start MQTT publishing.
Esp will automatically connect but all configuration value are not present
> 11. Access the IP address of the ESP8266 from the cell phone's browser.
Impossible to find esp8266ap in wifi connection
> 12. Tap the cog icon and navigate to the AutoConnect menu.
Impossible, I see in terminal my Serial print debug, esp8266 is connect but configuration data are not present (look a this image)
[image]

> 13. Go to `MQTT Setting` menu, open the _MQTT broker settings_. The entire settings are still there.
Impossible
> The `mqttRSSI_NA.ino` saves Thingspeak channel settings in the EEPROM area. That area usually overlaps with the AutoConnect credentials storage area. Therefore, for sketches that use the EEPROM area [...]

Bighoneypot, 2022-01-25:
I did not understand how to use in this sketch AutoConnectConfig::boundaryOffset

Hieromon, 2022-01-25:
@Bighoneypot Did you test mqttRSSI_NA without modification from the library example? If the problem reappears without modification, it may be that something is lurking in your environment. We first need to isolate the problem.

1. Access to the ESP8266

> Access the IP address of the ESP8266 from the cell phone's browser.
> Impossible to find esp8266ap in wifi connection

The serial monitor will show the IP address of the ESP8266 after WiFi connection. Can you use that to access it?

2. Library dependency

https://user-images.githubusercontent.com/35570068/151003151-56daa40d-ddf7-444f-92f0-8596cce181b0.png

There may be a library conflict. The correct dependencies are as follows:
```
Dependency Graph
|-- <AutoConnect> 1.3.2
|   |-- <PageBuilder> 1.5.2
|   |   |-- <ESP8266WebServer> 1.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <LittleFS> 0.1.0
|   |-- <ArduinoJson> 6.19.1
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <LittleFS> 0.1.0
|   |-- <Ticker> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266httpUpdate> 1.3
|   |   |-- <ESP8266HTTPClient> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <SD> 2.0.0
|   |   |-- <SDFS> 0.1.0
|   |   |   |-- <ESP8266SdFat> 2.0.2
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <ESP8266HTTPClient> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <PubSubClient> 2.8
|-- <EEPROM> 1.0
```

3. Dump EEPROM area

First, I recommend you check whether the data has been written correctly in the EEPROM area. Dumping the EEPROM may give you clues. You can easily dump the EEPROM area using [ESPShaker](https://github.com/Hieromon/ESPShaker).

```
ESPShaker 1.4.3 - 00d4e700,Flash:4194304,SDK2.2.2-dev(38a443e),core3.0.2
Type "help", "?" for commands list.
WiFi.mode(OFF)
WiFi.Status(WL_DISCONNECTED)

> ?

ESPShaker 1.4.3 - 00d4e700,Flash:4194304,SDK2.2.2-dev(38a443e),core3.0.2
Commands:
apconfig [AP_IP] [GW_IP] [NETMASK]
autoconnect on|off
[... 82 lines cut ...]
AutoConnectConfig config;

// Defines the custom data should be stored in EEPROM.
typedef struct {
  char  data1[8];
  char  data2[8];
  char  data3[8];
} EEPROM_CONFIG_t;
EEPROM_CONFIG_t eepromConfig;
...
// Declares to reserve the EEPROM_CONFIG_t area for a Sketch using.
config.boundaryOffset = sizeof(eepromConfig);
portal.config(config);
...
strcpy(eepromComfig.data1, "data1");
strcpy(eepromComfig.data2, "data2");
strcpy(eepromComfig.data3, "data3");

// Use getEEPROMUsedSize to access the EEPROM with the appropriate region size.
EEPROM.begin(portal.getEEPROMUsedSize());
EEPROM.put<EEPROM_CONFIG_t>(0, eepromConfig);
EEPROM.commit();
EEPROM.end();
...
```

Bighoneypot, 2022-01-25:
@Hieromon 
YOU Write this:
Access to the ESP8266
Access the IP address of the ESP8266 from the cell phone's browser.
Impossible to find esp8266ap in wifi connection.

I'm

Captive portal is non present because because at the second restart it has the wifi connection data in memory so it is connected to the internet.
Instead the connection data for the thingspeaks channel is deleted.

Why don't you do a function for captive portal open mode when writeapikey, user apikey or publish interval are Zero.
In this way, if the rom data is deleted, the captive portal opens again and is visible in the WiFi networks of the smartphone

Hieromon, 2022-01-25:
@Bighoneypot 
It's a little strange. You are viewing the Thingspeak channel settings on your serial monitor. However, you do not know the IP address displayed by mqttRSSI_NA.ino. Why?

> Instead the connection data for the thingspeaks channel is deleted.

Which sketch you used? Have you encountered this phenomenon with mqttRSSI_NA.ino? Or is it your sketch with some kind of modification?
I don't see this phenomenon in mqttRSSI_NA.ino in my environment.

> Why don't you do a function for captive portal open mode when writeapikey, user apikey or publish interval are Zero.
In this way, if the rom data is deleted, the captive portal opens again and is visible in the WiFi networks of the smartphone

I can not see its usefulness. User data is stored in the EEPROM area whether it is under the Captive Portal or not. This is what the AutoConnectConfig::boundaryOffset w/AutoConnect::getEEPROMUsedSize API is for.

Hieromon, 2022-01-25:
The initial value leak in mqttRSSI_NA.ino seems to have been fixed, so I consider this problem to be fixed. v1.3.3 will be released.

Bighoneypot, 2022-01-25:
@Hieromon 

ESPshaker:
```
18:16:33.300 -> 
18:16:33.300 -> > 
18:16:54.651 -> > ESPShaker 1.4.3 - 0033dc01,Flash:2097152,SDK2.2.2-dev(38a443e),core3.0.2
18:17:08.396 -> Commands:
18:17:08.396 -> apconfig [AP_IP] [GW_IP] [NETMASK]
18:17:08.396 -> autoconnect on|off
18:17:08.396 -> begin [SSID [PASSPHRASE]] [#wait]
18:17:08.396 -> config {IP GW [NETMASK] [DNS1] [DNS2]}|erase
18:17:08.396 -> delay MILLISECONDS
18:17:08.396 -> discon [ap]
18:17:08.396 -> eeprom {addr [ADDRESS]}|{clear BYTE LENGTH}|{write DATA}|{read LENGTH}
18:17:08.396 -> event on|off
18:17:08.396 -> fs start|dir|{file PATH}|format|info|{remove PATH}|{rename PATH NEW_PATH}
18:17:08.396 -> gpio {get PORT_NUM}|{set PORT_NUM high|low}
18:17:08.396 -> heap 
18:17:08.396 -> hostname [HOSTNAME]
18:17:08.396 -> http {get url}|{on uri PAGE_CONTENT}
18:17:08.442 -> mode ap|sta|apsta|off
18:17:08.442 -> mqtt {server SERVER [PORT]}|{con CLIENT_ID AUTH PASS}|{pub {TOPIC PAYLOAD [#r]}}|{sub TOPIC [QoS]|stop}|close
18:17:08.442 -> persistent on|off
18:17:08.442 -> ping REMOTE_HOST
18:17:08.442 -> reset 
18:17:08.442 -> scan 
18:17:08.442 -> sleep none|light|modem|{deep SLEEP_TIME}
18:17:08.442 -> smartconfig start|stop|done
18:17:08.442 -> show 
18:17:08.442 -> softap {SSID [PASSPHRASE] [CHANNEL]}|discon
18:17:08.442 -> start web|{dns [DOMAIN]}|{mdns HOST_NAME SERVICE PROTOCOL [PORT]}
18:17:08.442 -> station 
18:17:08.442 -> status 
18:17:08.442 -> stop dns|web
18:17:08.442 -> wps 
18:17:08.442 -> > 
18:17:29.486 -> > eeprom addr:0x000, read:
18:17:41.304 -> 000  00 00 00 00 20 64 6e 73  7c 77 65 62 00 00 00 00  .... dns |web....
18:17:41.304 -> 010  5d 7d 7c 7b b4 13 36 01  00 00 00 00 20 64 6e 73  ]}|{⸮.6. .... dns
18:17:41.304 -> 020  7c 77 65 62 00 00 00 00  43 45 20 50 b4 13 39 01  |web.... CE P⸮.9.
18:17:41.304 -> 030  00 00 00 00 5b 50 4f 52  54 5d 7d 00 00 00 00 00  ....[POR T]}.....
18:17:41.350 -> 040  53 53 49 44 b4 13 36 01  73 74 61 72 74 20 77 65  SSID⸮.6. start we
18:17:41.350 -> 050  62 7c 7b 64 6e 73 20 5b  44 4f 4d 41 49 4e 5d 7d  b|{dns [ DOMAIN]}
18:17:41.350 -> 060  7c 7b 6d 64 6e 73 20 48  4f 53 54 5f 4e 41 4d 45  |{mdns H OST_NAME
18:17:41.350 -> 070  20 53 45 52 56 49 43 45  20 50 52 4f 54 4f 43 4f   SERVICE  PROTOCO
18:17:41.350 -> 080  4c 20 5b 50 4f 52 54 5d  7d 00 00 00 00 00 00 00  L [PORT] }.......
18:17:41.350 -> 090  00 00 00 00 00 00 00 00  20 43 4c 49 b4 13 3f 01  ........  CLI⸮.?.
18:17:41.350 -> 0a0  36 01 00 00 74 20 77 65  62 7c 7b 64 6e 73 20 5b  6...t we b|{dns [
18:17:41.396 -> 0b0  44 4f 4d 41 49 4e 5d 7d  7c 7b 6d 64 6e 73 20 48  DOMAIN]} |{mdns H
18:17:41.396 -> 0c0  4f 53 54 5f 4e 41 4d 45  20 53 45 52 56 49 43 45  OST_NAME  SERVICE
18:17:41.396 -> 0d0  20 50 52 4f 54 4f 43 4f  4c 20 5b 50 4f 52 54 5d   PROTOCO L [PORT]
18:17:41.396 -> 0e0  7d 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  }....... ........
18:17:41.396 -> 0f0  00 00 00 00 b4 13 4a 01  00 00 36 01 00 00 00 00  ....⸮.J. ..6.....
18:17:41.396 -> 100  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.396 -> 110  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.443 -> 120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.443 -> 130  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.443 -> 140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.443 -> 150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.443 -> 160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.443 -> 170  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.443 -> 180  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.443 -> 190  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.489 -> 1a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.489 -> 1b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.489 -> 1c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.489 -> 1d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.489 -> 1e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
18:17:41.489 -> 1f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ........ ........
[... 60 lines cut ...]
|   |   |-- <EEPROM> 1.0
|   |   |-- <Ticker> 1.0
|   |   |-- <ESP8266HTTPClient> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266httpUpdate> 1.3
|   |   |   |-- <ESP8266HTTPClient> 1.2
|   |   |   |   |-- <ESP8266WiFi> 1.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <DNSServer> 1.1.1
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM> 1.0
|-- <Lifely Agrumino Lemon> 1.4.4
|   |-- <EEPROM> 1.0
|   |-- <Wire> 1.0
|-- <ESP8266HTTPClient> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
Building in release mode
```

**Sorry I don't understand how to implement and configure AutoConnect :: getEEPROMUsedSize in the sketch**

Hieromon, 2022-01-25:
@Bighoneypot First of all, please answer the following questions so that I can capture the phenomenon you are encountering.

1. Did you run mqttRSSI_NA.ino without modifying it, or did the disappearance of the channel settings for Thingspeak occur in the mqttRSSI.ino sample sketch provided by the AutoConnect library?

2. If you modify it, is your sketch making additional writes to the EEPROM area? The meaning of "addition" includes the following modifications:
    - Changed `mqtt_param_t` structure in `param.h`
    - Writing to the EEPROM area without using `mqtt_param_t`
    - mqttRSSI_NA.ino uses the loadParams and saveParams functions to access the EEPROM area. Modifications other than these functions to access the EEPROM area.

3. In the dependency graph you posted, `<param>` is shown at the top level. Is this the latest `param.cpp` and `param.h` shipped with AutoConnect example's `mqttRSSI_NA`?, And `<param>` refers to AutoConnect. Why is this?

4. The dump result you posted shows the sketch itself. According to the [ESP8266 Arduino core documentation](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html?highlight=eeprom#flash-layout), the EEPROM area of your ESP8266 module is out of alignment. I have no idea how this could happen. Does the board you are actually using match the board specified in the build system?

So, see the code snippet that I posted [above](https://github.com/Hieromon/AutoConnect/issues/451#issuecomment-1021330163). If a user sketch uses its own EEPROM area, that procedure is usually the norm. The `mqttRSSI_NA.ino` also follows the same procedure, but users require an understanding of the [EEPROM library](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html?highlight=eeprom#eeprom) API provided by the ESP8266 Arduino core. You also need to know the flash layout described in the ESP8266 documentation I mentioned [above](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html?highlight=eeprom#flash-layout). Without it, you will not be able to understand AutoConnect::getEEPROMUsedSize and AutoConectConfig::boundaryOffset.

Bighoneypot, 2022-01-26:
> @Bighoneypot First of all, please answer the following questions so that I can capture the phenomenon you are encountering.
>
> 1. Did you run mqttRSSI_NA.ino without modifying it, or did the [...]

@Hieromon 
Gday, the problem is when I try  to build and upload my firmware with platformio. I try this with Arduino Ide and problem are not present. I use my board Agrumino Lemon with my libraries, but, when i choice my board in Platformio, Platformio send me error.
Also, Arduinio Ide don't have problem with my board in board manager. Arduino install my board correctly.

Bighoneypot, 2022-01-26:
I find other problem, but I open other Issue.
Wait a 2 day to close this issue, thank

Bighoneypot, 2022-02-04:
@Hieromon bug is fixed, thank you. I close this Issue

closed 2022-02-04 by Bighoneypot