Hieromon/AutoConnect #476: AutoConnect captive portal shows menu bar but empty/blank body on ESP32

scoover, 2022-03-27:
I am trying to get AutoConnect to run on a Adafruit ESP32 Feather, but my captive portal looks like this.
[image]

When I connect and authenticate to the AutoConnect portal, I see the menu bar but the body appears empty (even after clicking "Configure New AP"). I can see that there are four invisible SSIDs hidden in the body; they can be unhidden when I disable the `.lap` css property `visibility: collapse;`. It appears that the scan is working but some code or overlay is preempting the captive portal UX.

Here is my setup:
- Adafruit ESP32 Feather
- Arduino IDE 1.8.19 
- AutoConnect 1.3.4
- PageBuilder 1.5.3

Miscellaneous notes:
- I can access the root page so the webserver seems to be working.
- I have tried the captive portal "Reset...", but that doesn't seem to improve anything.
- I'm not sure if it is related, but my serial port shows the "Restarted..." message but never shows the subsequent messages at the end of the `setup()` function. I also see no additional debug information when I define `AC_DEBUG` and `PB_DEBUG`.
- Unrelated, I'd love to rename the AP's SSID and password but experimentation with `apPortal.begin("My SSID","1234")` has made no difference.

I am essentially just using the [recommended sample code](https://hieromon.github.io/AutoConnect/basicusage.html):
```
#include <WiFi.h>
#include <WebServer.h>    
#include <AutoConnect.h>

WebServer webserver;
AutoConnect apPortal(webserver);

void rootPage(){
  char content[] = "Hello world!";
  webserver.send(200, "text/plain", content);
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println("Restarted...");
  
  webserver.on("/", rootPage);

  if (apPortal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  }
  else {
    Serial.println("Failed to start AutoConnect...");
  }
}

void loop() {
  apPortal.handleClient();
}
```

Thanks for this project and thanks for your help!

Hieromon, 2022-03-29:
Instead of extracting only a part of the debug trace of the ESP32 core, please present the AutoConnect debug trace output with `AC_DEBUG` enabled.
Also, if you can use multiple browsers (eg on other cellphones), I recommend trying access from that device's browser. This attempt eliminates the possibility that AutoConnect body content HTML is being hampered by browser extensions.

scoover, 2022-04-22:
Thanks for getting back to me, @Hieromon. I enabled the `AC_DEBUG` logging and this is what I saw when I:
1. Started the ESP32
2. Connected to `esp32ap`
3. Clicked the Open SSID tab

Please let me know if something jumps out at you. I'd love to get this working.

Thanks for supporting this project.
Randy

```
17:19:23.385 -> [AC] Set persistance
17:19:23.495 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
17:19:23.495 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
17:19:23.606 -> [AC] Current:
17:19:23.606 -> [E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
17:19:23.606 -> [AC] Preferences begin failed to import AC_CREDT
17:19:23.606 -> [AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
17:19:23.606 -> [E][WiFiSTA.cpp:220] begin(): connect failed!
17:19:23.606 -> [AC] WiFi.begin() failed
17:19:23.606 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
17:19:23.713 -> [AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
17:19:23.823 -> [AC] SoftAP esp32ap/12345678 Ch(1) IP:172.217.28.1 
17:19:23.823 -> [AC] http server started
17:19:23.823 -> [AC] DNS server started
17:19:32.357 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
17:19:32.468 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
17:19:32.758 -> [V][WebServer.cpp:291] handleClient(): New client
17:19:32.758 -> [V][Parsing.cpp:121] _parseRequest(): method: GET url: /hotspot-detect.html search: 
17:19:32.758 -> [AC] Host:,/hotspot-detect.html,ignored
17:19:32.758 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Host
17:19:32.758 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: captive.apple.com
17:19:32.758 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Connection
17:19:32.794 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: close
17:19:32.794 -> [V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
17:19:32.794 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: CaptiveNetworkSupport-428.0.0.0.1 wispr
17:19:32.794 -> [V][Parsing.cpp:255] _parseArguments(): args: 
17:19:32.794 -> [V][Parsing.cpp:238] _parseRequest(): Request: /hotspot-detect.html
17:19:32.794 -> [V][Parsing.cpp:239] _parseRequest():  Arguments: 
17:19:32.794 -> [E][WebServer.cpp:632] _handleRequest(): request handler not found
17:19:32.829 -> [AC] Detected application, captive.apple.com, 0.0.0.0
17:19:32.863 -> [V][WebServer.cpp:291] handleClient(): New client
17:19:32.863 -> [V][Parsing.cpp:121] _parseRequest(): method: GET url: /_ac search: 
17:19:32.863 -> [AC] Host:captive.apple.com,/_ac,generated:/_ac, allocated
17:19:32.863 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Host
17:19:32.900 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: 172.217.28.1
17:19:32.900 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Connection
17:19:32.900 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: close
17:19:32.900 -> [V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
17:19:32.900 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: CaptiveNetworkSupport-428.0.0.0.1 wispr
17:19:32.900 -> [V][Parsing.cpp:255] _parseArguments(): args: 
17:19:32.937 -> [V][Parsing.cpp:238] _parseRequest(): Request: /_ac
17:19:32.937 -> [V][Parsing.cpp:239] _parseRequest():  Arguments: 
17:19:32.937 -> [AC] Host:172.217.28.1,/_ac,already allocated
17:19:33.406 -> [V][WebServer.cpp:291] handleClient(): New client
17:19:38.419 -> [V][WebServer.cpp:291] handleClient(): New client
17:19:38.419 -> [V][Parsing.cpp:121] _parseRequest(): method: GET url: /hotspot-detect.html search: 
17:19:38.419 -> [AC] Host:172.217.28.1,/hotspot-detect.html,ignored
17:19:38.419 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Host
17:19:38.419 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: captive.apple.com
17:19:38.419 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Upgrade-Insecure-Requests
17:19:38.454 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: 1
17:19:38.454 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Accept
17:19:38.454 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
17:19:38.454 -> [V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
17:19:38.454 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko)
17:19:38.490 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Accept-Language
17:19:38.490 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: en-US,en;q=0.9
17:19:38.490 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Accept-Encoding
17:19:38.490 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: gzip, deflate
[... 84 lines cut ...]
17:19:55.096 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Host
17:19:55.096 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: captive.apple.com
17:19:55.096 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Connection
17:19:55.096 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: close
17:19:55.133 -> [V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
17:19:55.133 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: CaptiveNetworkSupport-428.0.0.0.1 wispr
17:19:55.133 -> [V][Parsing.cpp:255] _parseArguments(): args: 
17:19:55.133 -> [V][Parsing.cpp:238] _parseRequest(): Request: /hotspot-detect.html
[... cut ...]
17:19:55.172 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: close
17:19:55.210 -> [V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
17:19:55.210 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: CaptiveNetworkSupport-428.0.0.0.1 wispr
17:19:55.210 -> [V][Parsing.cpp:255] _parseArguments(): args: 
17:19:55.210 -> [V][Parsing.cpp:238] _parseRequest(): Request: /_ac
17:19:55.210 -> [V][Parsing.cpp:239] _parseRequest():  Arguments: 
17:19:55.210 -> [AC] Host:172.217.28.1,/_ac,already allocated
```

Hieromon, 2022-04-22:
@scoover The above log is a little strange to me.
1. I can't see the PageBuilder's log lines (it is enabled with PB_DEBUG), has it been intentionally deleted from your post?
2. You mentioned that the body content is empty when you click "Configure New AP". But the above log does not have access history to `/_ac/config`, it has only `/_ac/open`. Does the same phenomenon occur with "Open SSIDs"?

Other than that, there is nothing odd about it. Is it still a problem on the browser side? You can try disabling all extensions on the browser side for once or If you own an iPhone, I would suggest that you verify if this phenomenon also occurs on the iPhone.

scoover, 2022-04-22:
Sorry about that. `PB_DEBUG` logging is now included at bottom. 

As before, this log includes:
1. Started the ESP32
2. Connected to esp32ap
3. Clicked the "Open SSIDs" tab

I have tried using the AP with the macOS 12.3.1 Wifi Manager, macOS Safari, macOS Chrome, iOS (latest). In all cases, the page body of the captive portal is empty below the menu bar (screenshot in initial post).

Also, as I mentioned in the initial post, if I inspect the HTML and, if I disable the `.lap` css property `visibility: collapse;`,  I can see that there are four invisible SSIDs hidden in the body. So, it appears the scan is working but some code or overlay is preempting display of available networks in the captive portal.

[image]

One other (unrelated) thing I have noticed... when I run the `webserver` through `AutoConnect` (versus with AutoConnect commented out), the server no longer uses my handler (code below) for endpoints not explicitly called out with `webserver.on()` (i.e. my code checks if a SPIFFS file is available for the URI before returning a 404). Is this an issue with AutoConnect overriding my `onNotFound()` function? Or with `AutoConnect.handleClient();` not servicing the web events the same way as `webserver.handleClient();`? Is there a workaround for this?

```
  webserver.onNotFound([]() {
    if (!webpageFromSPIFFS(webserver.uri()))
      webserver.send(404, "text/plain", "404: Not Found");
  });
```

Thanks again for your help.

```
09:55:46.305 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
09:55:46.305 -> configsip: 0, SPIWP:0xee
09:55:46.305 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:55:46.305 -> mode:DIO, clock div:1
09:55:46.305 -> load:0x3fff0018,len:4
09:55:46.305 -> load:0x3fff001c,len:1044
09:55:46.340 -> load:0x40078000,len:10124
09:55:46.340 -> load:0x40080400,len:5856
09:55:46.340 -> entry 0x400806a8
09:55:47.730 -> Restarted...
09:55:47.840 -> [AC] Set persistance
09:55:47.949 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
09:55:47.949 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
09:55:48.056 -> [AC] Current:
09:55:48.056 -> [E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND
09:55:48.056 -> [AC] Preferences begin failed to import AC_CREDT
09:55:48.056 -> [AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
09:55:48.094 -> [E][WiFiSTA.cpp:220] begin(): connect failed!
09:55:48.094 -> [AC] WiFi.begin() failed
09:55:48.094 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
09:55:48.168 -> [AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
09:55:48.273 -> [AC] SoftAP esp32ap/12345678 Ch(1) IP:172.217.28.1 
09:55:48.273 -> [AC] http server started
09:55:48.273 -> [AC] DNS server started
09:56:06.199 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
09:56:07.259 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
09:56:08.329 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
09:56:10.332 -> [V][WebServer.cpp:291] handleClient(): New client
09:56:10.332 -> [V][Parsing.cpp:121] _parseRequest(): method: GET url: /hotspot-detect.html search: 
09:56:10.370 -> [AC] Host:,/hotspot-detect.html,ignored
09:56:10.370 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Host
09:56:10.370 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: captive.apple.com
09:56:10.370 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Connection
09:56:10.370 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: close
09:56:10.370 -> [V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
09:56:10.370 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: CaptiveNetworkSupport-428.0.0.0.1 wispr
09:56:10.412 -> [V][Parsing.cpp:255] _parseArguments(): args: 
09:56:10.412 -> [V][Parsing.cpp:238] _parseRequest(): Request: /hotspot-detect.html
09:56:10.412 -> [V][Parsing.cpp:239] _parseRequest():  Arguments: 
09:56:10.412 -> [E][WebServer.cpp:632] _handleRequest(): request handler not found
09:56:10.412 -> [AC] Detected application, captive.apple.com, 0.0.0.0
09:56:10.594 -> [V][WebServer.cpp:291] handleClient(): New client
09:56:10.594 -> [V][Parsing.cpp:121] _parseRequest(): method: GET url: /_ac search: 
09:56:10.632 -> [AC] Host:captive.apple.com,/_ac,generated:/_ac, allocated
09:56:10.632 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Host
[... 118 lines cut ...]
09:56:25.816 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Connection
09:56:25.816 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: close
09:56:25.816 -> [V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
09:56:25.816 -> [V][Parsing.cpp:228] _parseRequest(): headerValue: CaptiveNetworkSupport-428.0.0.0.1 wispr
09:56:25.816 -> [V][Parsing.cpp:255] _parseArguments(): args: 
09:56:25.816 -> [V][Parsing.cpp:238] _parseRequest(): Request: /hotspot-detect.html
09:56:25.853 -> [V][Parsing.cpp:239] _parseRequest():  Arguments: 
09:56:25.853 -> [E][WebServer.cpp:632] _handleRequest(): request handler not found
09:56:25.853 -> [AC] Detected application, captive.apple.com, 0.0.0.0
09:56:25.853 -> [V][WebServer.cpp:291] handleClient(): New client
09:56:25.853 -> [V][Parsing.cpp:121] _parseRequest(): method: GET url: /_ac search: 
09:56:25.853 -> [AC] Host:captive.apple.com,/_ac,generated:/_ac, allocated
09:56:25.853 -> [V][Parsing.cpp:227] _parseRequest(): headerName: Host
[... cut ...]
09:56:25.922 -> [V][Parsing.cpp:238] _parseRequest(): Request: /_ac
09:56:25.922 -> [V][Parsing.cpp:239] _parseRequest():  Arguments: 
09:56:25.922 -> [PB] HTTP_GET /_ac
09:56:25.922 -> [AC] Host:172.217.28.1,/_ac,already allocated
09:56:25.922 -> [PB] Chunked, HEAD CSS_BASE CSS_TABLE blk:1270 CSS_LUXBAR blk:1270 blk:1270 MENU_PRE blk:1270 MENU_AUX MENU_POST ESTAB_SSID WIFI_MODE WIFI_STATUS LOCAL_IP GATEWAY NETMASK SOFTAP_IP AP_MAC STA_MAC CHANNEL DBM CHIP_ID CPU_FREQ FLASH_SIZE FREE_HEAP blk:1010 
```

Hieromon, 2022-04-23:
@scoover Thanks for providing the PB_DEBUG log.

I suspected HTML content generation was failing when I first saw your post. Therefore, we needed the PageBuilder logs. Looking at the logs, I can't confirm such a problem. However, the screenshots you posted are very rare for me.
The above screenshot shows multiple detections of the same SSID on different channels (e.g. Fashion IoT, Fashion Seal). Do those detections match the actual WIFI environment? It detects 28 SSIDs (38 if you include hidden ones), And they all suggest that they are open (with no key icon). I have never experienced such an environment.

 Regarding your additional questions:

> the server no longer uses my handler (code below) for endpoints not explicitly called out with webserver.on() (i.e. my code checks if a SPIFFS file is available for the URI before returning a 404).

WebServer::on and WebServer::onNotFound are different exit. AutoConnect only overrides WebServer::onNotFound.  If you use the onNotFound handler it must be AutoConnect::onNotFound. https://hieromon.github.io/AutoConnect/adnetwork.html#404-handler

> AutoConnect.handleClient() not servicing the web events the same way as webserver.handleClient()?

I am not sure what you mean by events. To begin with, AutoConnect does not handle HTTP events. It is done via the WebServer class, and AutoConnect internally depends on the WebServer::handleclient. AutoConnect is just a URL request handler from the WebServer's point of view.

Hieromon, 2022-12-14:
Closing due to can't reproduce and no feedback in over 6 months.

closed 2022-12-14 by Hieromon