Hieromon/AutoConnect #482: Update from server when using mDNS fails

DropTarget, 2022-04-17:
Thank you very much for a great library! 

I am using AutoConnect for remotely deployed ESP32s and wish to add the HTTP update functionality from a remote server.  Using the "update" example provided with the library, I added mDNS registration using espmDNS as described in the documentation. All is working fine when I access AutoConnect using the WiFi.localIP address (192.168.1.122). The update proceeds normally from the remote server.  However, if I access AutoConnect using its mDNS registered hostname (actest.local), everything works except the upload file transfer.  The catalog is retrieved from the server, but when the file is selected and "update" is clicked, the GET for the selected file is never received by the file server. The update progress page is displayed but never updates.
   
I am building the firmware with platformio.  Following is the source file which is "update.ino" with espmDNS added:

*************************************************************************************************************************************

_/*
  Update.ino, Example for the AutoConnect library.
  Copyright (c) 2019, Hieromon Ikasamo
  https://github.com/Hieromon/AutoConnect
  This software is released under the MIT License.
  https://opensource.org/licenses/MIT

  This example presents the simplest OTA Updates scheme.
  When building this sketch, you may receive a compilation error message
  similar to the following:
  - text section exceeds available space in board
  This cause is the small text block size of the predetermined partition
  table. You can avoid this error by selecting Partition Scheme: from
  Arduino IDE's Tool menu and applying Minimal SPIFFS.
*/

#if defined(ARDUINO_ARCH_ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
using WebServerClass = ESP8266WebServer;
#elif defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#include <WebServer.h>
using WebServerClass = WebServer;
#endif
#include <AutoConnect.h>
#include <espMDNS.h>                             //<--------------------Added

// Update server setting page
static const char SETUP_PAGE[] PROGMEM = R"(
{
  "title": "Update setup",
  "uri": "/setup",
  "menu": true,
  "element": [
    {
      "name": "caption",
      "type": "ACText",
      "value": "OTA update setup",
      "style": ""
    },
    {
      "name": "isvalid",
      "type": "ACText",
      "style": "color:red"
    },
    {
      "name": "server",
      "type": "ACInput",
      "label": "Update server",
      "pattern": "^((([a-zA-Z]|[a-zA-Z][a-zA-Z0-9-]*[a-zA-Z0-9]).)*([A-Za-z]|[A-Za-z][A-Za-z0-9-]*[A-Za-z0-9]))|((25[0-5]|2[0-4][0-9]|[01]?[0-9][0-9]?)(\\.(25[0-5]|2[0-4][0-9]|[01]?[0-9][0-9]?)){3})$",
      "placeholder": "Your update server address"
    },
    {
      "name": "port",
      "type": "ACInput",
      "label": "port",
      "pattern": "[0-9]{1,4}"
    },
    {
      "name": "path",
      "type": "ACInput",
[... 473 lines cut ...]
port: 8000
uri:
[ 50888][E][WiFiClient.cpp:404] write(): fail on fd 49, errno: 104, "Connection reset by peer"
[AC] Host:actest.local,/_ac,generated:/_ac, allocated
[AC] Host:actest.local,/_ac,already allocated
[AC] Host:actest.local,/_ac/update,generated:/_ac/update, allocated
[AC] Host:actest.local,/_ac/update,already allocated
[AC] CB in AHEAD /_ac/update
[AC] Query 192.168.1.101:8000/_catalog?op=list&path=
[AC] Update server responded:
{"name":"HELOSESP32Test_firmware.bin","type":"bin","date":"04/15/22","time":"21:14:35","size":1070384}
{"name":"Simple_firmware.bin","type":"bin","date":"04/15/22","time":"09:03:48","size":1095392}
{"name":"updateserver.py","type":"file"}
{"name":"update_firmware.bin","type":"bin","date":"04/16/22","time":"20:10:28","size":1295808}
[AC] Host:actest.local,/_ac/update_act,generated:/_ac/update_act, allocated
[AC] Host:actest.local,/_ac/update_act,already allocated
[AC] fetch /_ac/update,elements stored
[AC] CB in AHEAD /_ac/update_act

*************No further activity *****************
*************************************************************************************************************************************
Thinking this might be an mDNS issue, I also tried to use the arduino_mdns library, but the results were the same.  Please let me know if I can provide any additional information.

Thanks,
DropTarget

DropTarget, 2022-04-21:
Additional information:
All above tests were performed by accessing AutoConnect from Windows 10 using Firefox browser.  I just tried it on my Windows 11 laptop with Firefox and it works fine.  This leads me to believe there might be an MDNS issue with Windows 10.

Hieromon, 2022-04-21:
@DropTarget Having said that, I too will try with the sketch code you posted. I may get different results in my environment. I'll keep you posted if I find out anything.

DropTarget, 2022-04-22:
@Hieromon
I opened port 5353 for UDP. I am able to resolve actest.local via powershell regardless of any wait time between tries.  It doesn't appear that ESP32 requires periodic update of MDNS.
Update result hang is the same on Windows 10 with port 5353 open. I have added additional AC_DEBUG statements in an effort to home in on where the behavioral difference is, but have not had any success yet.  I might need to set up JTAG debug to get to the root cause.

Hieromon, 2022-04-22:
@DropTarget I feel like this is not a cause with the Windows 10 mDNS responder and still think something happens on the ESP32 side.
I noticed in your log that it includes a connection reset by peer error. If this is the cause, I believe the XHR from AutoConnectUpdate is pending.

DropTarget, 2022-04-28:
@Hieromon I turned on the inspector and never saw any pending packets.  When using MDNS, it never makes it past the first update progress as seen here:
[image]

I have attached two wireshark captures, success and fail. In these traces, the nodes are:
192.168.1.113 Windows 10 Web client
192.168.1.124 Update server running updateserver.py -b 192.168.1.124
192.168.1.125 ESP32 running the test firmware 

In the success trace, note that just before the transfer starts at packet #757 is an ARP request from the web client requesting the esp32 IP address.  This is followed immediately by a response, and update progress, and the data transfer.

In the fail trace, at packet #842, is the corresponding ARP request but no response.  This is the last we hear from the web client or the update server.
 
Do you have any idea why the esp32 isn't responding to the ARP request?

[wireshark.zip](https://github.com/Hieromon/AutoConnect/files/8579288/wireshark.zip)

DropTarget, 2022-04-28:
@Hieromon I added WiFi.setSleep(false); immediately after successful return from Portal.begin but it made no difference. Please let me know if I can supply any additional information.

DropTarget, 2022-05-03:
@Hieromon 
I've not noticed any other anomalies in the network that would lead to a clue on this issue. It appears that I have one device in the network with a class A address.  It's an iRobot Roomba that connects to the cloud.  If I get a chance, I'll disconnect it from the network, reset the gateway, and try again. I confirmed again today that this doesn't seem to be an issue with Windows 11 acting as web client. This could simply be due to timing or other factors not causing the same set of conditions.  I also confirmed successful update operation with Ubuntu Server 20.0.4 acting as web client.  If I get any additional information, I'll let you know.  In the meantime, we can close this issue since I have workarounds.

Thanks for your help and a fantastic library.

closed 2022-05-03 by DropTarget