Hieromon/AutoConnect #488: Uploading file from Web Browser on Android Phone

ozturkahmetcevdet, 2022-04-29:
Hello Folks,
First of all, thanks for this beautiful library. The interface is great and very easy to use and stable.

"Choose file" (AutoConnectUploadHandler -> HTTPUpload) does not bring up the file selection page when I want to "update" the web page via Portal on Android phone. But it works fine on other operating systems and computers.

Is there a way you can suggest me?

Hieromon, 2022-04-29:
@ozturkahmetcevdet What is "Choose file" (AutoConnectUploadHandler -> HTTPUpload), is it regular file upload by [AutoConnectOTA](https://hieromon.github.io/AutoConnect/otabrowser.html#regular-file-uploading-using-autoconnectotaenhanced-wv120)? Your question is too vague and I don't know how to answer it.

If your sketch complies with the AutoConnect specification, but you have a question about its behavior, please provide specifics. In that case, please follow the submission form you are guided through in the [documentation](https://hieromon.github.io/AutoConnect/otabrowser.html#regular-file-uploading-using-autoconnectotaenhanced-wv120). It will help us resolve the issue quickly.

Incidentally, the HTML implementation of the "Choose file" button for AutoConnectOTA (which appears as an item in the AutoConnect menu via the **AutoConnectConfig::ota** setting) is the `input type="file"` element. You can check the support status of this HTML element for each browser at the following link, regardless of OS or device:
https://developer.mozilla.org/en-US/docs/Web/HTML/Element/input/file#browser_compatibility

Hieromon, 2022-04-30:
The adoption of input type=file in AutoConnectOTA and the non-support of this HTML element in Android WebView is a specification conflict that should be brought to the attention of other users. Therefore, this topic is moved to Discussions.