Hieromon/AutoConnect #496: Undesired AP is created without password after connecting to WiFi (question)

brawner, 2022-06-06:
After we add configure a new AP, our ESP32 will create a new AP with an seemingly randomly generated name without a password that publishes the auto connect setup page. We don't need this AP because we already provide that on the esp's webserver, so I've been looking for ways to deactivate it without conflicting with AutoConnect's code.

Is there a way to deactivate this new AP after connecting?

If it matters, we're also using ESP-NOW to connect with other boards, so we can interfere with that Wifi use either.

Thanks!

Hieromon, 2022-06-07:
@brawner 
> After we add configure a new AP, our ESP32 will create a new AP with an seemingly randomly generated name without a password that publishes the auto connect setup page. 

I am not sure what you mentioned  above.
1. AutoConnect does not randomly generate AP names or disable passwords when ESP32 is activated in WiFi_AP mode.
2. [Here](
https://hieromon.github.io/AutoConnect/adconnection.html#detects-connection-establishment-to-ap), the documentation describes an example sketch of adding an AP on the "Configure New AP" page, establishing a connection to the AP, and then deactivating the ESP32 AP mode.

Hieromon, 2022-12-14:
Closing due to no feedback in over 6 months.

closed 2022-12-14 by Hieromon