Hieromon/AutoConnect #502: AutoConnect doesn't work with ESP S3

fabiencomte, 2022-06-18:
Hello,
I cannot get a working wifi with ESP32 S3 devkit.

It fails in AutoConnect.cpp at 
      // Advance configuration for STA mode. Restore previous configuration of STA.
      // _loadAvailCredential(reinterpret_cast<const char*>(current.ssid));
      if (!_configSTA(_apConfig.staip, _apConfig.staGateway, _apConfig.staNetmask, _apConfig.dns1, _apConfig.dns2))
        return false;

Traces are
[AC] WiFi.config(IP=192.168.10.10, Gateway=192.168.10.1, Subnetmask=255.255.255.0, DNS1=192.168.10.1, DNS2=0.0.0.0)
[AC] failed, AC::begin abort

Hieromon, 2022-06-27:
@fabiencomte Under what conditions did the following logs you mention appear? Have you ever successfully connected to the desired access point with the static configuration shown there?

> [AC] WiFi.config(IP=192.168.10.10, Gateway=192.168.10.1, Subnetmask=255.255.255.0, DNS1=192.168.10.1, DNS2=0.0.0.0)
> [AC] failed, AC::begin abort

I don't have the ESP32-S3 modules so I can't reproduce the problem, but as long as I use the ESP32-S2, the phenomenon you encountered doesn't appear. The number of cores is different between S2 and S3, but it is unlikely that the difference is causing the problem.

To determine the cause, I need to know what is happening within the [WiFiGneric.cpp](https://github.com/espressif/arduino-esp32/blob/43e5e02224ef4f819376b6094189440d78d4039f/libraries/WiFi/src/WiFiGeneric.cpp#L86-L216) procedure. Please enable the core debug to Verbose in the board menu of Arduino IDE and then post FULL log. I will use its log as a clue to continue an investigation.

Hieromon, 2022-07-20:
@fabiencomte I was able to obtain an ESP32-S3 and tried to reproduce the issue.
In my environment, I have no reproduce the event of failure in static IP assignment as you have encountered. WiFi.Config with static IP works fine.

But, I discovered another strange phenomenon. Credentials are not saved to the Preferences after a WiFi connection is established. (Preferences API return code does not report failure)
Something unexpected seems to be happening with ESP32-S3, but I have not yet identified the cause. As progress is made, I will post here.

Also, could you post the `CORE_DEBUG_LEVEL=5` traces?

Hieromon, 2022-07-20:
@fabiencomte I have identified the cause of the credentials not being saved.
I tested Arduino IDE 1.8.19 w/esp32 core 2.0.4, it works fine.
In esp32 arduino-core 2.0.3 environment, neither Preferences nor SPIFFS work properly with ESP32-S3.
If you are using PlatformIO as your build environment and encountering this problem, it is because the version of esp32 arduino-core is not compliant.The latest [version 5.0.0 of the platform-espressif32](https://github.com/platformio/platform-espressif32/releases/tag/v5.0.0) framework is esp32 arduino-core 2.0.3; 2.0.4 is not yet supported.

ark02, 2022-08-12:
I can confirm that credentials are also not being saved on 

ESP32D WROOM, WROVER 
and ESP32E - DevKitC

using platformio 6.1.3 core, 
5.1.0 platform-espressif32

Hieromon, 2022-12-14:
Closing due to can't reproduce. I have verified and AutoConnect with ESP32 S3 devkit works fine.

closed 2022-12-14 by Hieromon