Hieromon/AutoConnect #519: PIO: PageBuilder.h:239:21: error: 'File' was not declared in this scope

JanHBade, 2022-08-19:
Hello,

first thanks for your great work in this lib...

I have an error....
```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220619 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- Wire @ 2.0.0
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- TFT_eSPI @ 2.4.75
|   |-- SPI @ 2.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|-- ArduinoJson @ 6.19.4
|-- AutoConnect @ 1.3.6
|   |-- PageBuilder @ 1.5.3
|   |   |-- FS @ 2.0.0
|   |   |-- LittleFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- SPIFFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- WebServer @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- ArduinoJson @ 6.19.4
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- SD @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- DNSServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- EEPROM @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- Preferences @ 2.0.0
|   |-- Ticker @ 2.0.0
|   |-- HTTPClient @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- HTTPUpdate @ 2.0.0
|   |   |-- HTTPClient @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |   |-- WiFi @ 2.0.0
|   |   |-- Update @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- WebServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- FS @ 2.0.0
|-- WiFi @ 2.0.0
|-- PageBuilder @ 1.5.3
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\Release\src\HttpUpdateServer.cpp.o
Compiling .pio\build\Release\src\main.cpp.o
Generating partitions .pio\build\Release\partitions.bin
Compiling .pio\build\Release\lib47c\SPI\SPI.cpp.o
Compiling .pio\build\Release\libb80\FS\FS.cpp.o
Compiling .pio\build\Release\libb80\FS\vfs_api.cpp.o
Compiling .pio\build\Release\liba73\SPIFFS\SPIFFS.cpp.o
Compiling .pio\build\Release\lib723\Wire\Wire.cpp.o
Archiving .pio\build\Release\lib47c\libSPI.a
In file included from src/main.cpp:6:
.pio/libdeps/Release/TFT_eSPI/TFT_eSPI.h:882:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
Compiling .pio\build\Release\lib244\WiFi\WiFi.cpp.o
Compiling .pio\build\Release\lib244\WiFi\WiFiAP.cpp.o
Compiling .pio\build\Release\lib244\WiFi\WiFiClient.cpp.o
Compiling .pio\build\Release\lib244\WiFi\WiFiGeneric.cpp.o
Archiving .pio\build\Release\libb80\libFS.a
Compiling .pio\build\Release\lib244\WiFi\WiFiMulti.cpp.o
Archiving .pio\build\Release\liba73\libSPIFFS.a
Compiling .pio\build\Release\lib244\WiFi\WiFiSTA.cpp.o
Archiving .pio\build\Release\lib723\libWire.a
In file included from src/main.cpp:9:
.pio/libdeps/Release/PageBuilder/src/PageBuilder.h:239:21: error: 'File' was not declared in this scope
     std::shared_ptr<File> _file;      /**< File for file: mold */
                     ^~~~
.pio/libdeps/Release/PageBuilder/src/PageBuilder.h:239:21: note: suggested alternative:
In file included from .pio/libdeps/Release/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:137,
                 from .pio/libdeps/Release/TFT_eSPI/TFT_eSPI.h:72,
                 from src/main.cpp:6:
C:/Users/jhb/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.h:47:7: note:   'fs::File'
 class File : public Stream
       ^~~~
In file included from src/main.cpp:9:
.pio/libdeps/Release/PageBuilder/src/PageBuilder.h:239:25: error: template argument 1 is invalid
     std::shared_ptr<File> _file;      /**< File for file: mold */
                         ^
Compiling .pio\build\Release\lib244\WiFi\WiFiScan.cpp.o
*** [.pio\build\Release\src\main.cpp.o] Error 1
```

Hieromon, 2022-08-20:
@JanHBade 

Is there an FS.h include conflict with TFT_eSPI_ESP32.h:137?
https://github.com/Bodmer/TFT_eSPI/blob/333f0f845c27a8ccca28b6f2bb97ce3a8dfc0e0a/Processors/TFT_eSPI_ESP32.h#L137

> In file included from .pio/libdeps/Release/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:137,
>                 from .pio/libdeps/Release/TFT_eSPI/TFT_eSPI.h:72,
 >                from src/main.cpp:6:
>C:/Users/jhb/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.h:47:7: note:   'fs::File'
>

I tried to reproduce the issue, but I have no error in my environment with the following #include directive sequence:

```cpp
#include <FS.h>
#include <SPI.h>
#include <TFT_eSPI.h>
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
```

```log
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (5.1.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20004.220818 (2.0.4)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 80 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AutoConnect @ 1.3.6
|   |-- PageBuilder @ 1.5.3
|   |   |-- FS @ 2.0.0
|   |   |-- LittleFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- SPIFFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- WebServer @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- ArduinoJson @ 6.19.4
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- SD @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- DNSServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- EEPROM @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- Preferences @ 2.0.0
|   |-- Ticker @ 2.0.0
|   |-- HTTPClient @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- HTTPUpdate @ 2.0.0
|   |   |-- HTTPClient @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
[... 130 lines cut ...]
Compiling .pio\build\nodemcu-32s\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\nodemcu-32s\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\nodemcu-32s\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\nodemcu-32s\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\nodemcu-32s\FrameworkArduino\main.cpp.o
Compiling .pio\build\nodemcu-32s\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\nodemcu-32s\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\nodemcu-32s\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\nodemcu-32s\libFrameworkArduino.a
Archiving .pio\build\nodemcu-32s\libdd4\libAutoConnect.a
Linking .pio\build\nodemcu-32s\firmware.elf
Retrieving maximum program size .pio\build\nodemcu-32s\firmware.elf
Checking size .pio\build\nodemcu-32s\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.3% (used 40300 bytes from 327680 bytes)
Flash: [========  ]  79.7% (used 1044201 bytes from 1310720 bytes)
Building .pio\build\nodemcu-32s\firmware.bin
esptool.py v3.3
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
```

If you can show me the minimal code that can reproduce the problem, I can look into it some more.
See [FAQ](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository), and [more](https://hieromon.github.io/AutoConnect/faq.html#compile-error-due-to-file-system-header-file-not-found).

JanHBade, 2022-08-20:
#include <FS.h> does the trick :)

closed 2022-08-20 by JanHBade