Hieromon/AutoConnect #557: Autoconnect captive portal on Windows

fruktiz32, 2023-01-15:
Hello. I'm using this library for long time but noticed that it does not start captive portal on Windows. It connects to SoftAP but can not open captive page on Windows. Android works perfect as always. What I do wrong?

`WebServer Server;
AutoConnect       portal(Server);
AutoConnectConfig config("MyApp", "123456789"); //Set SoftAP SSID and password
Preferences preferences;

void setup() {
  
  Serial.begin(115200);

  if (WiFi.status() != WL_CONNECTED) {

    Serial.println(config.apid);
    Serial.println(config.psk);
    Serial.println(config.apip);

  }
  config.autoReconnect = false; // Enable saved past credential by autoReconnect option,even once it is disconnected.
  
  config.title = "MyApp";
  config.ota = AC_OTA_BUILTIN; //Adds OTA to page
  portal.config(config);
  portal.load(addonJson);
  portal.on("/saved", feelsOn, AC_EXIT_AHEAD);
`

Hieromon, 2023-01-15:
I can not reproduce on my Windows 10 Pro environment.
A captive portal automatically rises with the default browser, whether it's Chrome or Edge.
Could it be that your browser extension is preventing the captive portal from opening?

fruktiz32, 2023-01-15:
> I can not reproduce on my Windows 10 Pro environment. A captive portal automatically rises with the default browser, whether it's Chrome or Edge. Could it be possible that the browser extensions are preventing you from opening the captive portal?

Oh, yes I just checked on Microsoft Edge and it works perfect... Does not open on Chrome and Firefox. Thank you!

closed 2023-01-15 by fruktiz32