Hieromon/AutoConnect #560: Wrong Arcithecture

Tipmanfrank, 2023-01-19:
I am using Arduino IDE 2.0.3. When I compile I get the following fault:

In file included from C:\Users\FrankRosendahlLarsen\AppData\Local\Arduino15\libraries\SD\src/utility/Sd2Card.h:26:0,
                 from C:\Users\FrankRosendahlLarsen\AppData\Local\Arduino15\libraries\SD\src/utility/SdFat.h:29,
                 from C:\Users\FrankRosendahlLarsen\AppData\Local\Arduino15\libraries\SD\src/SD.h:20,
                 from c:\Data\Privat\Arduino\libraries\AutoConnect\src/AutoConnectCredential.h:37,
                 from c:\Data\Privat\Arduino\libraries\AutoConnect\src/AutoConnectCore.hpp:35,
                 from c:\Data\Privat\Arduino\libraries\AutoConnect\src/AutoConnectCoreImpl.hpp:13,
                 from c:\Data\Privat\Arduino\libraries\AutoConnect\src/AutoConnectExt.hpp:15,
                 from c:\Data\Privat\Arduino\libraries\AutoConnect\src/AutoConnectExtImpl.hpp:13,
                 from c:\Data\Privat\Arduino\libraries\AutoConnect\src/AutoConnect.h:15,
                 from C:\Data\Privat\Arduino\Projekter\Autoconnect1_DISP_IP_v1_virker\Autoconnect1_DISP_IP_v1_virker.ino:3:
C:\Users\FrankRosendahlLarsen\AppData\Local\Arduino15\libraries\SD\src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
Multiple libraries were found for "SD.h"
  Used: C:\Users\FrankRosendahlLarsen\AppData\Local\Arduino15\libraries\SD
  Not used: C:\Users\FrankRosendahlLarsen\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD
exit status 1

Compilation error: exit status 1

I have chosen ESP32 Dev board before compiling. What causes this fault? I have the same fault when I compile any of the examples.

Hieromon, 2023-01-20:
Your Arduino library environment is confusing due to several different SD libraries.

> C:\Users\FrankRosendahlLarsen\AppData\Local\Arduino15\libraries\SD\src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.
#error Architecture or board not supported.

This message is emitted by the SD library for the Arduino **AVR**, **not the Espressif core**. Are you sure you have chosen the right board?
Additionally you are using ESP32 arduino core version **1.0.6**. Is ffthis your intention?
> C:\Users\FrankRosendahlLarsen\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD

Tipmanfrank, 2023-01-20:
I uninstalled my Arduino 1 and 2 and removed all libraries. Then installed 2.0.3 and now it works.
I hate the 2.0.3 for compiling very slowly.
Problem solved.

Hieromon, 2023-01-20:
Close due to solved.

closed 2023-01-20 by Hieromon