Hieromon/AutoConnect #561: ACCheckbox/AutoConnectCheckbox doesn't work for esp8266 from AutoConnect 1.3.7

Evgen2, 2023-01-19:
[CheckBoxIssue.zip](https://github.com/Hieromon/AutoConnect/files/10456381/CheckBoxIssue.zip)
Construction like 

```
ACText(Caption, "Hello, world", "", "", AC_Tag_DIV);
ACCheckbox(ChB1, "chb", "chb");
//or AutoConnectCheckbox ChB1("ChB1","1", "m", false, AC_Behind , AC_Tag_BR);
ACSubmit(Apply, "Apply", HELLO_URI);
// AutoConnectAux for the custom Web page.
AutoConnectAux helloPage(HELLO_URI, "Hello", true, { Caption, ChB1,  Apply });

```

work well for esp8266 with AutoConnect 1.3.3 and lead to  Exeption(3) with 1.3.7 or 1.41
Code for esp32 work
[image]

Hieromon, 2023-01-20:
I have reproduced the issue. It is AutoConnect bug. The readout from PROGMEM is missing.

I will fix it soon and release as AutoConnect v1.4.2. You can evaluate the fix ​before the release with the patches branch.

Thank you for reporting the issue.

closed 2023-02-02 by Hieromon

Hieromon, 2023-02-02:
Merged #575