Hieromon/AutoConnect #567: Compilation error: exit status 1

escolta2013, 2023-01-23:
Arduino IDE 2.03
AutoConnect v. 1.4.1 and 1.4.0 
Board: LOLIN (wemos) D1 R1 & mini
Sketchs: all AutoConnect's examples
Error: Compilation error: exit status 1 

Two week ago all worked well! 
Any idea the what's the problem?

Hieromon, 2023-01-24:
Which version of the ESP8266 core version are you using?
AutoConnect does **not yet support** [ESP8266 Arduino Core 3.1.0](https://github.com/esp8266/Arduino/releases/tag/3.1.0) or later.

Currently working on it.

escolta2013, 2023-01-25:
Yes, I'm using ESP8266 Arduino Core 3.1.1

libraries\AutoConnect\src\AutoConnectElementBasis.h:130:38: warning: unused parameter 'value' [-Wunused-parameter]
  130 |   virtual void  response(const char* value) {}
      |                          ~~~~~~~~~~~~^~~~~
exit status 1
Compilation error: exit status 1

This is the same error for all autoconnect examples

escolta2013, 2023-01-25:
oh ok 
Thanks

Hieromon, 2023-01-27:
@escolta2013 I regret to inform you of bad news.

When I finished supporting ESP8266 Arduino core 3.1.1 and started testing, the core itself consumed more heap memory than core 3.0.2. As a result, there are more cases where the HTML generation of the AutoConnect menu runs out of memory. As far as I have verified, it is hard to say that even the *mqttRSSI.ino* example and the *Element.ino* example can be executed normally.

Currently, I am reviewing the structure that causes memory consumption, but the problem has spread to PageBuilder and it seems that it will take time to solve it.

The one staged on the [patches/553](https://github.com/Hieromon/AutoConnect/tree/patches/553) branch corresponds to ESP8266 core 3.1.1, but it still suffers from memory consumption issues. If possible, I recommend applying that branch to verify how your sketch behaves with ESP8266 core 3.1.1.

Also, an interim workaround is to use [AutoConnectCore](https://hieromon.github.io/AutoConnect/basicusage.html#reducing-binary-size) or avoid using custom web pages whenever possible.

Evgen2, 2023-01-27:
> Two week ago all worked well!

Just found on new setup of platfomio a number of errors while compiling Autoconnect code
This is done at platform = espressif8266  and platform = espressif8266@4.1.0 in platformio.ini
On old setup of  platfomio   platform = espressif8266   is equivalent to platform = espressif8266@4.0.1 and all is fine

Hieromon, 2023-01-27:
@Evgen2 
Try to compile your sketch with the [patches/553](https://github.com/Hieromon/AutoConnect/tree/patches/553) branch applied.

Evgen2, 2023-01-27:
> @Evgen2 Try to compile your sketch with the [patches/553](https://github.com/Hieromon/AutoConnect/tree/patches/553) branch applied.

PLATFORM: Espressif 8266 (4.1.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30101.0 (3.1.1)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)

Build without errors and warnings

Hieromon, 2023-01-28:
@Evgen2 Thank you for results.

Even if the sketch is successfully built, AutoConnect may not function properly at runtime. If the browser receives like following response from the AutoConnect menu, the ESP8266 heap is exhausted. If the browser shows a corrupted page response from the AutoConnect menu as shown below, the ESP8266 heap is depleted.

[image]

Or, you can see this error factor when you specify `build_flags=-DDEBUG_ESP_HTTPSERVER` in `platformio.ini`.

```
HTTPServer: error short send after timeout (0 < 1270)
```

ESP8266 Arduino Core 3.1.0 or later is prone to heap memory exhaustion. In particular, AutoConnect uses many String objects with long durations of existence, causing a rapid decrease in remaining heap space. Memory depletion is especially noticeable in captive portal conditions.

A temporary workaround is to allocate a portion of the *ICHACHE* (ESP8266 instruction cache) to the heap area.

**In platformio.ini**

```
build_flags = -DPIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED
```
See:
[MMU - Adjust the Ratio of ICACHE to IRAM](https://arduino-esp8266.readthedocs.io/en/latest/mmu.html) - ESP8266 Arduino Core
[MMU - Adjusting ICACHE to IRAM ratio](https://docs.platformio.org/en/stable/platforms/espressif8266.html#id22) - Platformio Espressif8266

Besides, *PageBuilder*'s chunked response is not compatible with ESP8266 core 3.1.0 WebServer library.

Work on these renovations is ongoing.

Hieromon, 2023-02-02:
Merged #575

closed 2023-02-02 by Hieromon